ckb-next/ckb-next #482: System Extension by "Fumihiko Takayama" not showing up in Security & Privacy on macOS Catalina

Espionage724, 2019-10-08:
After installation of ckb-next 0.4.2, I got no prompt of the system extension. Upon starting ckb-next, I get two Warnings, one saying to allow that extension, and the other to allow the daemon in Input monitoring.

The extension prompt is not shown on the Security & Privacy page, nor is ckb-next present in Input monitoring.

I haven't done a reboot or device hot-plug yet.

- macOS 10.15 Catalina (19A583)
- ckb-next 0.4.2 (non-Legacy)
- Corsair Gaming K70 RGB RAPIDFIRE Keyboard
- Corsair Gaming HARPOON RGB Mouse
- [ckb-next-report-1570553735.gz](https://github.com/ckb-next/ckb-next/files/3703628/ckb-next-report-1570553735.gz)

tatokis, 2019-10-08:
Does anything happen if you run the following commands?
```
launchctl unload /Library/LaunchAgents/org.ckb-next.daemon_agent.plist
launchctl load /Library/LaunchAgents/org.ckb-next.daemon_agent.plist
launchctl start org.ckb-next.daemon

buddybuddybuddy, 2019-10-09:
hey tatokis 

[image]

buddybuddybuddy, 2019-10-09:
also mouse is still connected, and its not showing up
[image]

tatokis, 2019-10-09:
If you didn't get a restart prompt, then you can manually restart the daemon with

```
sudo launchctl unload /Library/LaunchDaemons/org.ckb-next.daemon.plist
sudo launchctl load /Library/LaunchDaemons/org.ckb-next.daemon.plist
```

That should get you up and running. I'm not sure why this didn't work in the first place though...

buddybuddybuddy, 2019-10-09:
asked for a password.
[image]

tatokis, 2019-10-09:
Yes, you have to enter the two lines separately. The first time it will ask for your password.
You can close it and try again

buddybuddybuddy, 2019-10-09:
ahhh

buddybuddybuddy, 2019-10-09:
legend 
[image]

exelban, 2019-10-10:
@tatokis the same problem with Corsair K70.
After update to Catalina and ckb-next 0.4.2 (non-legacy). I see the text: No devices connected.
Also in Input monitoring I don't see a ckb-next daemon.

I tried this to restart the daemon:
```
sudo launchctl unload /Library/LaunchDaemons/org.ckb-next.daemon.plist
sudo launchctl load /Library/LaunchDaemons/org.ckb-next.daemon.plist
```

But it still doesn't show.

I tried to restart the laptop and reinstalling the ckb. Nothing doesn't help.

MikesTooLz, 2019-10-11:
I already had Catalina installed and had it working via those commands and manually adding the daemon .

Got the prompt that a new version of ckb-next was available and that it had support for Catalina. so I installed it. After installed both keyboard and mouse would not show up. I unchecked the input permissions in preverences and then rechecked it and after 10 seconds it auto rand the commands and prompted for my password. after that it seemed like it was working but for some reason ever 10 minutes or so (random times) the mouse just stops working for 5-10 setons and the lights go out and come back on like if it was unplugged and reconnected. After 1 day of using the new version the mouse just stopped showing up in cks-next all together until I want and ran the unload/reload commands manually.

exelban, 2019-10-11:
@jpagano I tried these steps, but permission still not showing.
Also after: `/launchctl unload /Library/LaunchAgents/org.ckb-next.daemon_agent.plist `

I receive `/Library/LaunchAgents/org.ckb-next.daemon_agent.plist: Could not find specified service`.

And `launchctl load /Library/LaunchAgents/org.ckb-next.daemon_agent.plist` saying: `/Library/LaunchAgents/org.ckb-next.daemon_agent.plist: Service is disabled`.

But the file `/Library/LaunchAgents/org.ckb-next.daemon_agent.plist` exist.

tatokis, 2019-10-11:
Try `launchctl load -w /Library/LaunchAgents/org.ckb-next.daemon_agent.plist`

I have absolutely no idea why this happens. I just did a clean installation of ckb-next and I got the permission prompt...

tatokis, 2019-10-11:
@MikesTooLz Can you attach another report please?

MikesTooLz, 2019-10-12:
[ckb-next-report-1570915299.gz](https://github.com/ckb-next/ckb-next/files/3721168/ckb-next-report-1570915299.gz)

been having to run the commands every 10 minutes to get the mouse to work.

Espionage724, 2019-10-12:
I've done a few macOS complete reinstalls:

- ckb-next always said that the extension wasn't allowed in Security & Privacy; there was no prompt for this that I could see, but it was always resolved by a reboot
- ckb-next-daemon never showed under Input Monitoring even after a reboot
- ckb-next-daemon only showed up under Input Monitoring when a keyboard was plugged in (couldn't get it with just a Corsair mouse), and I believe after re-loading the driver (either the `launchctl load` or `launchctl start` command)

Espionage724, 2019-10-12:
> ...after that it seemed like it was working but for some reason ever 10 minutes or so (random times) the mouse just stops working for 5-10 setons and the lights go out and come back on like if it [...]

Thought I was the only one with this problem; I mentioned it as a feature request for ckb-next to somehow recover the mouse after this happens: https://github.com/ckb-next/ckb-next/issues/484

iCUE recovers it somehow usually after 2-3 seconds of it disconnecting

samuelfine, 2019-10-13:
These instructions helped me get up and running with macOS Catalina, thanks!

Quick tip for anyone who's still having problems: my K65 is connected to my MacBook through a USB hub, and I needed to disconnect and reconnect the keyboard from the _hub itself_ - not the hub from the computer - to get the system to recognize the new keyboard. I don't remember the exact order of operations, but I'd recommend unplugging the keyboard from the hub and reconnecting before unloading/loading the driver if you're not seeing the Input Monitoring request.

exelban, 2019-10-14:
Ok, maybe it can be the problem.
I have MacBook Pro 2018. So I connect k70 throw 2 hubs. One to type C and another USB type A hub. I tried to reconnect the keyboard in different ways.

Deamon still not showing(

exelban, 2019-10-17:
@tatokis it's strange. But after update to 19A602 (macOS update), everything works.

Handsomfreddy, 2020-12-20:
Hello, I am having this same problem. I'm running Big Sur. Followed all the instructions above but the daemon does not get added to SIP. I added the CKB app to Full Disk Access under SIP but I have no ability to add it to Input monitoring. Any ideas? Thank you very much. Happy holidays!

Ravenslofty, 2020-12-20:
There were some suggestions given in #660.

Handsomfreddy, 2020-12-20:
Thank you reading now. ICUE is awful – totally lags my system and then lags the keyboard colors too.  Such garbage software. I miss my CKB-NEXT!

Handsomfreddy, 2020-12-20:
Managed to get the daemon loaded by following the manual instructions for adding ckb-next-daemon into >Security & Privacy >Privacy > Input Monitoring located in #660 - however CKB still shows the following message. Tried unplugging, replugging, waiting, restarting to no avail.  Any help would be greatly appreciated. 

[image]

closed 2022-06-27 by tatokis