ckb-next/ckb-next #486: Mission Control does not bind to any key, as a program or Function Key

NathanJava, 2019-10-17:
Hello, I'm trying to rebuild CKB after Catalina destroyed it, and everything is going well except I can no longer bind the mission control app to my Sniper key on Harpoon RGB..

I tried to bind it to a program (open '/Applications/Mission Control.app') but that no longer works. 
I also tried to use the F3 key for the same effect to no avail. I have reinstalled and Uninstalled several times but the app refuses to let mission control be bound to any key..

The daemon is permitted to input monitoring and I have also unplugged and replugged all peripherals.

Attached below is my report generated just now.

Thanks for your work!

[ckb-next-report-1571271172.gz](https://github.com/ckb-next/ckb-next/files/3736882/ckb-next-report-1571271172.gz)

NathanJava, 2019-10-22:
**[Fix This Issue:]**

Hey all, after you allow the next-daemon in S&P and this issue persists, be sure to check your Mission Control Prefs to verify that Catalina did not reset your bind for the application.

Below is a screenshot after I switched from Catalina default to a function key, which I then binded to the sniper button on my Harpoon to fix the issue. Hope this helps someone, anyone!

[image]

ItzSwirlz, 2019-12-30:
If this is fixed, please close it.

closed 2020-05-14 by ZirconiumX