ckb-next/ckb-next #488: Start cks-next at login not working on macOS Catalina

Espionage724, 2019-10-18:
- Settings -> [  ] Start ckb-next at login

I've done several macOS clean installs, and most of the time, that checkbox isn't enabled by-default, nor does it stick if I check it, quick ckb-next, and re-open it (it always starts back up without the checkbox being checked).

- I don't know if I've been doing anything differently between installs to affect it, but sometimes that checkbox would be auto-checked and ckb-next would auto-start fine after install
- For the times where it isn't auto-checked, seemingly no amount of re-installs or reboots fixes it

- [ckb-next-report-1571396487.gz](https://github.com/ckb-next/ckb-next/files/3743559/ckb-next-report-1571396487.gz)
- macOS 10.15 Catalina (19A602)
- ckb-next 0.4.2 (non-Legacy)
- Corsair Gaming HARPOON RGB Mouse
- Corsair Gaming K70 RGB RAPIDFIRE Keyboard

akshaylive, 2019-10-18:
+1 this also happens when the system goes to sleep and wake up.

The daemon is running but the bindings don't work.

```
/Library/Application Support/ckb-next-daemon
```

Espionage724, 2019-10-20:
Seems like [org.ckb-next.plist](https://github.com/ckb-next/ckb-next/blob/master/macos/org.ckb-next.plist) isn't being added to the `/Library/LaunchAgents` folder. Manually adding that file works fine with 0.4.2 and makes ckb-next start on login.

Here's a command to download it to that folder:

```
sudo curl 'https://raw.githubusercontent.com/ckb-next/ckb-next/master/macos/org.ckb-next.plist' -o '/Library/LaunchAgents/org.ckb-next.plist'
```

closed 2022-06-27 by tatokis