ckb-next/ckb-next #492: Manjaro Linux k55 keyboard no keys respond once ckb-next-daemon starts

buster802, 2019-10-24:
I have tried to get my corsair k55 keyboard setup with ckb but everytime I install ckb both via pacman and manual installation everything starts up but once the daemon detects my keyboard it stops functioning except for the lights minus the numlock, winlock, and capslock keys they can't turn on.I can unplug it and get it to work for a moment before it detects it again  I have installed this both on Manjaro and on Ubuntu and have tried with the usbquirks from the troubleshooting page.
I updated the firmware on the keyboard via ICUE on windows to the latest version so that should not be an issue.

I frequently see this when I do systemctl status ckb-next-daemon 

> os_usbsend (via device_keyboard.c:40): Connection timed out, res = 0xffffffff 

Side note when I try to configure anything on the keyboard via ckb-next such as lighting that does not do anything either

buster802, 2019-10-25:
I removed usbhid and replaced  it with usbcore but still got the same issues with nothing obvious to me in the report indicating any change other than  the core quirks being their
https://pastebin.com/J4zpC8TQ
Could IOMMU be the issue since I do have that enabled but I need it for something.

RenatoTeixeira, 2019-10-29:
I have a corsair strafe keyboard and seem to have the same issue, keyboard almost completely unresponsive on the latest linux manjaro. Actually had to reinstall the system as i didn't have another keyboard to try.

tatokis, 2019-10-29:
You can either set the keyboard to bios mode (F1 + WinLock) or boot to recovery and disable the daemon if this happens again.

@RenatoTeixeira which kernel are you on?

RenatoTeixeira, 2019-10-29:
I'm on 5.3.7-2-MANJARO, this was a few days ago tho.

tatokis, 2019-10-29:
I just tried both my K70 and M65 on 5.3.7 (custom) and it works.

tatokis, 2019-10-29:
Can either of you try an older kernel?

gravityfargo, 2024-05-02:
This issue has been marked as stale due to age and/or lack of activity. 

Please feel free to reopen it if you have more to add or if the issue still needs attention. However, it would be best to [open a new issue](https://github.com/ckb-next/ckb-next/issues/new/choose) and provide a newly generated report issue using our updated templates.

closed 2024-05-02 by gravityfargo