ckb-next/ckb-next #495: Per-application modes?

Aimela, 2019-10-27:
Currently, this application only seems to support manual switching between profiles/modes. What I would like is if I could set certain modes to automatically activate when a specific application is launched and fallback to the previously used mode when the application is closed. for example, this would benefit those who want different bindings for specific game, especially when considering devices with extra buttons(macro buttons and the Scimitar's side buttons).

I would definitely appreciate if such a feature was added(bonus points if it works with application in Wine).

tatokis, 2019-10-28:
I worked on it some time ago https://github.com/ckb-next/ckb-next/tree/xcb-profile-switch
There are still a few issues that need to be fixed, but feel free to tewst it

TheJuiceFR, 2020-11-28:
> I worked on it some time ago https://github.com/ckb-next/ckb-next/tree/xcb-profile-switch
> There are still a few issues that need to be fixed, but feel free to tewst it

Any update on that?

tatokis, 2020-11-29:
I'll rebase it and try again, but if I remember correctly there were some GUI bugs that were really annoying to fix, so I kinda gave up on it.

tatokis, 2020-12-09:
I finished it and it's now in master https://github.com/ckb-next/ckb-next/commit/de7d6468528aeb19d0cc01b2903ec20ae8b19699

closed 2020-12-09 by tatokis