ckb-next/ckb-next #496: ST100 sometimes gets detected twice and animations stop working

kraxarn, 2019-10-29:
## General information
Sometimes when starting the ckb-next application, my ST100 stand seems to be detected twice and the animations stop working. From the log:
```
ckb-next: Corsair RGB driver 0.4.2
[I] Root controller ready at /dev/input/ckb0
[I] Connecting Corsair Gaming K95 RGB PLATINUM Keyboard at /dev/input/ckb1
[I] Connecting Corsair Gaming SCIMITAR PRO RGB Mouse at /dev/input/ckb2
[I] Connecting Corsair ST100 LED Driver at /dev/input/ckb3
[I] Starting input thread for /dev/input/ckb3
[I] Received identification packet: 0e 01 00 00 00 01 00 00 02 00 01 00 1c 1b 34 0a 08 00 00 00 c2 ff 40 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 0>
[I] Setup finished for /dev/input/ckb3
[I] Starting input thread for /dev/input/ckb2
[E] process_input_urb (keymap.c:362): Unknown mouse data received in input thread 10 from endpoint 81
[E] process_input_urb (keymap.c:362): Unknown mouse data received in input thread 02 from endpoint 81
[I] Starting input thread for /dev/input/ckb1
[I] Received identification packet: 0e 01 00 00 01 01 00 01 14 03 01 01 1c 1b 3e 1b 01 01 01 30 c1 ff 40 01 46 00 01 01 00 00 00 00 00 00 00 00 01 00 00 01 00 00 02 01 02 01 aa 00 6c dc 02 00 00 00 00 00 0>
[I] Setup finished for /dev/input/ckb2
[I] Received identification packet: 0e 01 00 00 01 01 00 01 21 03 04 00 1c 1b 2d 1b 01 01 00 01 c0 ff 40 01 03 00 00 00 02 00 03 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 0>
[I] Setup finished for /dev/input/ckb1
[I] Connecting Corsair ST100 LED Driver at /dev/input/ckb4
[I] Starting input thread for /dev/input/ckb4
[I] Received identification packet: 0e 01 00 00 00 01 00 00 02 00 01 00 1c 1b 34 0a 08 00 00 00 c2 ff 40 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 0>
[I] Setup finished for /dev/input/ckb4
[E] process_input_urb (keymap.c:362): Unknown mouse data received in input thread 00 from endpoint 81
[E] process_input_urb (keymap.c:362): Unknown mouse data received in input thread 00 from endpoint 81
[E] os_usbsend (via device_mouse.c:64):  Device or resource busy, res = 0xffffffff
[I] Attempting reset...
[E] usbclaim (usb_linux.c:520): Failed to claim interface 0: Device or resource busy
[E] os_resetusb (via usb.c:613): resetusb failed: Device or resource busy
[E] usb_tryreset (usb.c:621): Reset failed. Disconnecting.
[I] Disconnecting /dev/input/ckb3
[I] Removed device path /dev/input/ckb3
[E] process_input_urb (keymap.c:362): Unknown mouse data received in input thread 10 from endpoint 81
[E] process_input_urb (keymap.c:362): Unknown mouse data received in input thread 02 from endpoint 81
[E] process_input_urb (keymap.c:362): Unknown mouse data received in input thread 00 from endpoint 81
[E] process_input_urb (keymap.c:362): Unknown mouse data received in input thread 00 from endpoint 81
```
I only have one ST100 connected, but it seems to be detected in both `/dev/input/ckb3` and `/dev/input/ckb4` and it therefor gets busy and disconnected. Restarting the application (not the daemon) fixes the issue.

### Source
Package from AUR.

## Logs & crash reports
[ckb-next-report-1572352627.gz](https://github.com/ckb-next/ckb-next/files/3783541/ckb-next-report-1572352627.gz)

## What is the program's current behavior?
Animations on ST100 stops working until the application is restarted.

## What is the expected behavior of the program?
The animations on ST100 works.

## Any other notes & comments?
My Corsair keyboard and mouse still has working animations and work as intended, it is only the ST100 animations that have issues.
Linux 5.3.7, Manjaro Linux, KDE Plasma 5.17.1, Qt 5.13.1

tatokis, 2019-10-29:
Can you attach the last few lines of dmesg immediately after the issue happens?

kraxarn, 2019-10-29:
The issue happens right after boot (on login), but to try and capture information about the ST100:
(unless you want something specific)
```
[    3.306007] usb 1-5: New USB device found, idVendor=1b1c, idProduct=0a36, bcdDevice=93.02
[    3.306009] usb 1-5: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[    3.306009] usb 1-5: Product: Corsair ST100 USB 2.0 Hub
[    3.306010] usb 1-5: Manufacturer: Corsair Components
[    3.306010] usb 1-5: SerialNumber: v5.0
[    3.307319] hub 1-5:1.0: USB hub found
[    3.307601] hub 1-5:1.0: 4 ports detected
...
[    3.800118] usb 1-5.1: New USB device found, idVendor=1b1c, idProduct=0a32, bcdDevice= 1.00
[    3.800124] usb 1-5.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[    3.800128] usb 1-5.1: Product: Corsair ST100 Headset Output
[    3.800132] usb 1-5.1: Manufacturer: Corsair Components Inc.
[    3.800135] usb 1-5.1: SerialNumber: v0.6
[    3.805270] input: Corsair Components Inc. Corsair ST100 Headset Output as /devices/pci0000:00/0000:00:14.0/usb1/1-5/1-5.1/1-5.1:1.3/0003:1B1C:0A32.0006/input/input21
[    3.859637] hid-generic 0003:1B1C:0A32.0006: input,hidraw3: USB HID v1.00 Device [Corsair Components Inc. Corsair ST100 Headset Output] on usb-0000:00:14.0-5.1/input3
...
[    4.270053] usb 1-5.2: New USB device found, idVendor=1b1c, idProduct=0a34, bcdDevice= 0.02
[    4.270059] usb 1-5.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[    4.270063] usb 1-5.2: Product: Corsair ST100 LED Driver
[    4.270066] usb 1-5.2: Manufacturer: Corsair
[    4.270069] usb 1-5.2: SerialNumber: 1910040426891CAF0390D859C71B00F5
[    4.277756] hid-generic 0003:1B1C:0A34.0007: hiddev3,hidraw4: USB HID v1.11 Device [Corsair Corsair ST100 LED Driver] on usb-0000:00:14.0-5.2/input0
```

tatokis, 2022-10-03:
This should've been fixed for a while now

closed 2022-10-03 by tatokis