ckb-next/ckb-next #499: HyperX Cloud Alpha Music Visualization bug

yuri2203, 2019-11-03:
I had a problem with my Alpha + Strafe CKB.
The Mviz is listening only the Mic and not the music itself. How can I fix this?
It is not Crosstalk, because the headset works perfectly.
When I was using my Cloud Stinger, it wasnt happening.

I have been suffering with this in Ubuntu 19.10 and Fedora 30.

I tried to reinstall the CKB, tried front and back jacks, sound input settings, new motherboard, etc.

Anyone has any clues?

Thanks.

tatokis, 2019-11-03:
https://github.com/ckb-next/ckb-next/wiki/Animations#on-linux-with-pulseaudio

yuri2203, 2019-11-03:
> https://github.com/ckb-next/ckb-next/wiki/Animations#on-linux-with-pulseaudio

Accurate! thanks <3

closed 2019-11-03 by yuri2203