ckb-next/ckb-next #500: Daemon not running

CoroIa, 2019-11-04:
First i wanted to say i am new to all of this but i have studied up for quite some time.. that being said i installed ubuntu on my chromebook and attempted to install ckb-next for ubuntu 16. the program is installed and opens up but i can only change Tab, Shift, etc. keybinds and can access nothing else this build was intended to deliver.

when i launch ckb i recieve the message:

The ckb-next daemon is not running. This program will not work without it!

Start it once with:
sudo systemctl start ckb-next-daemon
Enable it for every boot:
sudo systemctl enable ckb-next-daemon

i tried both in my terminal but i receive the message: Running in Chroot, Ignoring Request

any tips on where i might have went wrong when building?

OvermindDL1, 2019-11-04:
> i tried both in my terminal but i receive the message: Running in Chroot, Ignoring Request

You need to run on the main system root, not a chroot.  What system is this anyway I'm curious?

CoroIa, 2019-11-04:
im unsure the difference of the two as i said i am new so i do appologise for my ignorance.. that said i believe the answer to your question is, that im on an acer c720-2844 chromebook in developer mode with ubuntu installed on the system. as for which ubuntu i have i do not know how to check that..

i tried installing ubuntu 14 ckb with no luck, after uninstalling everything (as far as im aware i uninstalled everything dealing with ckb ubuntu 14) i then proceeded with ubuntu 16 download.. upon completion i am able to open ckb-next but can only map shift, tab, super, etc.. 

i hope this was the information you were inquiring about for further trouble shooting.. 

note that the main reason for me needing the software is literally just to change the polling rate of one profile from 1000hz to 125hz so i can plug and play native mnk on modern warfare on console as i am experiencing extreme pixel skipping and freeze framing with my look/aim mechanic

tatokis, 2019-11-04:
Chromebooks aren't supported. Ubuntu is running inside a chroot in the chromebook, which means it's not directly accessing anything in the chromebook.

You can try to manually start the daemon in a terminal by running `sudo ckb-next-daemon` but I doubt it'll work.

closed 2019-11-04 by tatokis

CoroIa, 2019-11-04:
(xenial)corola@localhost:~$ sudo ckb-next-daemon
[sudo] password for corola: 
ckb-next: Corsair RGB driver 0.4.2-unknown
[I] Root controller ready at /dev/input/ckb0

CoroIa, 2019-11-04:
this is what came up can you help me make sense of this? im really interested in this kind of stuff and would like to become more lingual in it

CoroIa, 2019-11-04:
i did this and i am in.. i can see my mouse and change the lighting along with the polling rate and many other things now

CoroIa, 2019-11-04:
where it said the daemon wasnt running it now says unknown but i have all the customization options for mouse and keyboard along with polling rate

OvermindDL1, 2019-11-04:
That works more than I would actually expect it to for a chromebook then.  ^.^;

You can build a user profile setup of systemd startup for it if you want, but you might need to google for that, I'm unsure what other oddities chromeos might throw that way.

CoroIa, 2019-11-05:
i would like to.. im just very new to this stuff and have no idea on how i would get started and if there is anyone who would be in my predicament.. the program is can run keyboard animation colors for my K55 but can NOT save them on the hardware, the M65 Pro CAN though, i can change polling rate every rgb led profiles configs mappings theres even a software update to 3.06 i can install AND SAVE!!! It"s a little exciting for me lol, but again it needs work otherwise its only feasible as a mouse software

OvermindDL1, 2019-11-05:
> what if this could be the first possible corsair software for chromebook..

I'm still surprised at how well it works on a chromebook as-is, it's rather awesome, lol.  ^.^

CoroIa, 2019-11-05:
lol same, now.. i have an embarrassing question.. when you say build a user profile set up for systemd startup. isnt that what i just did? or can you explain it to make a little more sense to me?
im a noob -_-

OvermindDL1, 2019-11-05:
Oh you know how systemd has startup script-like-things to load things in the right order on the system like the old sysv setups?  It also has that capability on individual user profiles as well.  If you can get access to the root area then you could do it there, but as a last resort you can always manually build one in your own user.  Or just add it to a startup script if not already running.  ^.^

CoroIa, 2019-11-05:
oh i see! that makes sense, and thank you btw 

do you think development of this program even further would be fruitful or of any use?

OvermindDL1, 2019-11-05:
> do you think development of this program even further would be fruitful or of any use?

Always, development is always wanted and encouraged!

CoroIa, 2019-11-06:
okieee.. i was wondering though why its only in demo mode for keyboard but for the mouse i can actually save to hardware?

even the keyboard light sequences can be put on the mouse and to be fair they should stay as keyboard lighting sequences.. lol the only one that is pleasing to the eye on the mouse is the "ripple" animation, where it breathes two colors on the corsair gaming mouse LED logo on the back of the mouse where the palm rests.