ckb-next/ckb-next #501: It works on Chromebook Saved to Hardware mouse and keyboard

CoroIa, 2019-11-04:
i was told this software for ubuntu would NOT work on a chromebook but low and behold it DOES i have the software pulled up, able to change dpi, all button mappings, all led lights, etc. 

i don't mean to sound disrespectful but my support request was closed with very minimal information on my problem..

thankfully tato DID give me the key to running this software on my chromebook and that is command
sudo ckb-next-daemon

by doing this the daemon became unknown instead of not running and seems to have pushed me through to the advanced settings in the hardware. for those experiencing my issue, it can be solved by copy pasting the sudo command above. just save all changes to hardware un plug and plug in desired device for use you will see your changes have stuck!!

ZirconiumX, 2020-05-14:
The K55 doesn't actually have any memory to save to, so save to hardware is disabled on it.

closed 2020-05-14 by ZirconiumX

ReViDiSwO, 2020-08-26:
hi, I've got a chromebook and when i try the command : "sudo ckb-next-daemon"

I get this message : 
"ckb-next: Corsair RGB driver 0.4.2-87-g627ccd8
[E] _mkdevpath (devnode.c:214): Unable to create /dev/input/ckb0: No such file or directory"

I follow this instruction for the installation : "https://github.com/ckb-next/ckb-next/wiki/Linux-Installation"
I use this method in particular (my linux version is Debian 10) : 
"Debian/Ubuntu (18.04+): sudo apt install build-essential cmake libudev-dev qt5-default zlib1g-dev libpulse-dev libquazip5-dev libqt5x11extras5-dev libxcb-screensaver0-dev libxcb-ewmh-dev libxcb1-dev qttools5-dev git libdbusmenu-qt5-dev

git clone https://github.com/ckb-next/ckb-next.git

cd ckb-next

./quickinstall"

Any idea how to fix this problem ?