ckb-next/ckb-next #502: No back-lighting on Windows/Super lock key. K70 Rapidfire on Linux Ubuntu

justJackjon, 2019-11-09:
Hi team, hoping you can help!

## General information
I noticed there was a similar ticket but it was already closed ( ticket #220 ) and unfortunately there was no fix I could try there.

I am running Ubuntu 18.04.3 LTS on a Dell XPS 13 7390. The K70 Rapidfire is connected via a Dell WD19TB Thunderbolt docking station.

### Source
Installed via: https://launchpad.net/~tatokis/+archive/ubuntu/ckb-next

## What is the program's current behaviour?
The super/windows lock key (above scroll lock on the K70) retains its functionality (it still prevents usage of the super key/windows key whilst active) but its backlighting does not activate, it is set to permanently off.

I know in the performance tab behaviour can be modified, but the checkbox is selected to indicate the windows lock and I have tried setting the colour to both white or red in the 'on' dialogue box. (My version of the K70 is red backlighting only).

## What is the expected behaviour of the program?
When activating and deactivating the lock key the backlighting should toggle accordingly.

Please let me know if you need further info!

Thanks in advance for any help you can offer :)

berkson, 2021-06-20:
The same occurs in my K55 with Ubuntu 20.04 Acer Aspire 5.

Tibladar, 2022-05-06:
Still an issue with K55.

Is there an option to enable/disable the lock LED in iCUE?
If yes, it would be possible to monitor the usb data.

I haven't found an option in iCUE yet, only to change the action.

gravityfargo, 2024-05-02:
This issue has been marked as stale due to age and/or lack of activity.  No report was supplied.

Please feel free to reopen it if you have more to add or if the issue still needs attention. However, it would be best to [open a new issue](https://github.com/ckb-next/ckb-next/issues/new/choose) and provide a newly generated report issue using our updated templates.

closed 2024-05-02 by gravityfargo