ckb-next/ckb-next #505: K57 RGB Wireless Keyboard

Brainwasher06, 2019-11-13:
Hello,

Is it possible to add my keyboard in your software ? If you do it would be much appreciated =)

- If possible, try the latest code by building the software manually. ( i used ./quickinstall , dont know if it's what you mean)

Thank you for your work and your time.

Have a good day
---

Enter the name of your device here and attach the report generated above :
K57 RGB Wireless Keyboard
[ckb-next-report-1573647243.gz](https://github.com/ckb-next/ckb-next/files/3841018/ckb-next-report-1573647243.gz)

tatokis, 2019-11-13:
It uses a completely new and redesigned protocol that we haven't looked into, and probably will not for a while. No one has shown interest in implementing this protocol.

oppositionsecurity, 2019-11-13:
Just grabbed this keyboard and would be interested as well.

87racer, 2019-11-13:
Just grabbed this keyboard and would be interested as well.

MarcinOrlowski, 2019-11-16:
You guys may want to write to Corsair's support pointing this ticket. Maybe they appear smart and provide you all required docs. Win-Win.

kaijuagenda, 2020-01-16:
Recently bought one myself. shame there's no support yet.

oppositionsecurity, 2020-01-16:
Recently returned 2 K57's actually. If you type 70+ WPM you will be sorely disappointed with the "selective n-key rollover". Very common 3 and 4 letter combinations will fail to type in rapid succession. As a side note the K95 is great, despite the cost, and is fully supported and allows for some cool effects using the top LEDs as notifications lights.

kaijuagenda, 2020-01-26:
> Recently returned 2 K57's actually. If you type 70+ WPM you will be sorely disappointed with the "selective n-key rollover". Very common 3 and 4 letter combinations will fail to type in rapid [...]

Not sure how well the measure holds up. I frequently break 90+ WPM on the K57 with no apparent issues, Nor can i replicate the problem you described. Either way I don't see the point of this response, unless the goal was to shill a more expensive peripheral.

superjcvd, 2020-06-09:
I am interested too :)
So far the keyboard works fine in bluetooth.
- the sound control keys are working
- the RGB isn't working
- the macro keys are not working

ThisNekoGuy, 2020-06-09:
I just got mine in the mail today, myself, it feels pretty great Lol
But yeah, it's unfortunate that there isn't support for it yet

ethergary, 2020-08-16:
I'm also interested, most functionality works straight from the keyboard on the K57 out of the box for Linux, but it would be nice to be able to have full control over RGB and the macro keys like in this (very nice) software port for the other devices.  I noticed that the K55 is supported which, other than wireless, appears virtually identical to the K57 at a glance as far as features and how it works (over USB at least, the wireless abilities not withstanding).  I seem to figure with it plugged into USB should it communicate the same as the K55?  And/or is it possible for testing purposes to 'fake' the K55 feature set in the ckb-next software to point to the K57 keyboard while plugged into USB within the software to test?  I'm currently using Ubuntu 20.04 (POP OS style) which recognizes the keyboard just fine otherwise, but naturally not able to read as connected via ckb-next.  I'm happy to go in and give it a go in the code, but some bread crumbs to point the way would be helpful.  If it's my problem to figure out, no worries - I just like the keyboard and this software.  And not particularly gifted at hardware driver programming.

erdoking, 2020-08-24:
HID report:

```
Bus 001 Device 008: ID 1b1c:1b6e Corsair USB 2.0 Hub

usbhid-dump -d 1b1c:1b6e
001:008:002:DESCRIPTOR         1598253662.528718
 06 42 FF 09 02 A1 01 15 00 26 FF 00 75 08 95 40
 09 02 81 02 C0

001:008:001:DESCRIPTOR         1598253662.529510
 06 42 FF 09 01 A1 01 15 00 26 FF 00 75 08 95 40
 09 01 81 02 95 40 09 01 91 02 C0

001:008:000:DESCRIPTOR         1598253662.531037
 05 01 09 06 A1 01 85 01 05 08 95 03 75 01 19 01
 29 03 15 00 25 01 91 02 95 05 91 03 05 07 75 01
 19 E0 29 E7 95 08 81 02 19 00 29 67 95 68 81 02
 19 87 29 8E 95 08 81 02 C0 05 0C 09 01 A1 01 85
 02 95 01 75 10 15 00 26 FF 03 19 00 2A FF 03 81
 00 C0

```
```
Bus 001 Device 011: ID 1b1c:1b62 Corsair CORSAIR K57 RGB Wireless Dongle

001:011:003:DESCRIPTOR         1598253890.755936
 05 0C 09 01 A1 01 85 02 95 01 75 10 15 00 26 FF
 03 19 00 2A FF 03 81 00 C0

001:011:002:DESCRIPTOR         1598253890.756411
 06 42 FF 09 02 A1 01 15 00 26 FF 00 75 08 95 40
 09 02 81 02 C0

001:011:001:DESCRIPTOR         1598253890.756867
 06 42 FF 09 01 A1 01 15 00 26 FF 00 75 08 95 40
 09 01 81 02 95 40 09 01 91 02 C0

001:011:000:DESCRIPTOR         1598253890.757430
 05 01 09 06 A1 01 85 01 05 08 95 03 75 01 19 01
 29 03 15 00 25 01 91 02 95 05 91 03 05 07 75 01
 19 E0 29 E7 95 08 81 02 19 00 29 67 95 68 81 02
 19 87 29 8E 95 08 81 02 C0
```

I need to install the corsair windows software first ;) havn't start windows 10 for month ...

erdoking, 2020-08-24:
[usb_corsair_k57.zip](https://github.com/ckb-next/ckb-next/files/5117388/usb_corsair_k57.zip)

I have created the captures - thank you very much for your engagement :)

1.) disconnected all USB devices
2.) stop iCUE
3.) start wireshark
3.) plugin keyboard (wired OR wireless)
4.) start iCUE
5.) activate some effects
6.) Repeat

On wireless we cant do the same things like on wired! maybe different protocols?
On wireless we cant program each key color different for example...

tatokis, 2020-09-11:
A bit late, but excellent.
It helped me reverse quite a bit of the protocol. You can load [bragishark](https://github.com/ckb-next/CueShark) in wireshark and have a look if you wish. 

Before I add the device to ckb, can you please make another capture, both with CUE off and on, simply pressing some keys and noting down which keys they are.
For example: Close CUE, start capture, type press "qwerty", stop capture.
And then also repeat with CUE running.

tatokis, 2020-09-11:
Oh, and also, what layout is your keyboard? @erdoking  (European/American/etc)

erdoking, 2020-09-11:
I updated to iCUE 3.33.246
My keyboard layout is European (qwertz)

I pressed the keys: 
qwertz left up right botton and G1-G6
(G6 open the windows menu)
---
At the moment i have no idea how to load bragishark  ... but i found it out ;)

bvcelari, 2020-10-08:
I got one of this Keyboards too, Is there anyway I can help? it's a pitty I cannot use ckb-next

magni08, 2020-10-12:
Also interested and willing to assist. 

Using qwerty layout on a manjaro distro.

DrajonsAreEpic, 2020-11-14:
I have a K57 and would love to be able to use use ckb-next. Willing to help, but don't have any practical coding abilities. Using a QWERTY layout on Pop!OS.

pmsobrado, 2020-11-24:
Any updates on this? I just got a K57 myself and I'm willing to help by any means possible. I have Windows and Ubuntu 18.04.

nmartin-cs, 2021-02-03:
Found this project after picking up a K57. CKB daemon running is systemd (Arch). Got most everything working using udev; few things I'd like to fix. Personally I could care less about the led lights ;) 
@tatokis I'd like to take a look at this protocol. You mentioned it is different than the earlier models? Can you point me at it? Sorry if you mentioned this before TL;TD

NeleBiH, 2021-07-18:
anyone figured how to see battery level in corsair wireless devices?

erdoking, 2021-10-11:
Have a look at this commit. I think tatokis is working hard for us ;)

https://github.com/ckb-next/ckb-next/commit/c45878d6edacb2da2b5e6a66ae9e5d36e14924f6

NeleBiH, 2021-12-17:
Small workaround for crontroling rgb functions on k57 using keys check manual page 3 and 4
[DARK_CORE_RGB_QSG.pdf](https://github.com/ckb-next/ckb-next/files/7737088/DARK_CORE_RGB_QSG.pdf)

tatokis, 2022-04-04:
I just added the K57 to the GUI. You need to have the latest code from git and run ckb-next-daemon with `--enable-experimental`.
The lock lights are completely broken in wireless mode, and you probably don't want to use RGB animations wirelessly (they are very laggy).
There's also a problem with adjusting the brightness. You need to make sure that the brightness on the keyboard is set to maximum before running the ckb-next GUI.

NeleBiH, 2022-04-05:
i found altenative solution using fn 1-9 keys to change everything and it works awesome.Thing missing is no possibility to see baterry level.If you need anything that can help you send me info how because i own dark core,k57,virtuoso,h100i watercooling

docsmooth, 2022-04-19:
@tatokis thanks, this first commit does recognize my keyboard on 5.13.19 and allow me to set some values, but I didn't play with it much.  I have my K57 *and* a RGB Dark Core Pro mouse, and enabling running "systemctl start ckb-next-daemon" (which I added  --enable-experimental to the systemd configuration" and then launching the GUI... disconnected my mouse entirely.  the CKB gui showed only 1 device connected.

I've been using iCUE in a Windows VM to configure RGB settings for now. I'd love to provide any details needed to help add dual-device support back.  (Is wireshark +Windows how everyone has been gathering the required usb tracing?)

FWIW, stopping ckb-next allowed the mouse to reconnect w/out having to reconfigure anything with iCUE.

tatokis, 2022-04-22:
@docsmooth Which dark core? Can you generate and attach a report with your mouse connected?

NeleBiH, 2022-04-24:
https://www.mediafire.com/file/cqbnsn6ddtwytus/ckb-next-report-1650737969.gz/file here is my file if you find anything usefull
In this you have corsair comander,h115 cooling aio,k57 wireless,dark core wireless and virtuoso

docsmooth, 2022-04-26:
[ckb-next-report-1650991548.gz](https://github.com/ckb-next/ckb-next/files/8565224/ckb-next-report-1650991548.gz)

# lsusb -d1b1c:
Bus 001 Device 003: ID 1b1c:1b62 Corsair CORSAIR K57 RGB Wireless Dongle
# /usr/local/bin/ckb-next-daemon --enable-experimental
ckb-next: Corsair RGB driver 0.4.4-198-gbe1d33b
[I] Root controller ready at /dev/input/ckb0
[I] Connecting CORSAIR K57 RGB Wireless Dongle at /dev/input/ckb1
[I] Starting input thread for /dev/input/ckb1
[E] os_usb_interrupt_out (via bragi_common.c:9): No such file or directory, res = 0xffffffff
[F] start_bragi_common (device_bragi.c:97): ckb1: Couldn't get bragi device mode. Aborting
[E] os_usb_interrupt_out (via bragi_common.c:23): No such file or directory, res = 0xffffffff
[I] Found EP 0x82 at /sys/devices/pci0000:00/0000:00:14.0/usb1/1-2/1-2:1.0/ep_82
[I] Found EP 0x84 at /sys/devices/pci0000:00/0000:00:14.0/usb1/1-2/1-2:1.1/ep_84
[I] Found EP 0x83 at /sys/devices/pci0000:00/0000:00:14.0/usb1/1-2/1-2:1.2/ep_83
[I] Found EP 0x81 at /sys/devices/pci0000:00/0000:00:14.0/usb1/1-2/1-2:1.3/ep_81
[I] Found EP 0x85 at /sys/devices/pci0000:00/0000:00:14.0/usb1/1-2/1-2:1.4/ep_85
[I] Found EP 0x86 at /sys/devices/pci0000:00/0000:00:14.0/usb1/1-2/1-2:1.5/ep_86
[E] os_usb_interrupt_out (via bragi_common.c:9): No such file or directory, res = 0xffffffff
[I] Found new bragi subdevice 1
[E] os_usb_interrupt_out (via bragi_common.c:9): No such file or directory, res = 0xffffffff
[E] os_usb_interrupt_out (via bragi_common.c:9): No such file or directory, res = 0xffffffff
[I] ckb2: Subkb vendor: 0xffff, product: 0xffff
[I] Disconnecting /dev/input/ckb2
[I] Removed device path /dev/input/ckb2
[I] Found new bragi subdevice 2
[E] os_usb_interrupt_out (via bragi_common.c:9): No such file or directory, res = 0xffffffff
[E] os_usb_interrupt_out (via bragi_common.c:9): No such file or directory, res = 0xffffffff
[I] ckb2: Subkb vendor: 0xffff, product: 0xffff
[I] Disconnecting /dev/input/ckb2
[I] Removed device path /dev/input/ckb2
[I] Found new bragi subdevice 3
[E] os_usb_interrupt_out (via bragi_common.c:9): No such file or directory, res = 0xffffffff
[E] os_usb_interrupt_out (via bragi_common.c:9): No such file or directory, res = 0xffffffff
[I] ckb2: Subkb vendor: 0xffff, product: 0xffff
[I] Disconnecting /dev/input/ckb2
[I] Removed device path /dev/input/ckb2
... (lots of finding/removing devices with that same error, then I move the mouse and get:)
[E] process_input_urb (keymap.c:731): Unknown bragi data received in input thread 00 from endpoint 82
[E] process_input_urb (keymap.c:731): Unknown bragi data received in input thread 00 from endpoint 82
[E] process_input_urb (keymap.c:731): Unknown bragi data received in input thread 00 from endpoint 82
[E] process_input_urb (keymap.c:731): Unknown bragi data received in input thread 00 from endpoint 82
[E] process_input_urb (keymap.c:731): Unknown bragi data received in input thread 00 from endpoint 82
[E] process_input_urb (keymap.c:731): Unknown bragi data received in input thread 00 from endpoint 82
[E] process_input_urb (keymap.c:731): Unknown bragi data received in input thread 00 from endpoint 82
[E] process_input_urb (keymap.c:731): Unknown bragi data received in input thread 00 from endpoint 82
[E] process_input_urb (keymap.c:731): Unknown bragi data received in input thread 00 from endpoint 82

docsmooth, 2022-04-26:
[ckb-log.txt](https://github.com/ckb-next/ckb-next/files/8565356/ckb-log.txt)

Here's the ckb-next --enable-experimental log from ONLY having the Dark Core RGB Pro SE plugged in via USB (through my docking station).  When ckb-next was killed, the mouse stopped working entirely, FWIW.

`# lsusb -d1b1c:
Bus 003 Device 010: ID 1b1c:1b7e Corsair CORSAIR DARK CORE RGB PRO SE Gaming Mouse

`

DenizUgur, 2022-07-30:
Has there been any update on this?

nlynzaad, 2022-12-29:
Thinking of purchasing a k57. Based on the above can someone confirm if the macro keys are working under the experimental flag? Not to bothered with the rgb.

ghostdevv, 2022-12-29:
@nlynzaad Works great for me on arch with macros and rgb with the flag

John-Gee, 2023-06-05:
I'm on Arch and enabled experimental but it still doesn't find the keyboard. Do I need to connect it by BT or with a wire instead of 2.4 to connect it?