ckb-next/ckb-next #509: Add ability to create pipe files in location other than /tmp

bennybenjaman, 2019-11-21:
## What is the program's current behaviour?
CKB has the pipe files hardcoded to use /tmp for the installation directory

## What would you like it to do instead?
I use multiple users to run applications in my  environment, for example I have a dedicated user account for Firefox. When I run firefox in KDE using kde su it is under another user which is not in the same group, so their home folder are not connected and permissions do not match.

eg: 

My CKB user is anon, running KDE and KDE SU to run applications under different users.
My firefox user is firefox.
Each users home directory permissions are 700
CKB is installed under my user account anon
If I create a pipe using CKB it creates the pipe file with permissions for anon
If I create a script to execute when my firefox directory is backed up to a remote directory I have to change the ckbpipe000 files permissions to allow for firefox to rw the file. Since the file is created in /tmp it is recreated  on each restart and the files permissions need to be modified again for the animation to properly execute.

I understand that ckb can only creates pipe files for the user running the application, so perhaps if there was a way to select where each ckbpipe file is created that would help prevent any modified permissions from being over written on restart in /tmp
## Any other notes & comments?

ZirconiumX, 2019-11-21:
The bot should really check the tags before closing issues, @tatokis.

tatokis, 2019-11-21:
It does. It's just that mistakes happen and I thought new-device was enhancement too.

```python3
# Check if it's one of the issues that require a dev-detect
if len(event.data["issue"]["labels"]) == 0:
    return
```