ckb-next/ckb-next #511: Problem with xorg configuration

FelipeMarcelino, 2019-12-06:
I have two keyboards and two configuration Xorg files. The first config file covers my HHKB Pro 2:

00-keyboard-system.conf:

Section "InputClass"
        Identifier "system-keyboard"
        Option "XkbLayout" "us"
        Option "XkbVariant" "intl"
	MatchUSBID  "0853:0100"
EndSection

And the second one is for my k95 platinum:

10-keyboard-corsair.conf:

Section "InputClass"
        Identifier "Corsair Gaming K95"
	MatchUSBID  "1b1c:1b2d"
        Option "XkbLayout" "br"
EndSection

When my pc doesn't have ckb-next, the layout works fine. However, after the installation and execution of the daemon process, the second layout match by USB ID change from br to us. 

The behavior expected is that the layout continues in br. Maybe the daemon is executing before Xorg?

penguineer, 2020-03-31:
ckb has it's own laoyut setting in the profile. Maybe that's overriding your xorg setting?

gravityfargo, 2024-05-02:
This issue has been marked as stale due to age and/or lack of activity. No report was supplied.

Please feel free to reopen it if you have more to add or if the issue still needs attention. However, it would be best to [open a new issue](https://github.com/ckb-next/ckb-next/issues/new/choose) and provide a newly generated report issue using our updated templates.

closed 2024-05-02 by gravityfargo