ckb-next/ckb-next #513: Provide for Windows too

tepperly, 2019-12-09:
It would be cool if this was also available for Windows because iCUE is that bad.

ZirconiumX, 2019-12-09:
We have a Windows branch for that, but it's very experimental.

rwilliams, 2019-12-23:
This would be great. I'm trying to avoid icue at all costs.

rwilliams, 2019-12-23:
@ZirconiumX Do we just follow the same build directions using the windows branch to test?

tatokis, 2019-12-26:
Unfortunately no. And it's highly unstable.

You will need msys2 with mingw-w64 in it.
You'll have to install a full dev env in it, Qt, possibly other dependencies, and then manually install libusbk and edit the cmake file to point to the right path.

You should be able to open the project in Qt Creator from within msys2

Once you compile the daemon and the gui, you will need to manually use zadig to install the libusbk kernel driver for all endpoints on your device.

Then you will need to install a signed vmulti driver from https://github.com/ecologylab/EcoTUIODriver

If I am not mistaken, we're using "vmultia", not the other ones.

Do note, while the daemon isn't running, your devices will be completely dead. If you don't have a hardware bios switch, then you can't switch to bios mode while in this state either.

I haven't worked on it in a while, so I can't give more detailed information about it.

rwilliams, 2019-12-26:
@tatokis That seems like way too much work. For giggles I tried it on WSL2 with Ubuntu 19.10 and I could get the app to run but the service won't run even when started manually. Maybe after some future updates to WSL2 it'll just work.

Rafagd, 2020-01-20:
Is it "nothing works anymore and never will" bad or does it work to some extent?
I'm currently using iCUE and it just randomly decides to not work for not reason at all anyway...

rwilliams, 2020-01-21:
I'd be willing to put some effort into this but I'd need a good bit of guidance. I can't use icue because the hardware monitoring causes all Ryzen cores to wake and causes well above average core voltages.

spartan117au, 2020-01-26:
It would be fantastic to have a Windows version. iCue is pretty much entirely busted.

maffe, 2022-07-21:
Duplicate of https://github.com/ckb-next/ckb-next/issues/301

closed 2022-07-23 by tatokis