ckb-next/ckb-next #514: K95 RGB Media Keys Not Working

AustinBachurski, 2019-12-17:
Fair warning, I'm new to Linux so sorry for my ignorance.  Using ckb-next with my K95 RGB, everything seems to work except for the media keys- play/pause, stop, skip, reverse.  Tried changing the key binding to just a keystroke and that part works, installed playerctl and the terminal commands control my media player, but I can't get the media keys on the keyboard to do anything.  Not sure what to do.  Thanks.

tatokis, 2019-12-17:
Media keys are handled by your desktop environment and have nothing to do with ckb. Either set up your environment so that it falls playerctl accordingly, or make binds in ckb for it with the execute program on key press option.

AustinBachurski, 2019-12-18:
Ahh ok, thanks very much!

closed 2019-12-18 by AustinBachurski