ckb-next/ckb-next #515: systemd daemon takes 30 to 40 seconds to stop

pakro, 2019-12-17:
# General information
I'm running Arch Linux with XFCE.
I'm using a Corsair K95 RGB keyboard (the old one with more G buttons) Corsair Sabre mouse and the Corsair MM800 mousepad. All of them are detected properly and thanks to this software I can even use the built in USB port on the mousepad now (didn't work without this software!). 

However there is one major annoyance. Since I have installed the software, reboots and shutdowns take quite a time because the systemd daemon takes ages to stop.

### Source
AUR running ckb-next 0.4.2-3

## Logs & crash reports

No logs since it is just an observation I've made on a freshly installed Arch. If I can provide some, just say something!

## What is the program's current behaviour?
Systemd wind down takes long because corsair keyboard and mice daemon (not sure if it is 100% correctly spelled) needs around 30-40 seconds to shutdown.

## What is the expected behaviour of the program?
The daemon winds down faster.

## Any other notes & comments?
This issue sounds a lot like https://github.com/ckb-next/ckb-next/issues/168 but it was closed back then.

tatokis, 2019-12-26:
Have you tried the quirks?
https://github.com/ckb-next/ckb-next/wiki/Troubleshooting#usbcore-quirks
https://github.com/ckb-next/ckb-next/wiki/Troubleshooting#usbhid-quirks

pakro, 2019-12-30:
Thank you for the response. Unfortunately the USBHID quirk won't do it on my setup as my disk is encrypted. 
I'm not sure if the USBCore quirk can be applied to a encrypted disk setup. Can it? It is not explicitly mentioned in the article but both quirks look very similar.

tatokis, 2019-12-30:
Both can work on an encrypted disk. Neither disable the built in kernel driver.

zfr77, 2020-01-14:
I've got same issue with the same keyboard (K95 RGB).
Adding quirks didn't help.

drwhut, 2020-04-03:
I had the same problem on Ubuntu Server 18.04, but only with the `ckb-next` PPA. I've since switched to using the `ckb-next-git` PPA (the one based on the master branch), and the issue has stopped, i.e. I don't have to wait 30-40 seconds for the daemon to stop.

pakro, 2020-04-07:
> I had the same problem on Ubuntu Server 18.04, but only with the `ckb-next` PPA. I've since switched to using the `ckb-next-git` PPA (the one based on the master branch), and the issue has stopped, i.e. I don't have to wait 30-40 seconds for the daemon to stop.

Hi! Thank you for the hint! I switched to ckb-next-git. Hopefully it also improves my stop times.

pakro, 2020-04-22:
> Hi! Thank you for the hint! I switched to ckb-next-git. Hopefully it also improves my stop times.

Unfortunately the switch did not help at first. However there was an update some days ago and I now don't have to wait that long for reboots. However I believe it broke the suspend mechanism. I've created the [issue](https://github.com/ckb-next/ckb-next/issues/515#issue-539251205).

atm0915, 2020-04-26:
I am also having this problem. Neither the git package or the quirks solved the problem for me.

ashtonx, 2020-05-25:
Try changing 0x20000408 to 0x4 on usbhid quirk, this one helped me. It's mentiond in the wiki page but it's easy to miss.

ATM-Tech, 2020-05-25:
I can confirm that the change to 0x4 fixed the problem for me.

pakro, 2020-05-26:
I'm using full disk encryption therefore 0x4 is no option for me.

BeCracked, 2020-11-23:
I'm still having a similar problem. Running Manjaro 5.8.18-1-MANJARO with KDE and a I think the same Corsair K95 RGB as OP and the stop jobs takes about 30 to 40 seconds.
I will try the quirks mentioned and see if that helps.

pakro, 2020-11-25:
Hi @BeCracked. If you are running Manjaro, do you use the ckb-next package or the ckb-next-git package? You can use  `pacman -Q ckb-next` to find out which one you are running. For me the problem is far better since I'm using the ckb-next-git package. I'm still having to wait a little longer sometimes but most of the times it is shutting down really quick. Maybe you can try that if quirks don't work for you because your setup does not allow it (e. g. disk encryption).

gravityfargo, 2024-05-02:
This issue has been marked as stale due to age and/or lack of activity. 

Please feel free to reopen it if you have more to add or if the issue still needs attention. However, it would be best to [open a new issue](https://github.com/ckb-next/ckb-next/issues/new/choose) and provide a newly generated report issue using our updated templates.