ckb-next/ckb-next #516: System Extension is not allowed; no prompt and no option in Mac Catalina

Ketenks, 2019-12-20:
[ckb-next-report-1576826540.gz](https://github.com/ckb-next/ckb-next/files/3987186/ckb-next-report-1576826540.gz)
- Search in open and closed [Issues](https://github.com/ckb-next/ckb-next/issues).
- If possible, try the latest code by building the software manually.
- File unrelated issues separately.

## General information

### Source
How and where did you get this program, e.g.:
  * Pre-built dmg for macOS

## Logs & crash reports

Attach:
report is attached

## What is the program's current behaviour?
devices are not recognized
## What is the expected behaviour of the program?

## Any other notes & comments?

tatokis, 2019-12-26:
Make sure `ckb-next-daemon` has input monitoring, not `ckb-next`

Ketenks, 2019-12-29:
I haven't been able to find it in the Security and Privacy options. That was all I got when I went to give those permissions. Am I supposed to 'Show Package Contents' in order to find it? I would figure there would be a prompt for it at least and in my case there wasn't.

During the installation it had permission problems that the 'Install.app' had authority over and it could not finish the installation normally but it went through it anyway and then left me with the two messages: "System Extension is not allowed" and "CKB-Next-Daemon needs input monitoring in Security and Privacy" but when I went there to give it those permissions neither of those programs were options.

jaspermutsaerts, 2020-08-18:
Today I got the same problem after upgrading to Catalina. But I found a workaround!
Couldn't add the daemon to Input Monitoring because when there are no apps in the list, it's not possible to manually add them. Only apps can add the first one.
So I decided to write a Swift simple command line app that does nothing else than trigger the dialog and itself to the list so there is an entry. For those without Xcode I converted it to an actual app. 

[Download Empty Input Monitoring Entry app ](https://github.com/ckb-next/ckb-next/files/5091404/Empty.Input.Monitoring.Entry.zip)

**Screenshots**
[image]
[image][image]
[image]
[image]
[image]

**Used Code**
Used code based on https://developer.apple.com/videos/play/wwdc2019/701/ 
I didn't actually see the video throughout, but considered this code shown harmless.

```swift
import Foundation

func callback(
    proxy: CGEventTapProxy,
    type: CGEventType,
    event: CGEvent,
    userInfo: UnsafeMutableRawPointer?
) ->Unmanaged<CGEvent>? {
    // don't really do anything
    return nil
}

// Register to listen to tapEvents: this will trigger an  input monitoring dialog
let eventMask:CGEventMask = 1 << CGEventType.keyDown.rawValue
let eventTap = CGEvent.tapCreate(
    tap: .cghidEventTap,
    place: .tailAppendEventTap,
    options: .listenOnly,
    eventsOfInterest: CGEventMask(eventMask),
    callback: callback,
    userInfo: nil
)
```

**Disclamer**:
Use at own risk. It did resolve this exact problem for me. Of course it's still just a workaround.

closed 2022-06-27 by tatokis