ckb-next/ckb-next #518: Macro's not working Ubunt 19.10

davidbdeath, 2019-12-23:
I tried this as a macro mapped to a button on my scimitar pro rgb and it does nothing. When i map in Mac OS it works fine.

+lctrl, +lbrace, -lbrace, -lctrl,

tatokis, 2019-12-26:
There's nothing bound to `Ctrl [` on Ubuntu.

Linux has actual support for back and forward buttons on mice. Just bind the buttons to 
Mouse -> Special Button -> Backward/Forward

closed 2019-12-26 by tatokis

davidbdeath, 2019-12-26:
1. you can create nearly any keyboard shortcut in Ubuntu.
2. Ctrl + [ is one of Fire Fox's default short cuts.

https://support.mozilla.org/en-US/kb/keyboard-shortcuts-perform-firefox-tasks-quickly?redirectlocale=en-US&as=u&redirectslug=Keyboard+shortcuts&utm_source=inproduct