ckb-next/ckb-next #519: Macro's not working Ubunt 19.10

davidbdeath, 2019-12-26:
Okay lets try this again and hopefully i don't get the same rude unintelligent responce.

I am unable to create a macro using any keyboard buttons on my SCIMITAR PRO RGB. Only mouse inputs are detected and manual input is also ignored. I posted an example last time but it was used as a an excuse to invalidate my issue. I don't have this problem in Mac OS High Sierra.

I'm using version 0.4.2

joe-mojo, 2019-12-28:
This is not abug but more a lack of documentation.
I would like a detailed doc about macro syntax :
 - what is the "macro name" of each key (lalt, ralt, lctrl and rctrl are in the daemon doc, but other keys ???)
 - what is the "macro sequence" text are used for ? If I type in this area everyting is cleared on "apply"
etc.

davidbdeath, 2019-12-29:
No, it's a bug. I use this software successfully on Mac OS. CKB is ignoring keyboard inputs when recording a macro for my scimitar. It will pick up mouse inputs. I can delete the mouse inputs input keys manually, for example "+lctrl, +lbrace, -lbrace, -lctrl" It doesn't work. If I change the delay with one of the radial buttons the mouse inputs that were recorded will come back and the manually inputed keys are erased. Even if start with manually entering the keys it doesn't work. I am getting the key names from the keymap.c file.

davidbdeath, 2019-12-29:
The UI/UX for creating a macro is not good.

tdis8629, 2019-12-29:
For the prior issue, you listed specifically `+lctrl,+lbrace,-lbrace,-lctrl`, stating that it works in MacOS, but not Ubuntu (Example given was Firefox, but that was after the issue was already closed). I do not have Ubuntu 19.10 (Arch and Manjaro), but I was able to type out the sequence, apply it to Button 12, and it went back a page in Firefox. As the listed combination is usually used specifically in MacOS for Forward/Back, and *nix/Windows OS's can read Mouse Button 4/5, tatokis *did* technically solve your issue with no further context given.

Issue #10 also shows the issue with translating keyboard <=> mouse macros, and I am interpreting it as the macros can only poll that device when recording. As I do not have a non-Corsair keyboard, I am unable to confirm it, though.

davidbdeath, 2019-12-29:
He made an assumption about what my problem was and providing a fix that was irrelevant. I had to correct him afterwards that the keyboard shortcut was indeed valid when using Ubuntu. I am using a corsair keyboard as well as a corsair mouse. Exact same process works under Mac OS and does not under Ubuntu. Macros can not be manually programed under Ubuntu 19.10 for my mouse. I haven't tried with the keyboard because it's unnecessary outside of gaming. I understand how the software is supposed to work. My computer skills are way above average.

ZirconiumX, 2019-12-29:
Hi, I'm ZirconiumX, and I'm the other developer for ckb-next.

Let's look at your original bug report, shall we?

> I tried this as a macro mapped to a button on my scimitar pro rgb and it does nothing. When i map in Mac OS it works fine.
>
> +lctrl, +lbrace, -lbrace, -lctrl,

- Do you realise how little context there is in this? If you try doing Ctrl-[ in a terminal, of course it will do nothing. Is that because it isn't actually running the macro, or because whatever you're using to test does not respond to that macro? There's no mention of it at all. Only after the bug report gets closed do you mention Firefox.
- Linux has special support for the back button, much like Windows does, so when you press it, it will generate a "back button" event, which for Firefox at least, will go to the previous page. This does not apply for Mac OS, which does not recognise this event; instead, by convention, Cmd-[ will go back. So, if you try to send a "back button" event on Mac OS, it won't be recognised, and if you try to send a Ctrl-[ command on Windows or Linux, it won't be recognised.
- So I tried pressing Ctrl-[ in Firefox (from which I am typing this from), and nothing happened. Sure enough, when you go through the keyboard shortcut list that you kindly linked, there is no mention at all of Ctrl-[ going to the previous page. Instead, they show alt-left arrow, or the backspace key.

What can we conclude from this? You have filed two bug reports for a macro not working, and this turns out to be because the command you have entered does nothing in Firefox. You have wasted the time of both myself and Tasos, and you have also made us annoyed at you by getting angry at *us* for *your* problem, which makes us significantly less likely to bother with your bug reports in the future.

Let me remind you that ckb-next contains *no warranty*, and as such we are under no obligation to develop this program, or deal with these bug reports at all, but we do it out of the goodness of our hearts. Do *not* waste that.

closed 2019-12-29 by ZirconiumX