ckb-next/ckb-next #520: Corsair Nightsword RGB Mouse

freddfx, 2019-12-28:
Requesting support for Corsair Nightsword RGB gaming mouse to be added. 

[ckb-next-dev-detect-report.gz](https://github.com/ckb-next/ckb-next/files/4007486/ckb-next-dev-detect-report.gz)

Settyness, 2020-04-15:
I would like to second this request. This software is amazing and I appreciate the hard work.

cayers71, 2020-05-04:
I third - please add support for the Nightsword mouse.

PranQ, 2020-07-19:
Fourth. Please. While I'm ok so far with the defaults, I'd still like to re-map the buttons and set static lighting.

jannis3005, 2020-07-25:
Fifth. Would be awesome

openaiken, 2020-10-05:
(6th). I'm strongly considering getting a nightsword soon and after looking at closed PRs for new mouse device support, I think it's something I can handle. no promises on timeline or success, but I'll update this issue once I make progress, and open a PR if I get all the keys and stuff working.

tatokis, 2020-10-07:
It should be a matter of adding the usb id like those other PRs, and then figuring out the correct mapping for the LEDs.
If you need help, you can pop in on IRC and hang around for a bit, and we'll be able to help.
https://github.com/ckb-next/ckb-next/pull/621/files This should be a good starting point.

omarsamad-plutoflume, 2020-10-08:
is this added yet?? :+1:

Bligzar, 2020-10-22:
(7th) If help is needed, I am willing to do my best. :)

openaiken, 2020-10-28:
I received my Nightsword last week. In comparing (via iCUE on Win10) it's functionality to my Harpoon RGB (in tandem with my K70 RGB Mk.2) I've determined it's probably going to be fairly complex compared to many of the other supported devices. I am going to clone the source code and get to work on this as soon as I can make time around school (thankfully I am free from the distraction that is Cyberpunk for a few more weeks). I'll keep y'all posted.

For what it's worth, I found that creating a decent profile for the mouse in iCUE and saving it to the hardware makes it completely perfectly usable in Linux, even without ckb-next support yet. Just throwing that out there for anyone thinking about getting one. Great mouse, love everything about it.

openaiken, 2020-10-28:
It would be helpful if someone could consider getting the art assets for it prepared. That area is not my strongsuit... Haha. I'll figure it out though if it comes down to me.

Settyness, 2020-10-29:
You're a legend. Thank you for your time and effort on this endeavor.

egberts, 2020-10-29:
One alpha tester,  available here.

egberts, 2020-10-29:
Looks very similar to a Sabre-class mouse.
```console
Bus 001 Device 002: ID 1b1c:1b5c Corsair CORSAIR NIGHTSWORD RGB Gaming Mouse
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1b1c Corsair
  idProduct          0x1b5c 
  bcdDevice            3.35
  iManufacturer           1 
  iProduct                2 
  iSerial                 3 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0042
    bNumInterfaces          2
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     195
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      29
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
```

egberts, 2020-10-29:
First stab.  Back-end (daemon), first-wave, Vendor and Device ID only.

https://github.com/egberts/ckb-next/commit/ee1171b2ef6056de3e6dd723c7c3d3952bed6cdb

openaiken, 2020-11-01:
Woohoo! Thank you for doing that part! I'm looking forward to testing it and trying to sort out the other features. I noticed that the Nightsword has separate features for when it's not being controlled by iCUE. I'll look at the sabre per your recommendation.

Unless you get to it before me, in which case all power to ya :)

openaiken, 2020-11-06:
Update: I've made a lot of progress. The device is recognized, all DPI and some lighting zones are adjustable by the GUI. I have a lot of semantics to work through and since the Nightsword has profile up/down keys instead of a profile/mode cycle key I might have to do some daemon work. Trying to make it happen tho

openaiken, 2020-11-08:
Just opened PR https://github.com/ckb-next/ckb-next/pull/656 :)

egberts, 2020-11-21:
This issue can be closed once it gets merge, given that the fix is now available (but not yet merged) in #656.

Settyness, 2020-11-25:
Thank you so much for making this a reality.

PranQ, 2020-12-19:
Thank you for getting the Nightsword into ckb-next! It's so nice to not have that default rainbow effect going and to assign my preferred DPI values. The only issue I am seeing is a discrepancy between the DPI selected and the illumination of the 3 LEDs above the thumb. If I click DPI up to max and then back down two steps, all three lights illuminate properly for the medium DPS setting. If I click DPI to the lower two values (two lights or one light illuminated), then sometimes the LEDs turn off completely. When clicking DPI up again, the wrong colors are displayed until reaching the top two DPI settings, which fixes the illumination. Other than that small issue, it's working well.

openaiken, 2020-12-19:
Can you try toggling all but 3 of the DPI settings off? I think that the dpi up/down keys just track the dpi settings and the lights are ignored, as simply a byproduct. I think that's what I did on mine for simplicity.

PranQ, 2020-12-20:
It works perfectly if I disable #s 1 and 2 in the Performance tab. Then I get all 3 lights in the proper colors when switching between them.
This works for me since I never find myself using the lower DPI settings I had configured anyway.
If having #1 and #2 enabled, the lights mess up when switching down from #3 to #2 and don't correct themselves until switching back up from #3 to #4.
Now I notice that the Sniper mode shows the color of mode #3 the first time it is pressed, and then the assigned color for 'sniper' when pressing it a second time. After that, mode 3 shows the sniper color the first time it is selected and is fixed by switching DPI up and down again. This is no biggie for me as I rarely, if ever, use that button.

closed 2022-12-02 by freddfx