ckb-next/ckb-next #522: Macros not working in Ubuntu with Scimitar Pro period!

davidbdeath, 2019-12-29:
Macros do not work in Ubuntu with CKB-Next and the scimitar pro RGB period. It does not matter what macro is entered it does not work period. Basically a fresh install of Ubuntu. Instead of being arrogant and deflecting by complaining about my example how about you actually look into the issue? The arrogant attitude is why IT people and developers get a bad name for being rude.

tdis8629, 2019-12-30:
Just a forewarning; this is probably the last time I'll make a comment about this, depending on how you respond.

[This](https://youtu.be/-I3BDpb7GRY) is a video of my computer, with a Scimitar PRO RGB (Yellow) and a macro that spells out "The Quick Brown Fox Jumped Over The Lazy Dog [!]" on said mouse. I put a live version of Ubuntu 19.10 onto a flash drive, built ckb-next, and attempted the same thing, along with Alt+F2, Ctrl+Alt+F3 (tty3), and Super+L, and they all worked. Granted, this is only on one computer, but it also proves it on multiple versions of Linux.

Please try this: Bind individual keys to the 12 buttons, and test to see if they all work (bind the home row, as an example). IF it doesn't work at all, or specific buttons don't work, then is is more information than we had before, instead of just "I tried this one combination and it doesn't work."

Let me just say this, too: The project is mainly being held together by two developers, with a few others helping in specific cases. I do not believe the prior two responses were condescending or arrogant, but there also wasn't enough information to figure out what was wrong. At the same time, though, no one ever assumed your proficiency with Linux or computers in general. Assumptions and general negativity do not help in a report or response, and just replying to the closed topic would have done more good than just creating more bug reports with increasing frustration.

davidbdeath, 2019-12-31:
I did some testing and created a video showcasing the issue. I was unable to record a macro but was able to manually enter macro. The clear button does not work, I had to unbind to clear. The macro's did not function properly. I tried +1, -1 and once triggered it would output 1 until I unplugged the mouse. I tried "+lctrl, +pgdn, -pgdn, -lctrl", and that did not change my active tab. It did work when I manually pressed "lctrl + pgdn" on the keyboard. Before I recorded the video I tried all 12 button. Maybe there is an issue with my Ubuntu install. I primarily only use GNU/Linux for servers.

https://youtu.be/u8ZAa3Q_kfE

tdis8629, 2019-12-31:
1. Looking at gedit, switching tabs is Ctrl+Alt+Page_Up/Down. Unless you have a custom shortcut, that may be why it isn't working.

2. The clear button is for the area labeled "Macro Sequence." I just tested it, and it clears that area, while also stopping the macro recording. The fact that it also doesn't clear the macro created while recording might actually be a bug.

3. I was able to replicate the repeating 1 issue: It seems that the macro sees a space [ ] as an endpoint, so it is only seeing +1, instead of +1,-1 (As if you are just holding down the 1 key). If you type out a macro on your keyboard (use Num- as an example key), you'll notice that the macro doesn't contain any spaces. This is most likely another bug that was overlooked.

3. For the issue of being unable to type the macros on other devices (usually involving the mouse), that seems to be a longstanding issue brought up in #10, (initially from mattanger/ckb-next#121) with [tatokis pointing out](https://github.com/ckb-next/ckb-next/issues/10#issuecomment-400202432) that it would possibly wouldn't be fixed. I can confirm it by plugging in my Strafe RGB keyboard to my laptop and the software failing to type out my macro combination from the attached keyboard.

davidbdeath, 2020-01-01:
Removing spaces seems to be the answer. Thanks for looking into it.

Since the OS lets me use any number of custom keyboard shortcuts. Focusing on what keyboard shortcuts i'm programing seems illogical. Especially when i'm stating it works when I use them directly with the keyboard. 

I used OBS to capture the recording without ever really using the software before and the recording came out jumpy and didn't capture me using that combo to switch tabs at the end.

I would suggest a revision in the UX to support programing keyboard shortcuts without having to program them as Macros. A built in reference list for the names of keys when creating Macros would be useful. If spaces are not ignored, when clicking apply space could be auto removed. I'm guessing they are being stored in an array. The software has the potential to be far better than iCue with a few UX improvements.

tdis8629, 2020-01-16:
I haven't forgotten about this issue, I just needed time to study the code with my limited C++ knowledge.

I am still trying to figure out how to truncate the whitespaces from the macro section, but I do have some ideas (as hacky as they are)

For the clear button, there is a comment about how to possibly progress with it (either turn it into a clear all, or set delays to 0). If you want to add it yourself, include the following to line 778 of [src/gui/rebindwidget.cpp](https://github.com/ckb-next/ckb-next/blob/master/src/gui/rebindwidget.cpp#L772): 
`ui ->pteMacroBox->clear();`

This will clear the entire box and stop recording (as it trips case(1) just below), but I don't believe that's ultimately the proper way to do it.

closed 2022-10-03 by tatokis