ckb-next/ckb-next #523: [K70 MK.2 RGB] Save to hardware disabled and saved profiles not loaded

albertored, 2019-12-30:
I'm using ckb-next built from source (from v0.4.2 tag) on ubuntu 16.04.

The "Save to hardware" button is always greyed out and the gui does not show the rgb profiles stored in keyboard internal memory.

tatokis, 2019-12-30:
Saving to hardware is not supported for newer devices that support hardware animations as the format has not been reverse engineered and corsair does not provide any information.

albertored, 2019-12-30:
I didn't know, sorry for the noise and thanks for the quick reply!

tatokis, 2019-12-30:
It's fine, don't worry! There wasn't an issue open about this specific keyboard, so it's difficult to find the relevant issue. 

I'll link it later if I don't forget so that you can subscribe to that for any possible updates regarding the situation

ItzSwirlz, 2019-12-30:
Thanks for opening... Have issue with Harpoon RGB. Worked at first with the normal presets but I couldn't seem, even purging config and reinstalling (Ubuntu via repository) to get it back and it didn't come back.

arBalasquide, 2020-03-04:
In that case, is there a way to have ckb-next running as a service to keep my settings active?

ZirconiumX, 2020-03-04:
ckb-next already runs as a daemon and GUI pair, and the GUI lets you save
animation profiles, so as long as the daemon and GUI are running, you'll
have your animations.

CyrilBrulebois, 2020-05-21:
Hey, and thanks for CKB, that's a very fun tool to play with.

I was also a bit surprised by the lack of “Save to hardware” feature. I can totally understand that saving/updating profiles is not supported at the moment. But is it also expected to see predefined profiles to go away as well?

Out of the box, there was a sliding rainbow, some pinwheel-based rainbow, and I don't remember the third one. Now, I'm stuck with a sliding rainbow (which isn't that bad) but I cannot change luminosity any more for example.

Trashing presets might have been an undesired side effect, but that looks a bit harsh? :)

massimopavoni, 2023-03-10:
Is there any light at the end of the tunnel for saving to hardware even just simple bindings and profiles with custom dpi colors? I have no interest in animations, but having to deal with another service and application opened in the background is kind of heavy.

gravityfargo, 2024-05-02:
This issue has been marked as stale due to age and/or lack of activity. A generated support was not supplied.

Please feel free to reopen it if you have more to add or if the issue still needs attention. However, it would be best to [open a new issue](https://github.com/ckb-next/ckb-next/issues/new/choose) and provide a newly generated report issue using our updated templates.

closed 2024-05-02 by gravityfargo