ckb-next/ckb-next #524: macOS Catalina quickinstall script failing (cmake)

joemulray, 2020-01-03:
Trying to install ckb-next onto macOS catalina, and the install script is failing. Also tried the manual install still no luck. Any ideas on what could be causing this issue?
```
➜  ckb-next git:(master) ./quickinstall
-- The C compiler identification is AppleClang 11.0.0.11000033
-- The CXX compiler identification is AppleClang 11.0.0.11000033
-- Check for working C compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc
-- Check for working C compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++
-- Check for working CXX compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found Git: /usr/local/bin/git (found version "2.19.0") 
-- ckb-next version: 0.4.2-26-g0c1b7b9 (Non-release)
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Success
-- Found Threads: TRUE  
-- Performing Test ICONV_SECOND_ARGUMENT_IS_CONST
-- Performing Test ICONV_SECOND_ARGUMENT_IS_CONST - Failed
-- Found Iconv: /usr/lib/libiconv.dylib
-- CoreFoundation found: /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.15.sdk/System/Library/Frameworks/CoreFoundation.framework
-- CoreGraphics found:   /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.15.sdk/System/Library/Frameworks/CoreGraphics.framework
-- IOKit found:          /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.15.sdk/System/Library/Frameworks/IOKit.framework
-- Searching for running ckb and/or ckb-next GUI
-- NOTE:
    Privileged access is required for operations upon the daemon at
    configure time as it is owned by root. If this is unacceptable,
    disable SAFE_INSTALL and prepare the filesystem manually.
-- launchd detected
-- Importing org.ckb-next.daemon.plist (launchd)
-- Running job with org.ckb-next.daemon plist detected
-- Found QuaZip: /usr/local/lib/libquazip.dylib  
qt 5.14.0
quazip 0.8.1
-- Found ZLIB: /usr/lib/libz.dylib (found version "1.2.11") 
-- Foundation found:   /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.15.sdk/System/Library/Frameworks/Foundation.framework
-- CoreAudio found:    /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.15.sdk/System/Library/Frameworks/CoreAudio.framework
-- AudioToolbox found: /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.15.sdk/System/Library/Frameworks/AudioToolbox.framework
-- Generating binary.qrc
-- Importing ckb-next-key.gpg
-- Generating image.qrc
-- Importing ckb-next-monochrome.png
-- Importing ckb-next.png
-- Importing icon_blank.png
-- Importing icon_mode.png
-- Importing icon_mode1.png
-- Importing icon_mode1_hardware.png
-- Importing icon_mode2.png
-- Importing icon_mode2_hardware.png
-- Importing icon_mode3.png
-- Importing icon_mode3_hardware.png
-- Importing icon_plus.png
-- Importing icon_profile.png
-- Importing icon_profile_hardware.png
-- Importing overlay_glaive.png
-- Importing overlay_harpoon.png
-- Importing overlay_ironclaw.png
-- Importing overlay_katar.png
-- Importing overlay_m65.png
-- Importing overlay_m95.png
-- Importing overlay_polaris.png
-- Importing overlay_sabre.png
-- Importing overlay_scimitar.png
[... 181 lines cut ...]
                                ^
/usr/local/opt/qt/lib/QtCore.framework/Headers/qcompilerdetection.h:643:55: note: expanded from macro 'Q_DECL_DEPRECATED_X'
#    define Q_DECL_DEPRECATED_X(text) __attribute__ ((__deprecated__(text)))
                                                      ^
/Users/josephmulray/Documents/REPOS/ckb-next/src/gui/extrasettingswidget.cpp:196:5: error: no type named 'setIdleTimer' in 'KbManager'; did you mean 'IdleTimer'?
    KbManager::setIdleTimer(checked);
    ^~~~~~~~~~~~~~~~~~~~~~~
    IdleTimer
/Users/josephmulray/Documents/REPOS/ckb-next/src/gui/idletimer.h:4:7: note: 'IdleTimer' declared here
class IdleTimer
      ^
/Users/josephmulray/Documents/REPOS/ckb-next/src/gui/extrasettingswidget.cpp:201:16: error: no member named 'setIdleTimer' in 'KbManager'
    KbManager::setIdleTimer(false);
    ~~~~~~~~~~~^
/Users/josephmulray/Documents/REPOS/ckb-next/src/gui/extrasettingswidget.cpp:202:16: error: no member named 'setIdleTimer' in 'KbManager'
    KbManager::setIdleTimer(true);
    ~~~~~~~~~~~^
3 errors generated.
make[2]: *** [src/gui/CMakeFiles/ckb-next.dir/extrasettingswidget.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
1 warning generated.
make[1]: *** [src/gui/CMakeFiles/ckb-next.dir/all] Error 2
make: *** [all] Error 2
An error occurred,
```

tatokis, 2020-01-03:
Please try again with the latest code. It should now be fixed
https://git.io/JepZk

joemulray, 2020-01-04:
working now! thank you @tatokis

closed 2020-01-04 by joemulray