ckb-next/ckb-next #526: Corsair QL Fans + Lighting Node Core support

lth1301, 2020-01-11:
I saw ckb-next support Corsair LL120 fans!
Would it be possible to support QL120 fans as well?

QL is the newest LED fans from Corsair and they use Lighting Node 'Core' for LED control. (LL fans use Lighting Node 'Pro')

I'm not a programmer, but I'll help you whatever I can.

Thanks

thespyboss, 2020-02-13:
@lth1301 If you want to try it, install this version of ckb. https://github.com/mrdev023/ckb-next
I went through the animations and i have a gist of what works. If it does not work. Ping me and we can put something in #453

lth1301, 2020-02-13:
Okay.

Process
1. I downloaded the source code from the link you provided.
2. Built an app using cmake.
3. installation process
 went smoothly. (ckb-next 0.4.1 installed)

Outcome
· ckb-next does not detect the QL Fans ('Lighting Node Core' in this case)
· it indicates that ckb-next 0.4.2 is available.
· Installed 0.4.2 but it does not detect QL Fans either...

[image]

thespyboss, 2020-02-13:
@lth1301 Ok! If you want to or I can, put in a request on the #453 discussion and see what we can get figured out! I'm guessing that the fans and lighting node core has a different way of running compared to the LL fan series and the Lighting node pro system.

gravityfargo, 2023-04-21:
closing in favor of #639
no detect report was given here

closed 2023-04-21 by gravityfargo