ckb-next/ckb-next #532: RGB Lights Turn Off During Screensaver

hoxaen, 2020-01-15:
I believe that what I am encountering is a bug within ckb-next, but as this is my first time writing an issue report in Github,  I could very well be wrong. 

I recently installed Debian 10 XFCE on my computer and added my usual applications and customizations. Since there is no official package for Debian, I compiled ckb-next from source via the instructions. I've done this before, never had any problems. Everything seems to work fine--I can change button macros, mouse speed, RGB theme, etc. However, when my computer is idle, and xscreensaver starts, the lights on both my keyboard and mouse turn off. The only light that stays on is the little brightness icon on my keyboard. When I move my mouse to get rid of the screensaver, either the mouse or the keyboard will light back up, but never both. The only way I've managed to turn the lights back on is to either unplug and plug either the mouse or keyboard back in, or go into ckb-next's settings and switch to a different RGB theme and back again to the one I had set. While this is not a dealbreaker for me, it is a minor aggravation, so I thought I'd bring it to your attention.

Also, I've experienced this in the past, but for whatever reason, if ckb-next is set to autostart in XFCE, when I log into my user account, the main setting screen always appears, even though it is launching with the --background command. I've tried to delay the start (an option in the settings), and I've also tried autostarting a shell script instead of the default auto-launcher with both --background and --session tried--neither seem to make any difference. I don't know if I should make a different issue report for this or not. 

My keyboard is the K68, and my mouse is the Scimitar Pro RGB. I'm not sure what other information would be helpful, but I would be happy to provide it. As I said earlier, running Debian 10 stable with no packages from testing or backlogs.

tatokis, 2020-01-15:
I just fixed the light issue https://github.com/ckb-next/ckb-next/commit/778e97b842c11050a54909b1647a0934dbcc07a6

Regarding the UI showing up on startup, make sure the autostart checkbox is checked in the settings tab, and then on a boot where the gui shows up, run `ps aux | grep ckb-next` and paste the output here.

hoxaen, 2020-01-16:
Thanks so much for the quick fix! I will have to see if it helps. How would I go about applying your patch? Do I just pull the git repository and do another ./quickinstall or is there a better way?

Here is the output:
`root       766  0.8  0.0 314264  4336 ?        Ssl  19:39   0:00 /usr/local/bin/ckb-next-daemon
matt      1221  3.1  0.5 1290224 86880 ?       Sl   19:39   0:01 /usr/local/bin/ckb-next -session 2d775f010-e1db-4be4-a32e-075ff4edca0c_1579135119_585416
matt      1485  0.0  0.0   3848   988 ?        S    19:39   0:00 /usr/local/libexec/ckb-next-animations/wave --ckb-run
matt      1486  0.1  0.0   3840  1000 ?        S    19:39   0:00 /usr/local/libexec/ckb-next-animations/random --ckb-run
matt      1487  0.2  0.0   3848  1060 ?        S    19:39   0:00 /usr/local/libexec/ckb-next-animations/ripple --ckb-run
matt      1570  0.0  0.0   6208   880 pts/0    S+   19:39   0:00 grep ckb-next`

That is with me attempting to start the program via my little shell script. If you need the output of the program's autostart .desktop file, I will provide that as well.

'#!/bin/bash
/usr/local/bin/ckb-next --session &'

tatokis, 2020-01-19:
Yeah, just git pull and quickinstall.

This should fix the session restore issue.
https://github.com/ckb-next/ckb-next/commit/cbc42667f09dc9fdaa1f53d43c99e0e753b6db5d

I think it may be a Qt bug since we already tell it to start minimized if we're restoring a session, but I don't know why it stopped working suddenly.

Do note, it's `-session`, not `--session`.

hoxaen, 2020-01-21:
Thanks again for the fixes! I pulled in the new version and everything seems to be working properly again.

closed 2020-01-21 by hoxaen