ckb-next/ckb-next #534: No prompt for Input monitoring on Catalina

mekeirc, 2020-01-17:
After updating ckb-next no longer works with my Corsair Strafe RGB after it working fine on Mojave.
The software claims I should enable the ckb-daemon in Security > Privacy > Input Monitoring but I never get the prompt and nothing is there (with no way to add software to it)

Things I've tried:
- Restarting ckb-next multiple times
- Restarting the Mac
- Unplugging/replugging the keyboard multiple times
- Running the ckb-next-daemon restart shell script in Package Contents

Neither seem to do the trick, it's a shame because I love this software and this was essential to getting my media keys working on the strafe.

Version: macOS Catalina 10.15.3 Beta (19D62e)
[ckb-next-report-1579258649.gz](https://github.com/ckb-next/ckb-next/files/4076495/ckb-next-report-1579258649.gz)

TheRealNewbie, 2020-01-24:
Same problem with me. Have you figured out the issue?

mekeirc, 2020-01-24:
> Same problem with me. Have you figured out the issue?

Nope, I haven't had working RGB lighting or media keys since upgrading unfortunately. I'm using hotkeys on BeardedSpice for temporary media controls.

mekeirc, 2020-02-14:
Update on this, it appears that the precompiled dmg packages are out of date and do not include the permissions required. Compiling from source using the quickinstall then restarting macOS fixes the issue. I can imagine the vast majority of macOS users will use these precompiled dmg files so they will need to be updated.

tatokis, 2020-03-01:
Does this help? 

http://ckb-next.info/mac/ckb-next_v0.4.2-41-gf4c9184_10.12-SDK.dmg

```
3e0ec5f41a6bcdcf07e767626b71cd5e73ecabd21c68d1caf871f80880c826c3  ckb-next_v0.4.2-41-gf4c9184_10.12-SDK.dmg
```

Ricardonacif, 2020-04-14:
My issue was fixed with the latest Catalina (10.15.4).

TheRealNewbie, 2020-04-23:
My issue has not been resolved actually. I cannot see daemon ask
requesting for permission in input monitoring. Am I missing something?

mekeirc, 2020-04-23:
@TheRealNewbie Have you tried downloading the repo and using the quick install? The issue was the DMG installers were out of date.

tatokis, 2020-04-23:
The dmg I linked above was up to date with the latest changes. Nothing else changed since that was mac specific, so if that dmg doesn't work, then the issue is something else.

TheRealNewbie, 2020-04-23:
I also tried the file you sent. It didn't work. How do I do the "repo and
quick install thing?"
Thanks

jaspermutsaerts, 2020-08-18:
Above didn't work for me either. I found (created) a workaround,  [a workaround](https://github.com/ckb-next/ckb-next/issues/516#issuecomment-675569874)
#516

ameola, 2020-10-08:
This didn't work for me. I had to manually add the Daemon to the Security & Privacy dialog. Open the dialog and hit the padlock. There's a + sign that allows you to add new entries. Add the Daemon. It's located under Macintosh HD/Libaray/Application Support/ckb-next-daemon. Once done, you'll need to restart the daemon:

sudo launchctl unload /Library/LaunchDaemons/org.ckb-next.daemon.plist; sudo launchctl load /Library/LaunchDaemons/org.ckb-next.daemon.plist

jaspermutsaerts, 2020-10-09:
> This didn't work for me. I had to manually add the Daemon to the Security & Privacy dialog. Open the dialog and hit the padlock. There's a + sign that allows you to add new entries. 

This the part I created a workaround for: if there aren’t any entries yet, the + isn’t there either. But adding it manually is indeed the fix for most.