ckb-next/ckb-next #540: Lenovo Y740 Keyboard/Fan Lights

FullerBot, 2020-01-24:
Hello! I'm wondering if you'd eventually be able to support lighting controls for the Lenovo Y740 laptop, which uses iCue for it's lighting controls. It has fan lights, keyboard lights and a symbol on the back of the laptop that can all be controlled/adjusted.

If you want someone to test stuff, I'm happy to help.

Thanks for your time,
FullerBot

bogdanm, 2020-03-01:
I'd love to see this too! I can help if needed.

bogdanm, 2020-03-02:
If it helps, this is what dmesg reports:

```
[    2.329404] usb 1-6: New USB device found, idVendor=048d, idProduct=c935, bcdDevice=13.11
[    2.329405] usb 1-6: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[    2.329406] usb 1-6: Product: Corsair Gaming K95 RGB PLATINUM Keyboard
[    2.329406] usb 1-6: Manufacturer: Corsai
[    2.457715] usb 1-7: new high-speed USB device number 4 using xhci_hcd
```

I'm trying to patch the source and use this VID/PID instead of the default ones, but I hit some weird apt dependency issues that I couldn't solve yet and I can't even compile the original sources.

markuswall, 2020-05-25:
Check out https://github.com/ckb-next/ckb-next/issues/330, seems to be the same HW (I just got the legion branch working on my Y740)

ismu, 2020-05-26:
@markuswall Do you test legion branch on Lenovo Y740 15 inches? Can you set one color for all keys? Fn and keyboard lighting key works, can you turn on/off the backlight on linux?

markuswall, 2020-05-28:
@ismu Yes, against the 15 inch Y740. I did very brief testing, as I just got the laptop. I tested setting the same base color for all keys, as well as some individual keys and fan side ports in another color just to see if it was feasible.

ismu, 2020-05-31:
@markuswall Could you give me a any contact to you? I have some questions about linux on this laptop, what is working what not.

Magicrafter13, 2021-08-12:
I'd also like this. I have the 17" model, and the keyboard layout is different.

FullerBot, 2023-02-11:
Hello! It's been a while. 

Just updated CKB-Next (also updated the os, am on 22.04 now), issue still persisting.

I have the CKB-Next logs, with and without my external keyboard (Corsair K95 Platinum XT), if anyone sees this and can utilize the information within them. 

To clarify: I have a Lenovo Y740 17 inch version, specifically a Lenovo Legion Y740-17IRHg

Thanks,
FullerBot

FullerBot, 2023-02-11:
[ckb-next-report-1676117757.gz](https://github.com/ckb-next/ckb-next/files/10713639/ckb-next-report-1676117757.gz)
[ckb-next-report-1676117645.gz](https://github.com/ckb-next/ckb-next/files/10713640/ckb-next-report-1676117645.gz)