ckb-next/ckb-next #541: Constant CPU Usage

aaronw2, 2020-01-31:
When show animated is enabled, ckb-next consumes a constant 12% of CPU even when there is no keyboard activity and when it is minimized. If there is no keyboard activity or when it is minimized it should not be attempting to animate the keyboard.

stefanotorresi, 2020-03-11:
I am also affected by this. I'm using the `ckb-next-git` package from Arch User Repositories.

ckb-next-dev-detect - Wed Mar 11 12:53:06 CET 2020

```
Kernel: Linux breakthrough 5.4.23-1-MANJARO #1 SMP PREEMPT Fri Feb 28 22:45:29 UTC 2020 x86_64 GNU/Linux

OS Version:
LSB Version:	n/a
Distributor ID:	ManjaroLinux
Description:	Manjaro Linux
Release:	19.0.2
Codename:	Kyria

USB Devices:
Couldn't open device, some information will be missing
Couldn't open device, some information will be missing

Bus 001 Device 025: ID 1b1c:1b37 Corsair 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1b1c Corsair
  idProduct          0x1b37 
  bcdDevice            3.08
  iManufacturer           1 
  iProduct                2 
  iSerial                 3 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0042
    bNumInterfaces          2
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     132
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
[... 363 lines cut ...]
	libk5crypto.so.3 => /usr/lib/libk5crypto.so.3 (0x00007f4fde019000)
	libcom_err.so.2 => /usr/lib/libcom_err.so.2 (0x00007f4fde013000)
	libkrb5support.so.0 => /usr/lib/libkrb5support.so.0 (0x00007f4fde004000)
	libkeyutils.so.1 => /usr/lib/libkeyutils.so.1 (0x00007f4fddffd000)
	libresolv.so.2 => /usr/lib/libresolv.so.2 (0x00007f4fddfe4000)
	librt.so.1 => /usr/lib/librt.so.1 (0x00007f4fddfd9000)
	libsndfile.so.1 => /usr/lib/libsndfile.so.1 (0x00007f4fddf5b000)
	libasyncns.so.0 => /usr/lib/libasyncns.so.0 (0x00007f4fddf53000)
	libGLdispatch.so.0 => /usr/lib/libGLdispatch.so.0 (0x00007f4fdde9c000)
	libGLX.so.0 => /usr/lib/libGLX.so.0 (0x00007f4fdde69000)
	libfreetype.so.6 => /usr/lib/libfreetype.so.6 (0x00007f4fddd9b000)
	libgraphite2.so.3 => /usr/lib/libgraphite2.so.3 (0x00007f4fddd74000)
	liblzma.so.5 => /usr/lib/liblzma.so.5 (0x00007f4fddd4c000)
	liblz4.so.1 => /usr/lib/liblz4.so.1 (0x00007f4fddd2a000)
	libgcrypt.so.20 => /usr/lib/libgcrypt.so.20 (0x00007f4fddc0c000)
	libicudata.so.65 => /usr/lib/libicudata.so.65 (0x00007f4fdc15b000)
	libpcre.so.1 => /usr/lib/libpcre.so.1 (0x00007f4fdc0e6000)
	libFLAC.so.8 => /usr/lib/libFLAC.so.8 (0x00007f4fdc0a7000)
	libogg.so.0 => /usr/lib/libogg.so.0 (0x00007f4fdc09c000)
	libvorbis.so.0 => /usr/lib/libvorbis.so.0 (0x00007f4fdbe70000)
	libvorbisenc.so.2 => /usr/lib/libvorbisenc.so.2 (0x00007f4fdbbc7000)
	libX11.so.6 => /usr/lib/libX11.so.6 (0x00007f4fdba85000)
	libbz2.so.1.0 => /usr/lib/libbz2.so.1.0 (0x00007f4fdba72000)
	libgpg-error.so.0 => /usr/lib/libgpg-error.so.0 (0x00007f4fdba50000)
```

Kodikuu, 2020-05-21:
I had the ckb-next-daemon using 100% of a core after building on commit f4c9184. I updated to commit ca9bf57, cleaned, and reinstalled, and the appear appears to have been resolved.

This is with static white on a K95 with an outward yellow shimmer-ripple on the bar at the top, and a matching yellow shimmer-wave on my Scimitar Pro. I'm using OpenSUSE Tumbleweed, fully updated.

Not sticking around to provide more info, I've got too much on my plate already.

veilofsecurity, 2020-07-28:
This also exists on MacOS (i've seen it on multiple versions, tried a Macbook, and even a clean install) with the K95 Platinum. Same symptoms as others, constant 99-100% usage of a core.

gravityfargo, 2024-05-02:
This issue has been marked as stale due to age and/or lack of activity. 

Please feel free to reopen it if you have more to add or if the issue still needs attention. However, it would be best to [open a new issue](https://github.com/ckb-next/ckb-next/issues/new/choose) and provide a newly generated report issue using our updated templates.

closed 2024-05-02 by gravityfargo