ckb-next/ckb-next #543: Keyboard not detected: K95 RGB Platinum XT

rporter220, 2020-02-07:
I just purchased a K95 RGB Platinum XT keyboard hoping that it would be compatible with ckb-next since the K95 Platinum is on the supported keyboard list. The keyboard does not seem to be picking up the keyboard with the latest git version of ckb-next.

I'm thinking that since this is technically a different model it probably has a different device ID or something along those lines.

If there is anything I can provide, please let me know and I assist any way I can to help getting this keyboard supported. :-)

Thank you!

rporter220, 2020-02-09:
If it is just a USB Device ID issue, I hope this helps:
USB Device ID reported is `1b1c:1b89`

tatokis, 2020-03-01:
The keyboard uses the completely new protocol that no one has implemented. Unless someone steps up and does it, it won't be supported for quite a while

rporter220, 2020-05-12:
Unfortunately I'm not a software developer, but is there any way I can assist for someone to start putting this in their pipeline?  

I'm willing to purchase a keyboard for someone to begin this or if there is anything else I can contribute, just let me know! :-)

rporter220, 2020-05-12:
file:///home/richard/Documents/ckb-next-report-1589319382.gz

rporter220, 2020-05-12:
[ckb-next-report-1589319382.txt](https://github.com/ckb-next/ckb-next/files/4618580/ckb-next-report-1589319382.txt)

tatokis, 2020-09-16:
Can you please dump the HID descriptors?
`sudo usbhid-dump -d 1b1c:1b89`

ghost1347, 2020-10-01:
Hey I noticed he has not gotten back to you yet, Can I post mine?

tatokis, 2020-10-01:
Sure

ghost1347, 2020-10-02:
003:006:003:DESCRIPTOR         1601607982.430057
 05 01 09 02 A1 01 85 05 09 01 A1 00 05 09 19 01
 29 20 15 00 25 01 95 20 75 01 81 02 05 01 09 30
 09 31 16 01 00 26 FF 7F 75 10 95 02 81 06 09 38
 15 81 25 7F 75 08 95 01 81 06 C0 C0

003:006:002:DESCRIPTOR         1601607982.433051
 06 42 FF 09 02 A1 01 15 00 26 FF 00 95 40 75 08
 09 02 81 02 C0

003:006:001:DESCRIPTOR         1601607982.436041
 06 42 FF 09 01 A1 01 15 00 26 FF 00 95 40 75 08
 09 01 81 02 95 40 09 01 91 02 C0

003:006:000:DESCRIPTOR         1601607982.439051
 05 01 09 06 A1 01 85 01 05 07 19 E0 29 E7 15 00
 25 01 75 01 95 08 81 02 19 00 29 97 15 00 25 01
 75 01 95 98 81 02 05 08 19 01 29 03 25 01 75 01
 95 03 91 02 75 05 95 01 91 01 C0 05 0C 09 01 A1
 01 85 02 05 0C 19 00 2A FF 0F 15 00 26 FF 0F 75
 10 95 02 81 00 C0 05 01 09 80 A1 01 85 06 19 81
 29 83 15 00 25 01 95 03 75 01 81 02 95 01 75 05
 81 01 C0

ghost1347, 2020-10-03:
is there anything else you need for it?

tatokis, 2020-10-07:
This will do for now. Thank you

rporter220, 2020-10-07:
OMG I'm sorry.. This notification was filtered out in my email.

I checked and my descriptors are mostly the same.. Posting just in case

```
003:002:003:DESCRIPTOR         1602053415.782483
 05 01 09 02 A1 01 85 05 09 01 A1 00 05 09 19 01
 29 20 15 00 25 01 95 20 75 01 81 02 05 01 09 30
 09 31 16 01 00 26 FF 7F 75 10 95 02 81 06 09 38
 15 81 25 7F 75 08 95 01 81 06 C0 C0

003:002:002:DESCRIPTOR         1602053415.785484
 06 42 FF 09 02 A1 01 15 00 26 FF 00 95 40 75 08
 09 02 81 02 C0

003:002:001:DESCRIPTOR         1602053415.788474
 06 42 FF 09 01 A1 01 15 00 26 FF 00 95 40 75 08
 09 01 81 02 95 40 09 01 91 02 C0

003:002:000:DESCRIPTOR         1602053415.792467
 05 01 09 06 A1 01 85 01 05 07 19 E0 29 E7 15 00
 25 01 75 01 95 08 81 02 19 00 29 97 15 00 25 01
 75 01 95 98 81 02 05 08 19 01 29 03 25 01 75 01
 95 03 91 02 75 05 95 01 91 01 C0 05 0C 09 01 A1
 01 85 02 05 0C 19 00 2A FF 0F 15 00 26 FF 0F 75
 10 95 02 81 00 C0
```

ghost1347, 2020-10-20:
Is support going to be added? if so when?

FryDay, 2020-12-21:
Has there been any recent progress on this?

RoyvanEmpel, 2020-12-27:
Would really appreciate it if there was support for it. :)
Have been waiting for a while now :/

rvdavid, 2020-12-27:
Just got my keyboard and fell into the rabbit hole of trying to figure out why my "k95 platinum" was not being detected and found out that I had a k95 platinum xt.

I was able to set my keyboard up clunkily using windows via virtualbox and usb passthrough.
Merry Christmas and all that. I'll check back in march.

rrediske, 2021-01-24:
Mine's a little different, I'm on Manjaro Plasma 5.9.16-1, I'm quite interested in helping out if you need testing or information. 

sudo usbhid-dump -d 1b1c:1b89
[sudo] password for uglylinuxusername: 
005:003:003:DESCRIPTOR         1611461043.724591
 05 01 09 02 A1 01 85 05 09 01 A1 00 05 09 19 01
 29 20 15 00 25 01 95 20 75 01 81 02 05 01 09 30
 09 31 16 01 00 26 FF 7F 75 10 95 02 81 06 09 38
 15 81 25 7F 75 08 95 01 81 06 C0 C0

005:003:002:DESCRIPTOR         1611461043.724762
 06 42 FF 09 02 A1 01 15 00 26 FF 00 95 40 75 08
 09 02 81 02 C0

005:003:001:DESCRIPTOR         1611461043.724878
 06 42 FF 09 01 A1 01 15 00 26 FF 00 95 40 75 08
 09 01 81 02 95 40 09 01 91 02 C0

005:003:000:DESCRIPTOR         1611461043.725613
 05 01 09 06 A1 01 85 01 05 07 19 E0 29 E7 15 00
 25 01 75 01 95 08 81 02 19 00 29 97 15 00 25 01
 75 01 95 98 81 02 05 08 19 01 29 03 25 01 75 01
 95 03 91 02 75 05 95 01 91 01 C0 05 0C 09 01 A1
 01 85 02 05 0C 19 00 2A FF 0F 15 00 26 FF 0F 75
 10 95 02 81 00 C0 05 01 09 80 A1 01 85 06 19 81
 29 83 15 00 25 01 95 03 75 01 81 02 95 01 75 05
 81 01 C0

joffie87, 2021-02-02:
I just jumped ship from win10 and ended up here too. I would also greatly appreciate it if we could get this one supported. Thank you for your efforts, it sounds fantastic so far.

dcat52, 2021-02-06:
@tatokis 
I just got a XT keyboard today. Running Kubuntu 20.10 as my only system. I haven't done driver development before but have several years of CS/Robotics background with my CE degree.
I am eager to learn this type of stuff, If you can provide some guidance to get me started I can dedicate time to this.

1. To start off, I noticed in the wiki pages, it mentions Corsair Protocol 2. I am guessing this would be what we call Protocol 3 then?
2. What is the best way to start, get my Windows device up and running with a probe on the data pins of the USB? Or use some software and attempt to write to the device and listen for the response?

Ravenslofty, 2021-02-06:
If you want to help, it's probably best you join us in IRC.

The Wiki document is out of date (I wrote it after all).

Ravenslofty, 2021-02-06:
And quite importantly: don't put a probe across your USB connector in case you short it and kill it.

tatokis, 2021-03-18:
There should be some support for the K95 Platinum XT in the `k95-xt` branch. There are some key mapping issues, and the enter key currently gets stuck when the GUI starts, but other than that it should be okay to test.

pinheadmz, 2021-03-19:
My keyboard is:

`Bus 001 Device 003: ID 048d:c935 Integrated Technology Express, Inc. Corsair Gaming K95 RGB PLATINUM Keyboard`

I installed the k95-xt branch: `ckb-next 0.4.3-71-gbd1cd3b`

but the qt app still says `No devices connected`

thoughts on troubleshooting? Thank you.

tatokis, 2021-03-19:
@pinheadmz That's not a K95 Platinum XT, let alone a K95 Platinum.
Only thing it shares with the actual K95 Platinum is the name. Have a look at the legion branch instead, but that's horribly out of date at this point and there never was GUI support for it added.

TheWebMachine, 2021-05-03:
Hi all! Great work here. I just built from the `k95-xt` branch and was thrilled to see things working so well. I did find the following observations that may be of interest:

1. The Enter key didn't get stuck at GUI start for me; awesome!
2. Switching polling rates seems to crash out the service and a full system reboot is needed to get ckb-next to cooperate again...prior to reboot the keyboard reverts to default HID and still works fine
3. I can't seem to get the locking keys working right. To be precise: pressing Caps, Scroll, Num locks do toggle within the OS properly, but no matter what option I select in ckb-next, no status is rendered on the keyboard (flying blind). Setting an option like 'RGB Indicator' always results in the status at _service start_ being displayed by the keys with no status change being reflected...so it would seem changes to Caps, Scr, Num lock status in OS are not being updated by ckb-next-daemon and sent to the keyboard's LEDs, perhaps?
4. I too suffer the numpad mismapping with 7, 8, 9, 0, and . (dot); but this is a known issue: #711 

I haven't dug much deeper than that yet. Here's my report (the service crash is from the attempt to switch polling rates):
[ckb-next-report-1620015354.txt](https://github.com/ckb-next/ckb-next/files/6412827/ckb-next-report-1620015354.txt)

Shall I post separate issues for 2 and 3 or are they already widely known by the team?

tatokis, 2021-05-03:
@TheWebMachine 
Other than the crash with the pollrate switch, everything else is a known issue. Can you perhaps attach gdb and send the backtrace with debug symbols?

As for the indicator lights, make sure they are all off, stop the daemon `sudo systemctl stop ckb-next-daemon`, install wireshark, add yourself to the wireshark group and run lsusb to get the bus id your keyboard is plugged into.
Run `sudo modprobe usbmon && sudo setfacl -m u:$USER:r /dev/usbmon*` and open wireshark. Double click on usbmonX where X is the bus id you noted down earlier.
Then, light up first Num Lock, then Caps Lock (by pressing the respective keys). For scroll lock you can run `xset led named 'Scroll Lock'`.

Once finished, save the capture and attach it here.

TheWebMachine, 2021-05-04:
I'm using a stock kernel so not sure if I can get the debug but I'll attempt to pull the backtrace nonetheless.

I couldn't get wireshark to show the usbmon devices no matter what I did with the wireshark usergroup, dpkg-config, and `setfacl`, so I had to run as root to capture. That said, I did get the capture. 

NumLk
CapsLk
I did press the ScrLk button on the keyboard
Then I sent the xset command (pasted, so only Enter key was pressed on keyboard)
Ended Capture
[K95xt-Capture-20210503.zip](https://github.com/ckb-next/ckb-next/files/6418813/K95xt-Capture-20210503.zip)

tatokis, 2021-05-04:
The led indicators should work now

TheWebMachine, 2021-05-04:
> The led indicators should work now

Confirmed. They are working perfectly now. I'll give the backtrace on the polling rate change crash a try after I'm done working for the day.

TheWebMachine, 2021-05-09:
[gdb-ckb.txt](https://github.com/ckb-next/ckb-next/files/6446979/gdb-ckb.txt)

tatokis, 2021-05-09:
Other than the bug with the packet timeout, I don't see a crash anywhere in your backtrace.

TheWebMachine, 2021-05-09:
That's because the daemon completely locks up immediately after. Sig9 doesn't kill it; reboot times out waiting on it to die and forces through the reboot. I can't get anything else out of it once the timeout occurs.

tatokis, 2021-05-09:
Run continue after gdb picks up the SIGUSR2, and then once it locks up, press Ctrl C. It might give more info as to what's going on.

TheWebMachine, 2021-05-09:
[gdb-ckb-2.txt](https://github.com/ckb-next/ckb-next/files/6446991/gdb-ckb-2.txt)
Doesn't appear to be anything else of interest.

tatokis, 2021-05-09:
Ah, I see now. It looks like a deadlock. I also don't see any debug symbols from ckb itself. How did you build it?
Can you add `-DCMAKE_BUILD_TYPE=Debug -DSANITIZE_THREAD=1` to the cmake command and try again?

It should spit out information when the deadlock occurs.

TheWebMachine, 2021-05-09:
Sorry. I forgot I got switched back to Release in my build script when I pulled the fixes for mismapped keys. Debug flags added this time around...
[gdb-ckb-3.txt](https://github.com/ckb-next/ckb-next/files/6447035/gdb-ckb-3.txt)

tatokis, 2021-05-09:
Do you get anything from thread sanitizer if you run it without gdb?

TheWebMachine, 2021-05-09:
This is all I got when I ran it without gdb and trigger the crash:
````
# ./ckb-next-daemon 
ckb-next: Corsair RGB driver 0.4.3-75-g12f582d
[I] Root controller ready at /dev/input/ckb0
[I] Connecting CORSAIR K95 RGB PLATINUM XT Mechanical Gaming Keyboard at /dev/input/ckb1
[I] Starting input thread for /dev/input/ckb1
[I] Found EP 0x81 at /sys/devices/pci0000:00/0000:00:01.2/0000:02:00.0/0000:03:08.0/0000:08:00.3/usb3/3-2/3-2.2/3-2.2:1.0/ep_81
[I] Found EP 0x84 at /sys/devices/pci0000:00/0000:00:01.2/0000:02:00.0/0000:03:08.0/0000:08:00.3/usb3/3-2/3-2.2/3-2.2:1.1/ep_84
[I] Found EP 0x83 at /sys/devices/pci0000:00/0000:00:01.2/0000:02:00.0/0000:03:08.0/0000:08:00.3/usb3/3-2/3-2.2/3-2.2:1.2/ep_83
[I] Found EP 0x82 at /sys/devices/pci0000:00/0000:00:01.2/0000:02:00.0/0000:03:08.0/0000:08:00.3/usb3/3-2/3-2.2/3-2.2:1.3/ep_82
[I] Setup finished for /dev/input/ckb1
[I] Stopping input thread for /dev/input/ckb1
[W] os_usbrecv (via bragi_common.c:23): ckb1: Timeout while waiting for response
[I] Attempting reset...
[E] os_resetusb (via usb.c:650): resetusb failed: No such device
[E] usb_tryreset (usb.c:658): Reset failed. Disconnecting.
[I] Disconnecting /dev/input/ckb1
[I] Caught internal signal SIGUSR2
````
...then I had to forcefully kill it (which did let me SIGKILL this time).
I've never used ThreadSanitizer. Am I missing a step beyond the build flag you had me include?

Since this mentions issues resetting USB, allow me to illuminate some more on my system, in case it matters any:
ASUS ROG Crosshair VIII Formula MB...so _all_ the backpanel ports are USB3 (even the dedicated "ezFlash" BIOS port). The keyboard is plugged into a bog-standard Gen1 port (not Gen2 SS+). I'd have to go find a USB2 header unit to plug into the MB if I wanted to test that out, as my case doesn't even have any USB2 ports built into it.

Bitwolfies, 2021-09-11:
Is the branch still worth using? Seems like everyone was saying it was almost done and now, nothing.

tatokis, 2021-09-11:
The support for it is in master now, but requires `--enable-experimental` to be passed to ckb-next-daemon. One of the numpad keys still doesn't function, as I've been meaning to rewrite the input code now that I have more keyboards to compare to.

rrediske, 2021-09-13:
I enabled the --enable-experimental flag in the systemd unit file and I can verify that the daemon starts with that flag using ps, but I don't see Platinum XT keyboard support.

I am running version 0.4.4-1 and it still sees my Corsair Scimitar mouse just fine.

I'm anxious to at least get the keyboard lighting to work as every time I boot, I have to load up a Windows 10 VM, run the iCue application and pass through USB support just to turn the lights on so I can see the keys on my keyboard :-)￼

tatokis, 2021-09-13:
> I am running version 0.4.4-1

It is not in a release. It is in master. If you are on Ubuntu you can use the ckb-next-git PPA, on Arch you can use the ckb-next-git package from the AUR, or you can build it yourself using the instructions in the wiki.

rrediske, 2021-09-13:
I run Manjaro, so I removed the package and installed the git version, ckb-next 0.4.4-99-g761ba48d and it still doesn't show the keyboard in the gui. I had to restart the daemon manually, so I know it's running the new version. Both versions, this one and the package 0.4.4-1 do identify both devices using the ckb-next-dev-detect script.

FryDay, 2021-09-13:
@rrediske I am using the same version and it is working for me in Arch. What does `systemctl status ckb-next-daemon.service` output for you? This is mine for reference:

```
Sep 13 07:26:59 Pennywise systemd[1]: Started Corsair Keyboards and Mice Daemon.
Sep 13 07:26:59 Pennywise ckb-next-daemon[15185]: ckb-next: Corsair RGB driver 0.4.4-99-g761ba48d
Sep 13 07:26:59 Pennywise ckb-next-daemon[15185]: [I] Root controller ready at /dev/input/ckb0
Sep 13 07:27:00 Pennywise ckb-next-daemon[15185]: [I] Connecting CORSAIR K95 RGB PLATINUM XT Mechanical Gaming Keyboard at /dev/input/ckb1
Sep 13 07:27:00 Pennywise ckb-next-daemon[15185]: [I] Starting input thread for /dev/input/ckb1
Sep 13 07:27:00 Pennywise ckb-next-daemon[15185]: [I] Found EP 0x81 at /sys/devices/pci0000:00/0000:00:08.1/0000:2b:00.3/usb7/7-2/7-2:1.0/ep_81
Sep 13 07:27:00 Pennywise ckb-next-daemon[15185]: [I] Found EP 0x84 at /sys/devices/pci0000:00/0000:00:08.1/0000:2b:00.3/usb7/7-2/7-2:1.1/ep_84
Sep 13 07:27:00 Pennywise ckb-next-daemon[15185]: [I] Found EP 0x83 at /sys/devices/pci0000:00/0000:00:08.1/0000:2b:00.3/usb7/7-2/7-2:1.2/ep_83
Sep 13 07:27:00 Pennywise ckb-next-daemon[15185]: [I] Found EP 0x82 at /sys/devices/pci0000:00/0000:00:08.1/0000:2b:00.3/usb7/7-2/7-2:1.3/ep_82
Sep 13 07:27:00 Pennywise ckb-next-daemon[15185]: [I] Setup finished for /dev/input/ckb1
```

rrediske, 2021-09-13:
Huh, mine looks different, with a few errors to add more mystery:

```systemctl status ckb-next-daemon.service
● ckb-next-daemon.service - Corsair Keyboards and Mice Daemon
     Loaded: loaded (/usr/lib/systemd/system/ckb-next-daemon.service; disabled; vendor preset: disabled)
    Drop-In: /etc/systemd/system/ckb-next-daemon.service.d
             └─override.conf
     Active: active (running) since Mon 2021-09-13 10:07:37 CDT; 28min ago
   Main PID: 345352 (ckb-next-daemon)
      Tasks: 4 (limit: 38396)
     Memory: 1.5M
        CPU: 1.631s
     CGroup: /system.slice/ckb-next-daemon.service
             └─345352 /usr/bin/ckb-next-daemon

Sep 13 10:07:37 onyx ckb-next-daemon[345352]: [I] Root controller ready at /dev/input/ckb0
Sep 13 10:07:37 onyx ckb-next-daemon[345352]: [I] Connecting CORSAIR SCIMITAR RGB ELITE Gaming Mouse at /dev/input/ckb1
Sep 13 10:07:37 onyx ckb-next-daemon[345352]: [I] Starting input thread for /dev/input/ckb1
Sep 13 10:07:37 onyx ckb-next-daemon[345352]: [I] Found EP 0x81 at /sys/devices/pci0000:00/0000:00:01.3/0000:02:00.0/usb1/1-5/1-5:1.0/ep_81
Sep 13 10:07:37 onyx ckb-next-daemon[345352]: [I] Received identification packet: 0e 01 00 00 01 01 00 01 20 03 06 03 1c 1b 8b 1b 01 01 01 30 c1 ff 40 40 0>
Sep 13 10:07:37 onyx ckb-next-daemon[345352]: [I] Setup finished for /dev/input/ckb1
Sep 13 10:07:54 onyx ckb-next-daemon[345352]: [E] process_input_urb (keymap.c:623): Unknown mouse data received in input thread 10 from endpoint 81
Sep 13 10:07:54 onyx ckb-next-daemon[345352]: [E] process_input_urb (keymap.c:623): Unknown mouse data received in input thread 02 from endpoint 81
Sep 13 10:25:26 onyx ckb-next-daemon[345352]: [E] process_input_urb (keymap.c:623): Unknown mouse data received in input thread 10 from endpoint 81
Sep 13 10:25:26 onyx ckb-next-daemon[345352]: [E] process_input_urb (keymap.c:623): Unknown mouse data received in input thread 02 from endpoint 81```

FryDay, 2021-09-13:
It doesn't look like you are properly enabling experimental features:
```
CGroup: /system.slice/ckb-next-daemon.service
             └─345352 /usr/bin/ckb-next-daemon
```

Make sure /usr/lib/systemd/system/ckb-next-daemon.service looks like this:
```
# Copyright 2017-2018 ckb-next Development Team <ckb-next@googlegroups.com>
# Distributed under the terms of the GNU General Public License v2

[Unit]
Description=Corsair Keyboards and Mice Daemon

[Service]
ExecStart=/usr/bin/ckb-next-daemon --enable-experimental
Restart=on-failure

[Install]
WantedBy=multi-user.target
```

FryDay, 2021-09-13:
Are you overriding the service somehow?
```
Drop-In: /etc/systemd/system/ckb-next-daemon.service.d
             └─override.conf
```
Not sure what this is tbh.

rrediske, 2021-09-13:
I noticed that. I am not used to systemd, but I used systemctl edit ckb-next-daemon to add the --enable-experimental flag. It created an override.conf file and wouldn't load, so I deleted the file. But I see the file is back, although it's just 2 uncommented lines now.

Here's what I get from systemctl edit ckb-next-daemon:

```### Editing /etc/systemd/system/ckb-next-daemon.service.d/override.conf
### Anything between here and the comment below will become the new contents of the file

### Editing /etc/systemd/system/ckb-next-daemon.service.d/override.conf
### Anything between here and the comment below will become the new contents of the file

### Lines below this comment will be discarded

### /usr/lib/systemd/system/ckb-next-daemon.service
# # Copyright 2017-2018 ckb-next Development Team <ckb-next@googlegroups.com>
# # Distributed under the terms of the GNU General Public License v2
#
# [Unit]
# Description=Corsair Keyboards and Mice Daemon
#
# [Service]
# ExecStart=/usr/bin/ckb-next-daemon
# Restart=on-failure
#
# [Install]
# WantedBy=multi-user.target```

And here's what the override.conf file contains:

```### Editing /etc/systemd/system/ckb-next-daemon.service.d/override.conf
### Anything between here and the comment below will become the new contents of the file```

FryDay, 2021-09-13:
@rrediske It looks like override.conf got cut off in your last comment. Please repost it.

rrediske, 2021-09-13:
Ah, sure enough, the enable-experimental flag is not shown when I do a ps. I think I need to untangle this override.conf file out as well. When I edit the unit file for ckb-next-daemon, do I just add the flag to the commented code below? In my first attempt, I added it uncommented to the top of the file, which created the override.conf file and then didn't let the daemon load with the error:

```Sep 12 23:54:17 onyx systemd[1]: ckb-next-daemon.service: Service has more than one ExecStart= setting, which is only allowed for Type=oneshot services.

rrediske, 2021-09-13:
Actually, it didn't get cut off. It's only 2 lines long, both commented out.

FryDay, 2021-09-13:
Run the following and then restart the service:
`sudo rm -rf /etc/systemd/system/ckb-next-daemon.service.d && sudo sed -i '/ExecStart/ s/$/ --enable-experimental/' /usr/lib/systemd/system/ckb-next-daemon.service`

FryDay, 2021-09-13:
And if that still doesn't work, copy and paste your entire `/usr/lib/systemd/system/ckb-next-daemon.service` file.

rrediske, 2021-09-13:
That worked!!! OMG this will be so much nicer every time I reboot. Granted, I reboot much less in linux, but still.

 I apologize for my mix of newb and enough knowledge to be too dangerous, thank you very much!

TLDR

I see what happened last night. When I did the systemctl edit ckb-next-daemon, nano opened, I added the ExecStart line with the flag to the top of that unit file, and when I saved changes I didn't even notice that it added some gibberish to the override.conf file name:

/etc/systemd/system/ckb-next-daemon.service.d/.#override.confdbe789046adb1417

I think I edited the file incorrectly; I need to learn systemd.

lenonk, 2021-09-21:
After following the instructions to use --enable-experimental, this is working beautifully for me.  I've only found 2 bugs so far.  The 7 key on the number pad doesn't work, which I believe you know about, and setting the "on" color for "indicate mute:" does not save, and is always the transparent color.

Other than that, everything, including key binding, is working like a charm for me.

I assume that bugs with this particular keyboard should be reported in this thread as opposed to opening a new issue.  Please let me know if that assumption is incorrect.

Viktini, 2021-09-27:
Just did `--enable-experimental` as well on my Platinum XT (UK layout) and everything works!

Well, except for the backslash key. No light on it at all, and it seems to have inherited the functions on the 7 key on the number pad (probably related to above, worth mentioning anyway). The # key also has no lighting either.

tatokis, 2021-09-30:
Please try again the latest code from master. I have rewritten the input handling code, and this will either break everything, or fix the numpad 7 issue, as well as add support for European/ISO layouts. Please test it with both the GUI running and not running (do not SIGKILL it, please quit it normally, otherwise the keyboard will stay in software mode).
I'd suggest having a spare keyboard while testing, in case some keys get stuck pressed down and you have to unplug it, or input straight up doesn't work.

Additionally, please check if the RGB mapping is broken or if there are any more keys left that aren't lit properly. I have merged the keyboard and mouse RGB functions to make maintaining this easier, and I might have broken it in the process.

lenonk, 2021-09-30:
Ok, so, it works fine as long as the GUI is running.  The 7 ( <--- Typed from my numpad) works, all the other keys seem to work.  When the GUI is not running, the LEDs work, but no input works at all.

I fear I wasn't clear in my earlier post.  The actual light on the kb for "indicate mute" works just fine, it's just not represented properly in the GUI.  It's always the transparent color.  Sorry for any confusion.

tatokis, 2021-09-30:
@lenonk can you please test the input without the GUI running again? This should fix it https://github.com/ckb-next/ckb-next/commit/05512995c2fbbf6a8e8d9cd85c5b4b6403e56e9e

As for the indicate mute option, it's checking the default pulseaudio device, so if that remains muted on your system, it will always be in the transparent state. If it functions correctly with it unchecked, it has nothing to do with the keyboard and it's unrelated.

lenonk, 2021-09-30:
Yep.  Looks good now.  Thanks for all your work on this.

[18 more replies]

closed 2022-10-03 by tatokis