ckb-next/ckb-next #545: Have to unload/load ckb-next-daemon on every wake from sleep or boot to get it working

laxxity, 2020-02-18:
Since upgrading to Catalina (10.15.2) I found that the original ckb stopped working for my Corsair Sabre RGB mouse, so I upgraded to ckb-next. 

All was working fine until I rebooted or woke my Mac from sleep, after which my settings were not active and lighting effects were not working (just showed white LED).

I tried re-installing a couple of times and found that it worked after the install but again stopped after sleep/wake or reboot.

Checked that the ckb-daemon process is running:

    0 80663     1   0  8:27am ??         0:13.77 /Library/Application Support/ckb-next-daemon
  501 80670     1   0  8:27am ??         0:59.84 /Applications/ckb-next.app/Contents/MacOS/ckb-next
  501 80687 80670   0  8:27am ??         0:01.14 /Applications/ckb-next.app/Contents/Resources/ckb-next-animations/gradient --ckb-run
  501 80688 80670   0  8:27am ??         0:01.10 /Applications/ckb-next.app/Contents/Resources/ckb-next-animations/gradient --ckb-run
  501 80689 80670   0  8:27am ??         0:01.25 /Applications/ckb-next.app/Contents/Resources/ckb-next-animations/gradient --ckb-run

I then took a look at the scripts and found I could get it running again by unloading/loading the daemon with the following script I cobbled together :

#!/usr/bin/env bash

echo "Please enter your password to restart ckb-next-daemon"
sudo launchctl unload /Library/LaunchDaemons/org.ckb-next.daemon.plist
sudo launchctl load -w /Library/LaunchDaemons/org.ckb-next.daemon.plist
echo "Finished!"

Any idea why ckb-daemon isn't working after reboot / wake from sleep?

Thanks.

Attached ckb report.
[ckb-next-report-1582014705.gz](https://github.com/ckb-next/ckb-next/files/4218215/ckb-next-report-1582014705.gz)

Kristaphonie, 2020-02-27:
I'm having the same problem.

tatokis, 2020-03-01:
Looks like for some stupid reason it doesn't have permission to access the devices until you manually restart it.
I have no idea why, but it has to do with Catalina's annoying new permission system that isn't well thought out.

Try removing ckb-next-daemon from input monitoring and manually readding it using the + button.

Path should be `/Library/Application Support/ckb-next-daemon`

tatokis, 2020-03-01:
Give this new build also a try after completely uninstalling the old one using the uninstall button under settings.

http://ckb-next.info/mac/ckb-next_v0.4.2-41-gf4c9184_10.12-SDK.dmg

```
3e0ec5f41a6bcdcf07e767626b71cd5e73ecabd21c68d1caf871f80880c826c3  ckb-next_v0.4.2-41-gf4c9184_10.12-SDK.dmg
```

Kristaphonie, 2020-03-02:
So I tried the new build, but now my keyboard isn't connecting. The generated report is showing unknown errors and failure to open device. I'll post it here.

```
OS Version:
ProductName:	Mac OS X
ProductVersion:	10.15.3
BuildVersion:	19D76

Daemon log:
[E] os_usbsend (via device_keyboard.c:40): Got return value 0xe00002e2
[E] os_usbsend (via device_keyboard.c:40): Got return value 0xe00002e2
[E] os_usbsend (via device_keyboard.c:40): Got return value 0xe00002e2
[E] os_usbsend (via device_keyboard.c:40): Got return value 0xe00002e2
[E] os_usbsend (via device_keyboard.c:40): Got return value 0xe00002e2
[E] os_usbsend (via device_keyboard.c:40): Got return value 0xe00002e2
[E] os_usbsend (via device_keyboard.c:40): Got return value 0xe00002e2
[E] os_usbsend (via device_keyboard.c:40): Got return value 0xe00002e2
[E] os_usbsend (via device_keyboard.c:40): Got return value 0xe00002e2
[E] os_usbsend (via device_keyboard.c:40): Got return value 0xe00002e2
[E] os_usbsend (via device_keyboard.c:40): Got return value 0xe00002e2
[E] os_usbsend (via device_keyboard.c:40): Got return value 0xe00002e2
[E] os_usbsend (via device_keyboard.c:40): Got return value 0xe00002e2
[E] os_usbsend (via device_keyboard.c:40): Got return value 0xe00002e2
[E] os_usbsend (via device_keyboard.c:40): Got return value 0xe00002e2
[E] os_usbsend (via device_keyboard.c:40): Got return value 0xe00002e2
[E] os_usbsend (via device_keyboard.c:40): Got return value 0xe00002e2
[E] os_usbsend (via device_keyboard.c:40): Got return value 0xe00002e2
[E] os_usbsend (via device_keyboard.c:40): Got return value 0xe00002e2
[E] os_usbsend (via device_keyboard.c:40): Got return value 0xe00002e2
[E] os_usbsend (via device_keyboard.c:40): Got return value 0xe00002e2
[E] os_usbsend (via device_keyboard.c:40): Got return value 0xe00002e2
[E] os_usbsend (via device_keyboard.c:40): Got return value 0xe00002e2
[E] os_usbsend (via device_keyboard.c:40): Got return value 0xe00002e2
[E] os_usbsend (via device_keyboard.c:40): Got return value 0xe00002e2
[E] os_usbsend (via device_keyboard.c:40): Got return value 0xe00002e2
[E] os_usbsend (via device_keyboard.c:40): Got return value 0xe00002e2
[E] os_usbsend (via device_keyboard.c:40): Got return value 0xe00002e2
[E] os_usbsend (via device_keyboard.c:40): Got return value 0xe00002e2
[E] os_usbsend (via device_keyboard.c:40): Got return value 0xe00002e2
[E] os_usbsend (via device_keyboard.c:40): Got return value 0xe00002e2
[E] os_usbsend (via device_keyboard.c:40): Got return value 0xe00002e2
[E] os_usbsend (via device_keyboard.c:40): Got return value 0xe00002e2
[E] os_usbsend (via device_keyboard.c:40): Got return value 0xe00002e2
[E] os_usbsend (via device_keyboard.c:40): Got return value 0xe00002e2
[E] os_usbsend (via device_keyboard.c:40): Got return value 0xe00002e2
[E] os_usbsend (via device_keyboard.c:40): Got return value 0xe00002e2
[E] os_usbsend (via device_keyboard.c:40): Got return value 0xe00002e2
[E] os_usbsend (via device_keyboard.c:40): Got return value 0xe00002e2
[E] os_usbsend (via device_keyboard.c:40): Got return value 0xe00002e2
[E] os_usbsend (via device_keyboard.c:40): Got return value 0xe00002e2
[E] os_usbsend (via device_keyboard.c:40): Got return value 0xe00002e2
[E] os_usbsend (via device_keyboard.c:40): Got return value 0xe00002e2
[E] os_usbsend (via device_keyboard.c:40): Got return value 0xe00002e2
[E] os_usbsend (via device_keyboard.c:40): Got return value 0xe00002e2
[E] os_usbsend (via device_keyboard.c:40): Got return value 0xe00002e2
[E] os_usbsend (via device_keyboard.c:40): Got return value 0xe00002e2
[E] os_usbsend (via device_keyboard.c:40): Got return value 0xe00002e2
[E] os_usbsend (via device_keyboard.c:40): Got return value 0xe00002e2
[E] os_usbsend (via device_keyboard.c:40): Got return value 0xe00002e2
[E] os_usbsend (via device_keyboard.c:40): Got return value 0xe00002e2
[E] os_usbsend (via device_keyboard.c:40): Got return value 0xe00002e2
[E] os_usbsend (via device_keyboard.c:40): Got return value 0xe00002e2
[E] os_usbsend (via device_keyboard.c:40): Got return value 0xe00002e2
[E] os_usbsend (via device_keyboard.c:40): Got return value 0xe00002e2
[E] os_usbsend (via device_keyboard.c:40): Got return value 0xe00002e2
[E] os_usbsend (via device_keyboard.c:40): Got return value 0xe00002e2
[E] os_usbsend (via device_keyboard.c:40): Got return value 0xe00002e2
[E] os_usbsend (via device_keyboard.c:40): Got return value 0xe00002e2
[E] os_usbsend (via device_keyboard.c:40): Got return value 0xe00002e2
[E] os_usbsend (via device_keyboard.c:40): Got return value 0xe00002e2
[... 290 lines cut ...]
Launchctl start returned 0
HID agent encountered an unknown error
Launchctl start returned 0
HID agent encountered an unknown error
Launchctl start returned 0
HID agent encountered an unknown error
Launchctl start returned 0
HID agent encountered an unknown error
Launchctl start returned 0
HID agent encountered an unknown error
Launchctl start returned 0
HID agent encountered an unknown error
Launchctl start returned 0
HID agent encountered an unknown error
Launchctl start returned 0
HID agent encountered an unknown error
Launchctl start returned 0
HID agent encountered an unknown error
Launchctl start returned 0
HID agent encountered an unknown error
Launchctl start returned 0
HID agent encountered an unknown error
Launchctl start returned 0
HID agent encountered an unknown error
```

SwiftieH, 2020-05-16:
I'm having the same problem with Catalina 10.15.3. I try to restart ckb-next-daemon each time awake from sleep and it seems like working fine. The command for restarting ckb-next-daemon is:
`/Applications/ckb-next.app/Contents/Resources/daemon-restart.sh ; exit;`
Hopefully it would help.

audreyseo, 2020-12-23:
Nearly identical log as @Kristaphonie. 
[ckb-next-report-1608745213.gz](https://github.com/ckb-next/ckb-next/files/5736471/ckb-next-report-1608745213.gz)

I tried using the daemon-restart.sh script, which momentarily made the keyboard tab open, but then it closes about 5 seconds afterwards.