ckb-next/ckb-next #548: A pipe output for external drivers

TNZ9, 2020-03-01:
The only way to use a custom RGB device with ckb-next is to glitch the daemon by creating a fake RGB device known by the ckb daemon.

I'd like the user to be able to set some pipe devices up to allow an external driver (maybe user-written) to recieve instructions from the ckb daemon.

ZirconiumX, 2020-03-01:
Non-Corsair devices are fundamentally out of scope, because ckb is far too
monolithic to support the wide variety of possible ways to drive things. It
needs a complete rewrite and we do not have the energy for that.

>

TNZ9, 2020-03-01:
Ok, thank you a lot for the fast awnser, i'll look for another way to do it.

ZirconiumX, 2020-03-01:
The animations are not actually that complex; you could probably implement
them in a scripting language that calls your external driver.

tatokis, 2020-03-01:
You can also most likely make the GUI connect to another pipe of your choice with minimal hacking.
On the windows branch I made this class and then made the rest of the code use it. It could be a good starting point as all pipe access happens through it.

https://github.com/ckb-next/ckb-next/blob/windows/src/gui/daemonpipe.cpp
https://github.com/ckb-next/ckb-next/blob/windows/src/gui/daemonpipe.h