ckb-next/ckb-next #550: ckb-next-daemon disables keys, please help

prankousky, 2020-03-03:
Hi everybody,

I am running `ckb-next: Corsair RGB driver 0.4.2-28-g26dd69b`. Installed via `AUR`.

Just a few days ago, things started not working as expected any longer. Previously, everything worked, so I am not quite sure what could have caused this.

Currently, the keyboard is not displaying my custom color theme, but an animated rainbow. **Not all** shortcuts created in `ckb-next` work any longer. My keyboard has multimedia buttons. Those work fine! However, I have `<esc>:wq` as a shortcut, which will not execute when pressed.

However! When I run `sudo ckb-next-daemon`, the :wq shortcut **will** work again. **But** now my Windows key (`mod4`) will not work. I am no longer able to execute any of my `i3` shortcuts. On top, not even things like `ctrl+q` work. When I `sudo pkill ckb-next-daemon`, this will reverse: mod4 works, ctrl+<anything> works, **but** :wq does not work any longer.

I used to encounter this problem before, but would "simply" unplug my keyboard and plug it back in. Now I have to chose between being able to use my regular keys **or** custom color schemes and mapped buttons. Which I really do not understand.

What can I do to fix this? Thank you for your support :)

- Search in open and closed [Issues](https://github.com/ckb-next/ckb-next/issues).
- If possible, try the latest code by building the software manually.
- File unrelated issues separately.

**Click the `Generate Report` button in the `Settings` tab of the ckb-next user interface and attach the generated report to this issue.**

## General information

### Source
How and where did you get this program, e.g.:
  * Package from AUR/PPA/COPR/...
  * Built manually using quickinstall script
  * Pre-built dmg for macOS

## Logs & crash reports

Attach:
 * The generated report from the section above
 * Any additional crash reports you've managed to gather

## What is the program's current behaviour?

## What is the expected behaviour of the program?

## Any other notes & comments?

ZirconiumX, 2020-03-04:
Did you accidentally press the windows lock? Might be a stupid question but it doesn't hurt to check.

prankousky, 2020-03-04:
This might be a stupid question, but where would I have pressed the windows lock? I cannot find it anywhere in the settings... On `Modifier keys`, I got `Windows: Super`, `Ctrl: Control` and `Alt: Alt`. Any combination using the alt key works, so super and ctrl should work, too..?

closed 2020-03-04 by prankousky

jaclas, 2021-04-24:
> UPDATE 2: I think I got it... The little lock symbol on the top left button bar. Thank you! Solved and closed :)

I've encountered the same problem and can't figure it out, where is this blocker you write about? I can't find it.

prankousky, 2021-04-24:
I'm talking about the button here (don't mind the dust on my keyboard).

[image]

jaclas, 2021-04-24:
Shit...! 
I was looking in the ckb.next settings.... stupid me :-)

Big thanks, now life is easier :-)