ckb-next/ckb-next #551: Volume up and down key lights doesn't work

mauriciofauth, 2020-03-04:
## General information

K68 NRGB

### Source

* Built manually using quickinstall script

## What is the program's current behaviour?

The backlight of the Volume up and Volume down keys doesn't work.

## Any other notes & comments?

Works with v0.4.1, but doesn't work with v0.4.2.

tatokis, 2020-03-04:
Yeah, sorry about that, I screwed it up before the 0.4.2 release, but it's been fixed in master for a while now.
https://github.com/ckb-next/ckb-next/commit/7bbc4cfe7875a7e25fb959bc09a621c82a870b7b

tatokis, 2020-03-04:
...looking at that again, I'm not sure if it fixes anything, or why it would fix anything.

Can you please test the latest code from master?

mauriciofauth, 2020-03-04:
> Can you please test the latest code from master?
> 
> EDIT: Nevermind, I see now. It does fix it.

I just tested with master and it does fix it.
Hoping for a release with the fix soon.
Thank you.

4g0tt3nSou1, 2020-07-06:
Hi, Just asking about status of release containing the patch that fixes this. For obvious reasons (Too lazy srry.) I am waiting for a release rather than compiling from master. It's been some time so I figured a "bump" was in order :)