ckb-next/ckb-next #555: Failed to build.  Unsure what is useful to include here.

hydranmenace, 2020-03-16:
I searched as best I knew how for my issue.  I do ./quickinstall and it gets roughly halfway done and then I see errors.  Not sure what logs to attach, so I guess I'll just paste in the output from ./quickinstall.  Any help would be appreciated.  

hydran@hydramax ~/Downloads/ckb-next $ ./quickinstall
-- The C compiler identification is GNU 8.2.0
-- The CXX compiler identification is GNU 8.2.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found Git: /usr/bin/git (found version "2.24.1") 
-- ckb-next version: 0.4.2-50-gafd0e19 (Non-release)
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Check if compiler accepts -pthread
-- Check if compiler accepts -pthread - yes
-- Found Threads: TRUE  
-- Performing Test ICONV_SECOND_ARGUMENT_IS_CONST
-- Performing Test ICONV_SECOND_ARGUMENT_IS_CONST - Failed
-- Found Iconv: /usr/lib/libc.so
-- Checking for module 'libudev'
--   Found libudev, version 242
-- Found UDEV: /usr/lib/libudev.so  
-- Searching for running ckb and/or ckb-next GUI
-- NOTE:
    Privileged access is required for operations upon the daemon at
    configure time as it is owned by root. If this is unacceptable,
    disable SAFE_INSTALL and prepare the filesystem manually.
-- systemd detected
-- Generating and importing ckb-next-daemon.service (systemd)
Password: 
inactive
Failed to get unit file state for ckb-daemon.service: No such file or directory
inactive
Failed to get unit file state for ckb-next-daemon.service: No such file or directory
-- Found QuaZip: /usr/lib/libquazip5.so  
-- Found PulseAudioSimple: /usr/lib/libpulse-simple.so  
-- Found PulseAudio: /usr/lib/libpulse.so  
-- Found ZLIB: /usr/lib/libz.so (found version "1.2.11") 
-- Found X11: /usr/include   
-- Looking for XOpenDisplay in /usr/lib/libX11.so;/usr/lib/libXext.so
-- Looking for XOpenDisplay in /usr/lib/libX11.so;/usr/lib/libXext.so - found
-- Looking for gethostbyname
-- Looking for gethostbyname - found
-- Looking for connect
-- Looking for connect - found
-- Looking for remove
-- Looking for remove - found
-- Looking for shmat
-- Looking for shmat - found
-- Looking for IceConnectionNumber in ICE
-- Looking for IceConnectionNumber in ICE - found
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.2") 
-- Checking for module 'xcb'
--   Found xcb, version 1.13.1
-- Checking for module 'xcb-ewmh'
--   Found xcb-ewmh, version 0.4.1
-- Checking for module 'xcb-screensaver'
--   Found xcb-screensaver, version 1.13.1
-- Generating binary.qrc
-- Importing ckb-next-key.gpg
[... 110 lines cut ...]
[ 36%] Linking C executable ../../bin/snake
[ 37%] Linking C executable ../../bin/life
[ 38%] Building C object src/daemon/CMakeFiles/ckb-next-daemon.dir/led_keyboard.c.o
[ 39%] Linking C executable ../../../bin/mviz
[ 39%] Built target life
[ 39%] Built target snake
[ 40%] Building C object src/daemon/CMakeFiles/ckb-next-daemon.dir/led_mouse.c.o
[ 41%] Building C object src/daemon/CMakeFiles/ckb-next-daemon.dir/led_mousepad.c.o
[ 42%] Building C object src/daemon/CMakeFiles/ckb-next-daemon.dir/main.c.o
[ 42%] Built target mviz
[ 42%] Building C object src/daemon/CMakeFiles/ckb-next-daemon.dir/notify.c.o
[ 43%] Building C object src/daemon/CMakeFiles/ckb-next-daemon.dir/profile.c.o
[ 44%] Building C object src/daemon/CMakeFiles/ckb-next-daemon.dir/profile_keyboard.c.o
[ 45%] Building C object src/daemon/CMakeFiles/ckb-next-daemon.dir/usb.c.o
[ 46%] Building C object src/daemon/CMakeFiles/ckb-next-daemon.dir/profile_mouse.c.o
[ 47%] Building C object src/daemon/CMakeFiles/ckb-next-daemon.dir/usb_linux.c.o
gmake[2]: *** No rule to make target '/usr/lib/libudev.so', needed by 'bin/ckb-next-daemon'.  Stop.
gmake[2]: *** Waiting for unfinished jobs....
[ 48%] Building C object src/daemon/CMakeFiles/ckb-next-daemon.dir/input_linux.c.o
gmake[1]: *** [CMakeFiles/Makefile2:225: src/daemon/CMakeFiles/ckb-next-daemon.dir/all] Error 2
gmake[1]: *** Waiting for unfinished jobs....
[ 48%] Built target ckb-next_autogen
gmake: *** [Makefile:152: all] Error 2
An error occurred,
press enter to exit.

tatokis, 2020-03-16:
What distro is this?

` gmake[2]: *** No rule to make target '/usr/lib/libudev.so', needed by 'bin/ckb-next-daemon'. Stop.`
Does `/usr/lib/libudev.so` exist?

ZirconiumX, 2020-03-16:
You need to install libudev; it's normally a package like libudev-dev on
Debian derivatives.

>

hydranmenace, 2020-03-16:
Sabayon Linux. Gentoo based. I thought I had libudev because toward the beginning of the output it says 
-- Found Iconv: /usr/lib/libc.so
-- Checking for module 'libudev'
-- Found libudev, version 242
-- Found UDEV: /usr/lib/libudev.so
-- Searching for running ckb and/or ckb-next GUI

and I have 
hydran@hydramax /usr/lib $ ls *udev*
libgudev-1.0.so  libgudev-1.0.so.0  libgudev-1.0.so.0.2.0  libudev.so  libudev.so.0

hydranmenace, 2020-03-16:
I made it work?  Sorry I am really new to linux.  Sort of.  I try it again every several years and things seem to change a lot.  It would seem that /usr/lib/libudev.so is a symlink that wasn't actually pointing to anything?  After some newbi-googling i did ln -s libudev.so.1.6.14 libudev.so which may or may not have been a horrible mistake.... but it did build and install.  Seems to be working fine.  Thanks to those who replied.  If my "fix" is a terrible idea, please let me know.

tatokis, 2020-03-16:
Yeah, that sounds about right. Somehow your system had a broken libudev.
In theory, reinstalling it should fix the symlinks. I don't know, I've never used your distro.

Either way, good to hear you got it working.

closed 2020-03-16 by tatokis