ckb-next/ckb-next #557: WASD keys are always white

descendingnerdliness, 2020-03-21:
K70RGB
While this seems to be a gaming feature, I find it disturbing that the WASD keys are always white.
The color seems to be changeable, but they never follow the rest of the keys in color changes.  
It would be vastly preferable, to me at least, if the  WASD keys follow the animations or are the same as the rest of the keys.

ashirviskas, 2020-03-28:
@descendingnerdliness it's just the default profile. Just go there, delete it and create a one yourself to be whatever you want it to be.

descendingnerdliness, 2020-04-05:
Thanks for the reply and this driver. 
I found that animation and all of the keys could be selected and then the 'Set keys' button pressed to have them all behave the same.
Thanks.

closed 2020-04-05 by descendingnerdliness