ckb-next/ckb-next #558: Support for CORSAIR IRONCLAW RGB Wireless Mouse

MordicusEtCubitus, 2020-03-22:
Here is the report.

Thank for helping.

[ckb-next-dev-detect-report.gz](https://github.com/ckb-next/ckb-next/files/4365264/ckb-next-dev-detect-report.gz)

Here is the product:
https://www.corsair.com/us/en/Categories/Products/Gaming-Mice/Wireless-Gaming-Mice/IRONCLAW-RGB-WIRELESS-Gaming-Mouse/p/CH-9317011-NA

And I am using Linux kernel 5: 5.3.0-42

Is it supported ?

Thanks a lot

MordicusEtCubitus, 2020-03-22:
I have built the kernel from sources, but it is not recognized and I have the driver version mismatch (0.4.2) warning.

I have rebooted, but still not recognized

[image]

MordicusEtCubitus, 2020-03-22:
I have connecting using cable and wireless dongle: same issue

thespyboss, 2020-03-22:
@MordicusEtCubitus It seems to not be supported. You can use this link to check though -> https://github.com/ckb-next/ckb-next/wiki/Supported-Hardware  I don't know if the devs are working on support for it but look through the pull requests. If you could, put the new device label on this issue. Thanks!

MordicusEtCubitus, 2020-03-22:
I don't know how to add the label : new-device ? I cannot edit/click on label

MordicusEtCubitus, 2020-03-22:
Device detection run as root: much more interesting...

[ckb-next-dev-detect-report.gz](https://github.com/ckb-next/ckb-next/files/4365628/ckb-next-dev-detect-report.gz)

MordicusEtCubitus, 2020-03-22:
I have added comments in #387 to better explain issue and my first step to get the gui appearing...

MordicusEtCubitus, 2020-04-07:
As my device is recognized with number 1b66 (see lsusb above)  I have update usb.h like this:
```C
// #define P_IRONCLAW           0x1b5d
#define P_IRONCLAW           0x1b66
```

Then recompiled using quickinstall with wireless branch of project:

I can now see the device in the gui, but I loose mouse control and daemon says:

```bash
$ sudo ckb-next-daemon 
ckb-next: Corsair RGB driver 0.4.2-22-g38d6023
[I] Root controller ready at /dev/input/ckb0
[I] Connecting CORSAIR IRONCLAW RGB WIRELESS Gaming Dongle at /dev/input/ckb1
[I] Starting input thread for /dev/input/ckb1
[W] os_inputmain (usb_linux.c:369): Unable to read wMaxPacketSize for /ep_81, assuming 64
[W] os_inputmain (usb_linux.c:369): Unable to read wMaxPacketSize for /ep_82, assuming 64
[W] os_usbrecv (via firmware.c:24): ckb1: Timeout while waiting for response
[W] _start_dev (device.c:43): Unable to load firmware version/poll rate
[I] Setup finished for /dev/input/ckb1
^C
[I] Caught signal 2
[I] Disconnecting /dev/input/ckb1
[I] Stopping input thread for /dev/input/ckb1
[I] Removed device path /dev/input/ckb1
[I] Closing root controller
[I] Removed device path /dev/input/ckb0
```

My kernel is:

```
$ uname -a
Linux  5.3.0-45-generic #37~18.04.1-Ubuntu SMP Fri Mar 27 15:58:10 UTC 2020 x86_64 x86
```

MordicusEtCubitus, 2020-04-07:
As updated in title : I'm ready to offer a gift of €100 if this mouse is supported !

Thanks !

MordicusEtCubitus, 2020-05-15:
Thanks for keeping this issue active.

So, if you think you can do it:
- I push the bounty up to €500 if solved in next 6 months
- I can send you a wireless mouse if you don't have it yet
- You will have to explain how I will pay you if you are many people to work on this issue : how to split the bounty ?

On my side, I've tried to negotiate with Corsair that they help this project by publishing ICue source code or by publishing their wireless protocol, or contribute to this project.
They answered politely to my messages but always rejected my requests.
I think they don't worth that Linux users invest in their hardware as they do not play the game.

Nevertheless, I would like to support your efforts and use this mouse fully on linux now that I own it.

A funny point for Linux users will be that this software becomes better than original ICUE, and that the License prevents from Corsair to use it in their proprietary software. Just to put the shame on their company.

tatokis, 2020-05-15:
It is a lot of work, as it's a whole new protocol from scratch, which is why it hasn't been done yet.

Your bounty is being considered, but at the moment, @ZirconiumX and I are both swamped with coursework. It will probably be picked up by at least one of us in the near future.

Apologies for the prolonged silence in this thread.

ZirconiumX, 2020-05-15:
I've set a 4-week reminder for this, and I'd like my portion of the bounty - whatever it ends up being - to go to the [Teenage Cancer Trust](https://www.teenagecancertrust.org/). But there's a *lot* of work needed here, though one could argue it had to be done at some point.

fierogtfan, 2020-06-17:
I asked on one of the related requests that was closed in preference of this issue, so here it is again.  What steps or aspects of the code base are recommended that someone new to the project follow to help contribute getting this device added?

MordicusEtCubitus, 2020-06-17:
While waiting you give a kick start to this request, here is a proof of my good faith: a first donation made.
[image]

ZirconiumX, 2020-06-17:
So, here's my understanding so far. @audiohacked mentioned he wanted to take the bounty, and I'd rather not treat on the feet of others. I've asked him to commit his changes, whatever they are, and we can go from there.

Still, thank you for that, and even if @audiohacked takes the bounty I'd be willing to chip in the £100 that you donated to the TCT.

MordicusEtCubitus, 2020-06-17:
No issue for me. £100 is the proof that the bounty is not only words.  
I will give the full bounty €500 to @audiohacked if this is your choice.

Thanks for working on it.

ZirconiumX, 2020-06-17:
Oh, and @fierogtfan just so you know, I'm not ignoring you. I suggest you join us in the #ckb-next IRC channel (it's the IRC button in the readme); it's a bit easier to discuss and distribute tasks there compared to on GitHub.

audiohacked, 2020-06-17:
Just to show I have been working on it a bit: https://github.com/audiohacked/ckb-next/tree/feature/ironclaw_wireless

rporter220, 2020-07-16:
Is this the same protocol that's needed for the K95 Platinum XT?  If so I'll throw an additional $100 into the bounty. 
My issue on new protocol/keyboard #543

chetwisniewski, 2020-07-22:
I'd pitch in $100 as well to author or charity

mijofa, 2020-07-23:
> Just to show I have been working on it a bit: https://github.com/audiohacked/ckb-next/tree/feature/ironclaw_wireless

I'm struggling to make this work for me, the mouse works fine as a normal mouse until I fire up ckb-next then it starts flashing the indicator lights between blue, red, and white, and it doesn't work as a mouse anymore.
I'd like to get in and help make my mouse work, but I'm unfamiliar with the project at all, so just looking for a bit of a starting point to work with.

Electro707, 2020-07-25:
I have also been working on a fork to include this mouse: [https://github.com/Electro707/ckb-next/](https://github.com/Electro707/ckb-next/) 
Although mine wasn't working (until I saw what @audiohacked changed and saw it was the V3 override that I missed being the reason what it wasn't showing up on my GUI, so thanks for that :) ).

More importantly tough I have been doing some USB sniffing and analyzing the packets being sent by Icue, and I have posted everything in this repository [https://github.com/Electro707/Ironclaw_Wireless_Reverse_Engineering](https://github.com/Electro707/Ironclaw_Wireless_Reverse_Engineering)
So far I have figured out how to get a firmware version, change the DPI settings, and changing the LED colors (on the logo, scroll wheel, and the front individually). I haven't officially tested them tough. 
The repository also includes Wireshark snippets in case anybody wants to go thru them (currently trying to do that with initialization of the mouse thru Icue)

mijofa, 2020-07-26:
@Electro707, I tried your ironclaw_wireless branch yesterday and found it didn't recognise the wireless dongle's VID/PID, once I copied those in ([patch attached](https://github.com/ckb-next/ckb-next/files/4977075/Electro707-ironclaw_wireless-dongle.txt)
) it recognised the device, but dmesg was going a bit nuts with messages like this when the ckb-next GUI was active:

    [ 2752.963981] hid-generic 0003:1B1C:1B66.0026: hiddev1,hidraw2: USB HID v1.11 Device [Corsair CORSAIR IRONCLAW RGB WIRELESS Gaming Dongle] on usb-0000:00:1d.0-1.6/input2
    [ 2752.966195] usb 3-1.6: usbfs: process 9428 (ckb1 input) did not claim interface 3 before use
    [ 2752.966230] usb 3-1.6: usbfs: process 9428 (ckb1 input) did not claim interface 2 before use
    [ 2753.192900] usb 3-1.6: usbfs: process 9428 (ckb1 input) did not claim interface 2 before use
    [ 2753.276831] usb 3-1.6: reset full-speed USB device number 8 using ehci-pci
    [ 2753.391581] hid-generic 0003:1B1C:1B66.0027: hiddev1,hidraw2: USB HID v1.11 Device [Corsair CORSAIR IRONCLAW RGB WIRELESS Gaming Dongle] on usb-0000:00:1d.0-1.6/input2

And it won't even work as a standard mouse. I didn't tried wired, mostly because wireless is too important a feature to me to give up. Can do some testing if it's helpful though.

Electro707, 2020-07-26:
@mijofa I haven't added the wireless USB VID/PID yet, as I figured I should test it out with wired before I move on to the wireless dongle (Also the fact I use it wired on my desktop)

Electro707, 2020-07-27:
Update: I finally was able to change the LEDs on the mouse. Here is the video of it: [https://youtu.be/vUglFVp9CuM](https://youtu.be/vUglFVp9CuM) (Sorry for the quality, my phone doesn't have the best camera)

I did this by having a python program send the same packets captured from initializing the mouse thru Icue, then sending LED data thru. 
After that, i've started to have the Python program ignore some packets to be sent, and thus pinning down closer as to what is needed and what is not (basically Muntzing-ing the packets sent from Icue to the mouse) 

If anybody wants to try it out, clone my reverse engineering repo ([here](https://github.com/Electro707/Ironclaw_Wireless_Reverse_Engineering)), run *send_from_file.py*, then select *IronclawWireless_Startup_2.json* as the packets when the file dialog pops up (In the *Extracted JSON* folder). If you have the wireless version of it, you could change the PID on line 19 of the Python program. If it's successful, the mouse should blink between yellow and light-blue 10 times. 
**WARNING: THE PROGRAM MAY SOMEHOW BREAK YOUR MOUSE, AND THUS I AM NOT LIABLE FOR ANY DAMAGES DONE TO IT** (sorry, obligatory warning)

tatokis, 2020-07-27:
Can someone provide a wireshark capture of the mouse being plugged in (wired) with CUE already running?
As in, start capture, start CUE, plug mouse in, stop capture once the mouse shows up.

tatokis, 2020-07-28:
In the meantime, also try the code from the [bragi](https://github.com/ckb-next/ckb-next/tree/bragi) branch in wired mode.
It should have working RGB (excluding the DPI indicator) and pollrate switching.
I have no idea if mouse input works, so please also paste any output from the daemon.

mijofa, 2020-07-28:
Windows is sadly not working for me at the moment, so I can't get wireshark captures. I have tested that bragi branch though.

The mouse input *kinda* worked, it was very slow/jumpy and every now and then seemed to just reset the mouse position back to the top right of my screen. I couldn't find anywhere to configure the pollrate, but the slow/jumpiness could be that.
The RGB didn't seem to work, the default wave animation wasn't playing on the mouse when the GUI said it was anyway.

Here's the output from the ckb-next-daemon: [ckb-next-daemon.txt](https://github.com/ckb-next/ckb-next/files/4985567/ckb-next-daemon.txt)

tatokis, 2020-07-28:
Can you do a wireshark capture on linux (with usbmon) with the GUI closed?
Start capture, start daemon, see if mouse can move the cursor.
Then run `echo active > /dev/input/ckb1/cmd`.

If it fails, you can stop the capture and upload it.
If not, try sending some rgb data to it. `echo "rgb ff00ff" > /dev/input/ckb1/cmd`
Wired only for now.

Electro707, 2020-07-28:
@tatokis I have captured packets from Icue after the mouse has been plugged in, and I posted the capture [here!](https://github.com/Electro707/Ironclaw_Wireless_Reverse_Engineering/blob/master/Wireshark%20Captures/IronclawWireless_Startup_4.pcapng) There are other captures in the repo where I captured data with the mouse already plugged in but Icue was started (although I expect them to be the same in terms of packets sent out).
I also did run the bragi branch, and I have the same comments as @mijofa where the mouse response is slow and choppy. Here is my ckb-next-daemon output:
[CaptureIronclawWireless2020-7-28.txt](https://github.com/ckb-next/ckb-next/files/4986350/CaptureIronclawWireless2020-7-28.txt)
And here is my Wireshark capture with ckb-next-deamon: 
[capture_ckb_next_deamon.zip](https://github.com/ckb-next/ckb-next/files/4986402/capture_ckb_next_deamon.zip)

tatokis, 2020-07-30:
Is the mouse choppy before you run the ckb-next gui as well?

Electro707, 2020-07-31:
@tatokis It's only choppy when the LEDs are running. The mouse becomes responsive if the brightness is set to off (assuming it shuts off whatever LED thread is running). So I'm assuming it's in the way the LEDs are handled that slows down the program.

tatokis, 2020-08-04:
The jerky cursor movement is caused by the fact that we have to read a response for every single RGB packet we send.
I'll need to figure out a workaround. This happens with other mice as well, but we only read from them during init, so it only happens for a very short period of time.

I think I have also figured out how RGB lighting works, so can you please try the latest code from the branch? https://github.com/ckb-next/ckb-next/commit/5559935ed485f8485c14e2132e701c0689aaae5a

Electro707, 2020-08-07:
@tatokis I have tried out the last branch, and the initialization command doesn't work (in the sense of the software still isn't able to control the mouse's LED). 
A potential solution to the LED response issue is to have a flag and a buffer, where the flag would be set by the LED thread and it will indicate to the USB receiver thread to add data to a buffer and clear the flag (Don't know how well it would work, just an idea).

mijofa, 2020-08-07:
I just tested as well and am gettting the same results as last time, none of the LEDs turn on, and chopping input handling (which was still expected)

tatokis, 2020-08-07:
Please provide logs

tatokis, 2020-08-07:
Additionally, after sending the logs (to see what the device was complaining about), try the latest code. I think I see the problem
https://github.com/ckb-next/ckb-next/commit/4c05a928277b21a6b9793ba27ac4e780722c9871

Electro707, 2020-08-08:
I have edited my previous comment to include the logs for that branch.
Trying out 4c05a92, the software is able to control the LEDs :)

Electro707, 2020-08-08:
Here is a bug I found: When the `ckb-next-daemon` daemon is loaded, scrolling up or down will result in mouse movements to the right. I have attached the logs for when that happens. 

[ScrollWheel-Mouse-Logs.txt](https://github.com/ckb-next/ckb-next/files/5046574/temp_2020-8-8-7-31-25.txt)

Ravenslofty, 2020-08-08:
That probably just means that directions need to be mapped differently, then.

tatokis, 2020-08-08:
Does the cursor move when you scroll? Does it happen if the GUI is running or not? (Make sure to close it with the Quit button in settings.)

Electro707, 2020-08-09:
Yes the cursor moves to the right (scrolling up will make it move father than scrolling down). This only happens when the GUI is loaded.

Electro707, 2020-08-09:
As an update, I copied over the bragi protocol stuff over to my forked version and added the drawing for the ironclaw wireless (should be a 1-1 with the original model, as I traced over a picture of the mouse). Still got to move around and fix the button's placement (there isn't an easy correlation between the x-y coordinate of the keys and that of the svg's size).
[image]

tatokis, 2020-08-19:
@Electro707 I have rebased and force pushed to the bragi branch.
The lag issue should be fixed, and the mouse should now be usable while the GUI is running.

However, I do need the output of the daemon with DEBUG_USB_INPUT=1 while the GUI is running.
For example, move the mouse vertically and isolate the 64 byte input packets (if there are any), repeat with horizontal movement.
Then do the same with the rest of the buttons, again, while the GUI is running.
I've been having trouble reading the HID descriptor to figure the packets out.

Additionally, I think it'd be better if you opened a PR against the bragi branch with the art and the proper button positioning. If it's correct, I'll merge it. Regarding the positioning, it's mostly guesswork at this point.

Electro707, 2020-08-25:
@tatokis Sorry for the late reply, I've just been procrastinating on my life :upside_down_face: 
Anyways, sadly the LED doesn't work anymore on the mouse, and I have attached a log here where the daemon is started while the gui is running, and then is existed without moving the mouse: 
[ckb-next-no-led-2020-08-24-9-18-15.txt](https://github.com/ckb-next/ckb-next/files/5120911/ckb-next-no-led-2020-08-24-9-18-15.txt)

As for the buttons while the GUI is running, here are the captured output from ckb-next-daemon:
Horizontal mouse movements: [ckb-next-horz.txt](https://github.com/ckb-next/ckb-next/files/5120945/ckb-next-horz.txt)
Vertical mouse movements: [ckb-next-vert.txt](https://github.com/ckb-next/ckb-next/files/5120946/ckb-next-vert.txt)
Button clicks: [ckb-next-buttons.txt](https://github.com/ckb-next/ckb-next/files/5120948/ckb-next-buttons.txt)

As for the GUI's graphic I made and the keymaps, I'll finish working/polishing it and create a pr soon hopefully.

tatokis, 2020-09-01:
@Electro707 I have no idea what happened regarding the LED. It seems like I somehow undid my fix from last time, which doesn't really make sense.

Either way, lighting should be fixed, and I have hopefully properly implemented the input packets. Please test the buttons with the GUI running again.

tatokis, 2020-09-18:
Just a quick update.

Basic mouse functions work in wired mode now, including DPI control. Since DPI is handled by software, despite the device only having 3 stages, you can still set 5 through ckb and it'll work fine.
I'm not sure how this should be handled.

The DPI indicators can be controlled independently from the daemon only for now. I need to figure out how to make them indicate the DPI while also allowing for independent control. As in, you can set one to red, the next one to green, and the third one to blue.

```
echo rgb dpi0:ff0000 dpi1:00ff00 dpi2:0000ff > /dev/input/ckb1/cmd
```

Soon I'll finish getting the UI right with the image done by @Electro707 and then I'll move on to wireless.

Wired testing is still welcome.

mijofa, 2020-11-05:
So after over a month and a half of dealing with retail warranty/repair processes while physical retail weren't accepting customers in-person. I've ended up getting a different model mouse.
Wish I could help more, but I no longer have the device required for testing, thank you for your efforts though, and good luck to you all

Electro707, 2020-11-05:
@mijofa Was the issue with the mouse double-clicking? I am mentioning that because my mouse has the same issue every now and then, and looking into it the fault are the switches used (Chinese Omron switches). I am currently awaiting on replacement switches to see if the issue goes away.

mijofa, 2020-11-05:
Nah, was an issue with the scroll wheel. I felt/heard something snap when middle clicking once, from then on the scroll wheel only really worked if I was pushing it to the left at the same time.
Probably a similar issue with regards to cheap switches being used though. I got the feeling it would be an easy fix if I opened it up, but was still very much within warranty, so I got the retailer to sort it out instead

Electro707, 2020-12-05:
Just to trow an update, I have tried the recent branch with the dongle, and the left and right click do not seem to be working. Looking thru the daemon logs it looks like the dongle isn't sending back the full 64-byte data in which the program determines what to do with the HID input (only receives the 9 byte from the dongle), so I recon it has to do with initialization. I have included a log of the daemon with the dongle and usb, just initializing then left clicking a couple of times. 
[dongle.txt](https://github.com/ckb-next/ckb-next/files/5646642/dongle.txt)
[usb.txt](https://github.com/ckb-next/ckb-next/files/5646643/usb.txt)
I've traced the error message (on the dongle) to the setactive_bragi function, when it switches the mouse to software mode. If I have the chance and launch Icue I'll see what's different.

MordicusEtCubitus, 2020-12-05:
Hi,

First I would like to say that even if I do not participate in solving this issue my bounty of €500 still effective and I read all your posts
.
Also I would like to get your advice:

The wheel of my mouse IRONCLAW-RGB-WIRELESS is broken. 

Apparently this is a recurrent defect as seen in users comment: Many of the buyers in France have the same issue: https://www.ldlc.com/fiche/PB00271133.html#clients-review

So I think I will use the warranty to switch to this model:  HARPOON RGB WIRELESS
https://www.corsair.com/fr/fr/Cat%C3%A9gories/Produits/Souris-gaming/HARPOON-RGB-WIRELESS-Gaming-Mouse/p/CH-9311011-EU

But before doing that do you know if this mouse can be supported by the development you are working on, or should I keep the current model ?

Thanks for your reply.

Electro707, 2020-12-06:
@MordicusEtCubitus Issue #346 exists regarding that same model mouse. I haven't been as active in the development of this issue, but from what I could tell once the bragi protocol gets polished up (especially for the wireless dongle), adding that mouse should a relative breeze.

tatokis, 2020-12-06:
@MordicusEtCubitus Thank you.
The work needed is exactly the same, so you're safe to get the Harpoon wireless.

@Electro707 the problem is that those long packets get sent when the device is in software mode. In order to set a device to software mode through the dongle, you need to tell the dongle which device to send the message to. It's the first byte in the packet if I recall correctly.

I'll resume the work on this after I'm done with https://github.com/ckb-next/ckb-next/issues/495 which should be pretty soon.

Electro707, 2020-12-07:
I have captured the dongle's communication while Icue was starting out (with wireshark):
[ironclaw_w_dongle_init.zip](https://github.com/ckb-next/ckb-next/files/5649815/ironclaw_w_dongle_init.zip)
*Note: Use filter ```usb.device_address == 6 && usb.data_len == 64  && usbhid.data[0:2] != 08:02 && usbhid.data[0:2] != 00:02``` for an easier time analyzing the data*

I have noticed that it sends the same software mode initialization command ckb does (08:01:03:00:02), but there seems to be some paring commands going on. May not be as easy as changing a byte on the software mode packet (although I hope I'm wrong about that)

JPA1238, 2021-03-21:
Is there any update on the ability to use a wireless mouse like an ironclaw wireless or harpoon wireless? With Bluetooth or with dongle?

Thanks for adding wired support for these mice already.

NicTanghe, 2021-07-20:
Hello the mouse does not show up for me in wired mode.
ANd id verry much like to b able to assighn functions 2 it.

Am i doing something wrong or does the wired mode on the wireless mouse just also not work yet ?

tatokis, 2021-07-20:
You need to stop the daemon and run it with --enable-experimental. I've been meaning to add instructions on how to do that for some time now.

sniff122, 2021-08-18:
> You need to stop ckb-next-daemon and run it with `--enable-experimental`. I've been meaning to add instructions on how to do that for some time now.

Enabling experimental in the daemon made my dongle show up in ckb-next but the mouse doesnt show unless its plugged in over USB. Additionally, when i made a button mapping change, it doesnt apply with the mouse plugged in (or unplugged over wireless). Is this expected behaviour?

tatokis, 2021-08-18:
That is expected at the moment unfortunately. We need to find someone with both a keyboard and a mouse connected to the same dongle to figure out how to tell which packet comes from which device and then forward it to the appropriate virtual device for remapping to work. 
If I'm not mistaken, battery reading is also not working wirelessly at the moment.

MordicusEtCubitus, 2021-08-18:
Hi,

If you don't have such hardware with you I can buy one and send it to 
one of the dev team
Give me the model that you need
It will be a gift to your effort

Best regards,

Gaël,

[30 more replies]