ckb-next/ckb-next #559: Running "ckb-next -b" more than once opens GUI

ashirviskas, 2020-03-29:
## General information

### Source
1:0.4.2.r51.g50865dc-1 (latest git)

## Logs & crash reports

ckb-next-dev-detect - Sun Mar 29 15:39:45 EEST 2020

```
Kernel: Linux matitank 5.5.13-arch1-1 #1 SMP PREEMPT Wed, 25 Mar 2020 16:04:40 +0000 x86_64 GNU/Linux

OS Version:
LSB Version:	1.4
Distributor ID:	Arch
Description:	Arch Linux
Release:	rolling
Codename:	n/a

USB Devices:
Couldn't open device, some information will be missing

Bus 005 Device 003: ID 1b1c:1b2d Corsair Corsair Gaming K95 RGB PLATINUM Keyboard
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1b1c Corsair
  idProduct          0x1b2d 
  bcdDevice            3.21
  iManufacturer           1 
  iProduct                2 
  iSerial                 3 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0042
    bNumInterfaces          2
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     192
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
[... 255 lines cut ...]
	liblz4.so.1 => /usr/lib/liblz4.so.1 (0x00007f22d5c89000)
	libgcrypt.so.20 => /usr/lib/libgcrypt.so.20 (0x00007f22d5b6b000)
	libicudata.so.65 => /usr/lib/libicudata.so.65 (0x00007f22d40ba000)
	libpcre.so.1 => /usr/lib/libpcre.so.1 (0x00007f22d4046000)
	libFLAC.so.8 => /usr/lib/libFLAC.so.8 (0x00007f22d4007000)
	libogg.so.0 => /usr/lib/libogg.so.0 (0x00007f22d3ffc000)
	libvorbis.so.0 => /usr/lib/libvorbis.so.0 (0x00007f22d3dd0000)
	libvorbisenc.so.2 => /usr/lib/libvorbisenc.so.2 (0x00007f22d3b27000)
	libX11.so.6 => /usr/lib/libX11.so.6 (0x00007f22d39e5000)
	libbz2.so.1.0 => /usr/lib/libbz2.so.1.0 (0x00007f22d39d2000)
	libgpg-error.so.0 => /usr/lib/libgpg-error.so.0 (0x00007f22d39b0000)
```

## What is the program's current behaviour?

* Run `ckb-next -b` 2 times
* First time it starts in background
* Second time it produces this: `❯ ckb-next -b
QObject::startTimer: Timers cannot have negative intervals
ckb-next is already running. Exiting.
`  and opens GUI
## What is the expected behaviour of the program?
* Run `ckb-next -b` 2 times
* First time it starts in background
* Second time it does nothing as it's already running in background

tatokis, 2020-03-29:
What's your use case?
It's set like that so that if you have an autostart shortcut, and you manually open it, it will focus the GUI instead of doing nothing.

ashirviskas, 2020-03-29:
I added auto start script to i3wm. The issue is that when I restart i3, it opens a new ckb-next window and I have to manually close it every time. 

To be fair, `-b` option indicates that I want it to launch in the background, so this is an undocumented behaviour territory.

tatokis, 2020-03-29:
I see why that can be an issue, yeah. I'll see if I can make a quick patch for it.

tatokis, 2020-03-29:
https://github.com/ckb-next/ckb-next/commit/f7f93159753b603e021d0b6946f110a52380b4c0
this should fix it

ashirviskas, 2020-03-29:
That was fast.

Thanks, it's working perfectly right now!

closed 2020-03-29 by ashirviskas