ckb-next/ckb-next #560: USB Connection Loss when keyboard used with hub

dominik-code, 2020-04-01:
## General information

### Source
 * Package from AUR

## Logs & crash reports

[ckb-next-report-1585757877.gz](https://github.com/ckb-next/ckb-next/files/4416393/ckb-next-report-1585757877.gz)

## What is the program's current behaviour?

Everything is working fine. But after 15-60 seconds the keyboard will stop working (lights still work but no input is registered). Unplugging and plugging in the keyboard from the USB hub helps for the next 15-60 seconds. When the ckb-next tool is closed completely the keyboard works as expected without issues for at least 8 hours.

## What is the expected behaviour of the program?

Keyboard works after 60 seconds when tool is still open.

## Any other notes & comments?

The keyboard seems to work with the tool running without the USB hub, but for me this is not really an option.

Hub used (with power supply): ID 1a40:0201 Terminus Technology Inc. FE 2.1 7-port Hub

ZirconiumX, 2020-04-04:
Unfortunately, this is due to bugs in Corsair firmware, where the keyboard falls off the USB hub[1] and gets reset by the kernel. You should add the [USB HID](https://github.com/ckb-next/ckb-next/wiki/Troubleshooting#usbhid-quirks) and [USB core](https://github.com/ckb-next/ckb-next/wiki/Troubleshooting#usbcore-quirks) quirks to your kernel command line, as these mitigate *some* of the problems.

However, Corsair having difficulties with USB hubs is well documented across our issue tracker, and the only fix I can suggest is directly plugging your keyboard into your computer.

[1]: e.g. `ckb-next-daemon[1214]: [E] os_usbsend (via led_keyboard.c:163):  Cannot send after transport endpoint shutdown, res = 0xffffffff` (where the device falls off the bus) and `ckb-next-daemon[1214]: [E] os_usbsend (via firmware.c:24):  Connection timed out, res = 0xffffffff` (where the device stops responding).

dominik-code, 2020-04-06:
But the strange thing is, without the software running everything works fine. (even via the USB Hub). As the hub is used as a KVM switch plugging the keyboard directly is not possible currently for my use case. Guess i'll have to live with this Corsair bug for the next few months.

I'll try some fixes and report if i find a reliable solution.

Thanks anyway for your time and the software. I love it (used without a KVM switch).

closed 2020-04-06 by dominik-code

tatokis, 2020-04-06:
Some more suggestions to try if the quirks don't work: 

Make sure the kvm switch is plugged in on the rear USB ports. 
You can also try plugging it in to a USB2 instead of USB3 port and vice versa. 

We've had someone else with a kvm and a keyboard failing to respond to packets. The USB cable on their KVM switch was detachable, so they replaced it with a shorter length/better quality one and the keyboard started working fine. That might be something to consider if possible.