ckb-next/ckb-next #564: Keyboard + Mouse stop working after connected to Dell TB-16

usmcamp0811, 2020-04-18:
- Search in open and closed [Issues](https://github.com/ckb-next/ckb-next/issues).
- If possible, try the latest code by building the software manually.
- File unrelated issues separately.

**Click the `Generate Report` button in the `Settings` tab of the ckb-next user interface and attach the generated report to this issue.**

## General information

I have an XPS 15 (9570) + TB16  running Arch Linux with the `ckb-next-git` version installed.

Most of the time when I connect my `K70 LUX RGB` and `M65 RGB ELITE` to the USB ports on my `TB16` they do not work. `ckb-next` recognizes them, the RGB color scheme gets loaded but they just aren't working. I have noticed that I have about a second or two after I connect them that they do work. The same thing happens when connecting to a USB hub that is connected to the `TB16`. Sometimes though they work just fine, at least until my computer goes to sleep or is otherwise left unattended for some time. 

It doesn't seem like there is anything wrong with the USB ports(I've tried all ports) and Linux sees the keyboard and mouse. I even have a webcam and a smartcard reader hooked up to the `TB16` and they both work just fine. 

Presently I have my keyboard and mouse connected directly to the laptop and have ZERO problems. Not knowing much, my guess is it has maybe something to do with the polling speed of the USB ports when connected to the hub.. I dunno.. 

## Report
```
ckb-next-dev-detect - Sat Apr 18 09:26:46 CDT 2020

```
Kernel: Linux Darkhorse 5.6.3-arch1-1 #1 SMP PREEMPT Wed, 08 Apr 2020 07:47:16 +0000 x86_64 GNU/Linux

OS Version:

USB Devices:
Couldn't open device, some information will be missing
Couldn't open device, some information will be missing

Bus 001 Device 017: ID 1b1c:1b33 Corsair Corsair Gaming K70 LUX RGB Keyboard 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1b1c Corsair
  idProduct          0x1b33 
  bcdDevice            3.08
  iManufacturer           1 
  iProduct                2 
  iSerial                 3 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0042
    bNumInterfaces          2
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
[... 1331 lines cut ...]
	libkrb5support.so.0 => /usr/lib/libkrb5support.so.0 (0x00007f13728a0000)
	libkeyutils.so.1 => /usr/lib/libkeyutils.so.1 (0x00007f1372899000)
	libresolv.so.2 => /usr/lib/libresolv.so.2 (0x00007f1372880000)
	librt.so.1 => /usr/lib/librt.so.1 (0x00007f1372875000)
	libsndfile.so.1 => /usr/lib/libsndfile.so.1 (0x00007f13727f7000)
	libasyncns.so.0 => /usr/lib/libasyncns.so.0 (0x00007f13727ef000)
	libGLdispatch.so.0 => /usr/lib/libGLdispatch.so.0 (0x00007f1372738000)
	libGLX.so.0 => /usr/lib/libGLX.so.0 (0x00007f1372705000)
	libfreetype.so.6 => /usr/lib/libfreetype.so.6 (0x00007f1372637000)
	libgraphite2.so.3 => /usr/lib/libgraphite2.so.3 (0x00007f1372610000)
	liblzma.so.5 => /usr/lib/liblzma.so.5 (0x00007f13725e8000)
	liblz4.so.1 => /usr/lib/liblz4.so.1 (0x00007f13725c6000)
	libgcrypt.so.20 => /usr/lib/libgcrypt.so.20 (0x00007f13724a8000)
	libicudata.so.65 => /usr/lib/libicudata.so.65 (0x00007f13709f7000)
	libpcre.so.1 => /usr/lib/libpcre.so.1 (0x00007f1370985000)
	libFLAC.so.8 => /usr/lib/libFLAC.so.8 (0x00007f1370944000)
	libogg.so.0 => /usr/lib/libogg.so.0 (0x00007f1370939000)
	libvorbis.so.0 => /usr/lib/libvorbis.so.0 (0x00007f137070d000)
	libvorbisenc.so.2 => /usr/lib/libvorbisenc.so.2 (0x00007f1370464000)
	libX11.so.6 => /usr/lib/libX11.so.6 (0x00007f1370324000)
	libbz2.so.1.0 => /usr/lib/libbz2.so.1.0 (0x00007f1370311000)
	libgpg-error.so.0 => /usr/lib/libgpg-error.so.0 (0x00007f13702ed000)
```

```

tatokis, 2020-04-18:
There's nothing we can do about it. Linux and corsair devices don't play nicely with USB hubs, which I assume the TB16 is in this case.

You may try adding the usbcore quirks as per the wiki, but I don't think it will help in this case.
This issue has also been observed with long USB cables.

Maybe if you change the pollrate on the devices it might help.

gravityfargo, 2024-05-02:
This issue has been marked as stale due to age and/or lack of activity. An explanation/solution was presented as well. 

Please feel free to reopen it if you have more to add or if the issue still needs attention. However, it would be best to [open a new issue](https://github.com/ckb-next/ckb-next/issues/new/choose) and provide a newly generated report issue using our updated templates.

closed 2024-05-02 by gravityfargo