ckb-next/ckb-next #566: [Question] Change all devices modes from binding

JulyanoF, 2020-04-20:
For example, I can create a binding to change only keyboard mode (Special - Switch to mode).
Is it possible to create a key binding (keyboard) that changes all devices modes (with same names)?

iivvoo, 2020-05-15:
This would be really useful - not sure how likely it is to have multiple different mice (or keyboards) connected to your device.

For me I have three different scimitar mice on three different locations, permanently hooked to a docking station, and I have to redefine my bindings for each of them each time.

tatokis, 2020-05-15:
As a workaround, you can abuse the ckb-next --mode argument. You could make a binding that runs a program, with that program being ckb-next --mode "YourMode".

If you want a mode under a specific profile, it becomes ckb-next --profile "ProfileName" --mode "ModeName".

iivvoo, 2020-05-15:
> As a workaround, you can abuse the ckb-next 

I might be misunderstanding profiles and modes, but each mouse seems to have its own distinct tab, set of profiles/modes. Setting to "Profile A" "Mode B" which is defined on mouse `1500A021AF4504455xxxxxxxxxx01BC4` does not work if mouse `1701A007AF4504455xxxxxxxxxx01BC5` is plugged in

For now it's doable to just redefine everything for all three mice, but to me it would make sense to just treat them al as the same.

Thanks for your great work though! Apperciate it!

tatokis, 2020-05-15:
I don't really understand what you mean by that.

If you run in a terminal ckb-next --profile "Profile A" --mode "Mode B", it will go through all connected devices and switch them to Mode B inside Profile A, as long as they are named the same.

The idea would be that you get ckb-next to run that as a keybind.

It's not really possible to treat them the same.