ckb-next/ckb-next #567: ckb-next-daemon preventing suspend

pakro, 2020-04-22:
- Search in open and closed [Issues](https://github.com/ckb-next/ckb-next/issues).
- If possible, try the latest code by building the software manually.
- File unrelated issues separately.

**Click the `Generate Report` button in the `Settings` tab of the ckb-next user interface and attach the generated report to this issue.**

## General information
I remember there was an update some days ago. Today I realized that I can't suspend. I starts suspending and then I'm on the login screen (I lock my session when suspeding). Normally this works but honestly it sometimes has also its issues. But this time every time I try to suspend the system it throws me back to the login screen.

### Source
I'm using the AUR https://aur.archlinux.org/packages/ckb-next-git/.
Updates are made regularly with yay -Syu --devel

## Logs & crash reports

Here is the relevant output from `sudo journalctl --since "10 minutes ago"`

```
Apr 22 19:25:11 homestation systemd[1]: Reached target Sleep.
Apr 22 19:25:11 homestation systemd[1]: Starting Suspend...
Apr 22 19:25:11 homestation systemd-sleep[43087]: Suspending system...
Apr 22 19:25:11 homestation kernel: PM: suspend entry (deep)
Apr 22 19:25:11 homestation kernel: Filesystems sync: 0.010 seconds
Apr 22 19:25:12 homestation rtkit-daemon[1318]: Supervising 0 threads of 0 processes of 0 users.
Apr 22 19:25:12 homestation rtkit-daemon[1318]: Successfully made thread 43091 of process 1311 owned by '1000' RT at priority 5.
Apr 22 19:25:12 homestation rtkit-daemon[1318]: Supervising 1 threads of 1 processes of 1 users.
Apr 22 19:25:12 homestation rtkit-daemon[1318]: Supervising 0 threads of 0 processes of 1 users.
Apr 22 19:25:12 homestation rtkit-daemon[1318]: Successfully made thread 43092 of process 1311 owned by '1000' RT at priority 5.
Apr 22 19:25:12 homestation rtkit-daemon[1318]: Supervising 1 threads of 1 processes of 1 users.
Apr 22 19:25:32 homestation ckb-next-daemon[1020]: [I] System has woken from sleep
Apr 22 19:25:32 homestation kernel: Freezing user space processes ... 
Apr 22 19:25:32 homestation kernel: Freezing of tasks failed after 20.006 seconds (3 tasks refusing to freeze, wq_busy=0):
Apr 22 19:25:32 homestation kernel: pool-xfce4-pane D    0 14394   1217 0x00004084
Apr 22 19:25:32 homestation kernel: Call Trace:
Apr 22 19:25:32 homestation kernel:  ? __schedule+0x2e8/0x7a0
Apr 22 19:25:32 homestation kernel:  schedule+0x46/0xf0
Apr 22 19:25:32 homestation kernel:  request_wait_answer+0x127/0x270 [fuse]
Apr 22 19:25:32 homestation kernel:  ? wait_woken+0x70/0x70
Apr 22 19:25:32 homestation kernel:  ? wait_woken+0x70/0x70
Apr 22 19:25:32 homestation kernel:  fuse_simple_request+0x196/0x290 [fuse]
Apr 22 19:25:32 homestation kernel:  fuse_statfs+0xe5/0x140 [fuse]
Apr 22 19:25:32 homestation kernel:  statfs_by_dentry+0x67/0x90
Apr 22 19:25:32 homestation kernel:  vfs_statfs+0x16/0xc0
Apr 22 19:25:32 homestation kernel:  user_statfs+0x66/0xb0
Apr 22 19:25:32 homestation kernel:  __do_sys_statfs+0x31/0x60
Apr 22 19:25:32 homestation kernel:  do_syscall_64+0x4e/0x150
Apr 22 19:25:32 homestation kernel:  entry_SYSCALL_64_after_hwframe+0x44/0xa9
Apr 22 19:25:32 homestation kernel: RIP: 0033:0x7f5cce812efb
Apr 22 19:25:32 homestation kernel: Code: Bad RIP value.
Apr 22 19:25:32 homestation kernel: RSP: 002b:00007f5cc77fdab8 EFLAGS: 00000246 ORIG_RAX: 0000000000000089
Apr 22 19:25:32 homestation kernel: RAX: ffffffffffffffda RBX: 0000557012f58cc0 RCX: 00007f5cce812efb
Apr 22 19:25:32 homestation kernel: RDX: 000055701303a9a0 RSI: 00007f5cc77fdae0 RDI: 0000557012f35620
Apr 22 19:25:32 homestation kernel: RBP: 000055701302c550 R08: 00007f5cc000f188 R09: 00007f5cc000f190
Apr 22 19:25:32 homestation kernel: R10: 00005570129253e8 R11: 0000000000000246 R12: 00007f5cc77fdc40
Apr 22 19:25:32 homestation kernel: R13: 0000557012f32560 R14: 00007f5cc77fdd80 R15: 00007f5cc77fe700
Apr 22 19:25:32 homestation kernel: pool-xfce4-pane D    0 14395   1217 0x00004084
Apr 22 19:25:32 homestation kernel: Call Trace:
Apr 22 19:25:32 homestation kernel:  ? __schedule+0x2e8/0x7a0

```

Attach:
 * The generated report from the section above
 * Any additional crash reports you've managed to gather

## What is the program's current behaviour?

The daemon wakes the system from sleep.

## What is the expected behaviour of the program?

The daemon should not wake the system from sleep.

## Any other notes & comments?

--

ZirconiumX, 2020-04-22:
```
Apr 22 19:25:32 homestation kernel: Freezing of tasks failed after 20.006 seconds (3 tasks refusing to freeze, wq_busy=0):
Apr 22 19:25:32 homestation kernel: pool-xfce4-pane D    0 14394   1217 0x00004084
```
If I'm reading the kernel log correctly here, that's XFCE not suspending.

pakro, 2020-04-22:
You are right. Rebooted the system and tried to suspend it. It worked. Sorry for wasting your time. I will close this!

closed 2020-04-22 by pakro