ckb-next/ckb-next #571: Found this official SDK, so, Fan (RGB) control possible?

Xardas220, 2020-04-24:
I found this SDK in C++. So is this a simplification for the developers of this software? I think so. :)
So now it should be easier to implement the fan control and the RGB control of these.

https://github.com/CorsairOfficial/cue-sdk

Xardas

fleischie, 2020-04-24:
Certainly interesting to see they have an "official" "open source" SDK available. From a quick glance it only supports macos, as well as windows and I suppose it requires cue to run in the background.

So naively I don't see a good way to have ckb-next "run on" the cue sdk...

tatokis, 2020-04-24:
Yeah. All that SDK does is provide a way for applications to talk to CUE.
It is of no use to us unfortunately. Even if it was though, the license wouldn't be compatible.

closed 2020-04-24 by tatokis