ckb-next/ckb-next #572: Opening caja with macro key crashes ckb-next

the-very-edgy-yoshi, 2020-04-25:
I have a Corsair K55 keyboard that I use with ckb-next version 0.4.2-57-g322ce69, compiled from source on Ubuntu MATE 20.04. I use the G2 key to open my file manager, Caja. However, the first time attempting this crashes ckb-next. After reopening the program, using the G2 key to open Caja works fine.

ZirconiumX, 2020-05-14:
Can you try to run ckb-next under a debugger like GDB?

the-very-edgy-yoshi, 2020-05-20:
I ran the program under GDB after ckb-next was removed from my list of startup applications. Upon starting the program, the program worked normally when I ran it. Here are the results:

Starting program: /usr/local/bin/ckb-next 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7ffff26de700 (LWP 6871)]
[New Thread 0x7ffff1e06700 (LWP 6872)]
QObject::startTimer: Timers cannot have negative intervals
[Detaching after fork from child process 6873]
[Detaching after fork from child process 6874]
[New Thread 0x7ffff095f700 (LWP 6875)]
[New Thread 0x7fffe3f4f700 (LWP 6876)]
[New Thread 0x7fffe374e700 (LWP 6877)]
[New Thread 0x7fffe2f4d700 (LWP 6878)]
[New Thread 0x7fffe26fe700 (LWP 6879)]
[New Thread 0x7fffe1c2f700 (LWP 6880)]
Path to settings: "/home/lily/.config/ckb-next/ckb-next.conf"
[New Thread 0x7fffe0fda700 (LWP 6881)]
Using AppIndicator
[Detaching after fork from child process 6882]
Scanning "/usr/local/libexec/ckb-next-animations/gradient"
[Detaching after fork from child process 6883]
Scanning "/usr/local/libexec/ckb-next-animations/heat"
[Detaching after fork from child process 6884]
Scanning "/usr/local/libexec/ckb-next-animations/invaders"
[Detaching after fork from child process 6885]
Scanning "/usr/local/libexec/ckb-next-animations/life"
[Detaching after fork from child process 6886]
Scanning "/usr/local/libexec/ckb-next-animations/mviz"
[Detaching after fork from child process 6887]
Scanning "/usr/local/libexec/ckb-next-animations/pinwheel"
[Detaching after fork from child process 6888]
Scanning "/usr/local/libexec/ckb-next-animations/pipe"
[Detaching after fork from child process 6889]
Scanning "/usr/local/libexec/ckb-next-animations/rain"
[Detaching after fork from child process 6890]
Scanning "/usr/local/libexec/ckb-next-animations/random"
[Detaching after fork from child process 6891]
Scanning "/usr/local/libexec/ckb-next-animations/ripple"
[Detaching after fork from child process 6892]
Scanning "/usr/local/libexec/ckb-next-animations/snake"
[Detaching after fork from child process 6893]
Scanning "/usr/local/libexec/ckb-next-animations/wave"
[New Thread 0x7fffc9aa0700 (LWP 6894)]
[New Thread 0x7fffc91e6700 (LWP 6895)]
[Detaching after fork from child process 6896]
Starting "/usr/local/libexec/ckb-next-animations/wave"
[New Thread 0x7fffbffff700 (LWP 6897)]
[New Thread 0x7fffbf7fe700 (LWP 6898)]
[New Thread 0x7fffbeffd700 (LWP 6899)]
[Thread 0x7fffbffff700 (LWP 6897) exited]
[Thread 0x7fffe26fe700 (LWP 6879) exited]
[New Thread 0x7fffe26fe700 (LWP 6900)]
[Detaching after fork from child process 6901]
Downloaded new firmware list. 38 entries found.
[Detaching after fork from child process 6902]
[Thread 0x7fffe1c2f700 (LWP 6880) exited]
[Detaching after fork from child process 6938]
[Thread 0x7fffbf7fe700 (LWP 6898) exited]