ckb-next/ckb-next #573: Linux Mint Cinnamon Tricia 19.3: Default Debian 18.04 Installation Instructions (Newb Question/Issue)

Ketenks, 2020-04-26:
## General information
[ckb-next-report-1587892067.gz](https://github.com/ckb-next/ckb-next/files/4534987/ckb-next-report-1587892067.gz)

### Source
Default Debian 18.04 Installation instructions

## Logs & crash reports
attached

## What is the program's current behaviour?

Newb question or issue: in terminal I can get the program running by entering in
ckb-next
and then it will run the program however there is no perceivable means of making this program recognized as a real program outside of the terminal where I could pin it to taskbar and do similar things like the other installed programs.
I cannot "Bring to Foreground" because it is not anywhere to be seen in the panel of the desktop.

## What is the expected behaviour of the program?
the program needs to be exactly as though i had installed it on the package manager, instead of from the terminal so that I can pin it to taskbar and click on an executable and "Bring to Foreground" while it is in the tray. 

## Any other notes & comments?

tatokis, 2020-04-26:
Try logging out and back in (or rebooting). It's possible cinnamon for some reason didn't pick up on it

Ketenks, 2020-04-26:
Yes that was the situation. Logged back in today and it was being recognized. However, it does not have a "tray" in my panel or it is not going to the background? I have to keep it as a pinned program to access it again instead of the panel's tray. But oh well. It works for me so I think I'm good.

closed 2020-04-27 by Ketenks