ckb-next/ckb-next #575: dark core se pro not reconized

olivier-belleval, 2020-04-27:
Hi

first of all thank you for your job and help other to have good stuff working on linux
i have a k95 working perfectly on linux mint and just buy the dark core pro SE , But [ this fuc... but :( ]
it not reconized as say in the title 
here the log report 
[ckb-next-report-1587986881.gz](https://github.com/ckb-next/ckb-next/files/4539291/ckb-next-report-1587986881.gz)
thank for help

ZirconiumX, 2020-05-14:
Duplicate of #101.

olivier-belleval, 2020-05-14:
Hmmm you close an issue because it s the of another make 2 year ago ..... for the new mouse .... interesting

ZirconiumX, 2020-05-14:
Fundamentally the problem is the same: ckb-next doesn't have the necessary infrastructure for wireless devices, and the two unpaid, burned-out developers have prioritised other things in the meantime.

tatokis, 2020-05-14:
Either way, this is a bragi device

olokos, 2025-05-31:
At this point, all features, such as RGB, animations, zones, brightness and the mouse icon are visible in Lightning section.
Binding is a bit untested, but should work just fine.
Performance tab works well, DPI stages can be changed, disabled.

Sniper button changes profiles, instead of going into the sniper DPI stage, thats a small bug for sure.
Device tab has layout like for keyboard, not sure if that's correct.
Serial number gets displayed in wired mode and for the dongle, but mouse wireless serial does not appear.
This can lead to issues, if more than one mouse of this type is connected wirelessly, let us know if You have two of those, what happens.
FW version/bootloader version/battery % and charging work well.

Seems like there's no read/write access to hardware profiles, there is only `P - Demo` profile selectable, where new modes can be created, this is obviously a bug. Shouldn't cause too many issues though, apart from having to set it up manually the first time after installing ckb-next.

With latest PR merged, the next release/build of ckb-next will support Dark Core RGB Pro SE out-of-the-box, no experimental modes enabling needed, for both wired and wireless modes.

Let us know if You get any issues, on my side though, apart from the minor issues above, the mouse can be fully and properly used without too much worries now. I'm especially happy that somebody has made battery level and indicator working, since there's a nice ckb-next tray icon for its battery now :)

olokos, 2025-05-31:
```
bragi_process_notification (bragi_notification.c:36): ckb4: Unknown bragi notification for dev 4
ckb-next-daemon[1568]: [E] os_usb_interrupt_out (via bragi_common.c:80): Protocol error, res = 0xffffffff
ckb-next-daemon[1568]: [I] Attempting reset...
ckb-next-daemon[1568]: [I] Stopping input thread for /dev/input/ckb4
ckb-next-daemon[1568]: [E] os_resetusb (via usb.c:667): resetusb failed: No such device
ckb-next-daemon[1568]: [E] usb_tryreset (usb.c:675): Reset failed. Disconnecting.
ckb-next-daemon[1568]: [I] Disconnecting /dev/input/ckb4
ckb-next-daemon[1568]: [I] Removed device path /dev/input/ckb4
ckb-next-daemon[1568]: [I] Caught internal signal SIGUSR2
ckb-next-daemon[1568]: [I] ckb4: Bragi poll thread shutting down due to 4 (Interrupted system call)

ckb-next-daemon[7609]: [I] ckb1: bragi dongle hotplug thread finished
ckb-next-daemon[7609]: [I] ckb1: bragi dongle hotplug thread started
ckb-next-daemon[7609]: [I] Found new bragi subdevice 1
ckb-next-daemon[7609]: [I] ckb1: bragi dongle hotplug thread started
ckb-next-daemon[7609]: [W] bragi_usb_read (via bragi_common.c:9): ckb1: Timeout while waiting for response
ckb-next-daemon[7609]: [W] bragi_usb_read (via bragi_common.c:9): ckb3: Timeout while waiting for response
ckb-next-daemon[7609]: [W] bragi_usb_read (via bragi_common.c:9): ckb1: Timeout while waiting for response
ckb-next-daemon[7609]: [W] bragi_usb_read (via bragi_common.c:9): ckb3: Timeout while waiting for response
ckb-next-daemon[7609]: [W] bragi_usb_read (via bragi_common.c:9): ckb1: Timeout while waiting for response
ckb-next-daemon[7609]: [W] bragi_usb_read (via bragi_common.c:9): ckb3: Timeout while waiting for response
```

There's one, quite important issue remaining:
Going wireless->wired, like when your mouse discharges, might crash daemon and restarting daemon and replugging dongle is a way to get back from wired to wireless.

Above logs are from current crashes, similar to the ones I reported earlier, but that's probably the last remaining serious issue. Charging mouse with Type-C wall charger, instead of connecting it directly to the PC is another way around this issue.

Generally when booting wired or wireless, there's no issue, but when changing between the modes after booting into the OS, there's the issue.