ckb-next/ckb-next #576: Nothing happens when sending data to a pipe for a pipe animation

selplacei, 2020-04-29:
Using ckb-next 0.4.2-3 from the AUR on Arch Linux with the ZEN kernel; keyboard: K70 LUX RGB.

Generated report: https://eeeee.s-ul.eu/TCUt5teG

When sending something to a pipe associated with a pipe animation (such as with `echo "ff0000ff" > /tmp/ckbpipe000`), the color doesn't change.

Minimal steps to reproduce on my machine:
- Create new mode
- Press "New Animation..." and select Pipe
- Choose a new pipe that's never been used before (for example, 10)
- Execute `echo "ff0000ff" > /tmp/ckbpipe010`
- Nothing changes, neither on the preview nor on the actual keyboard.

What I've tried:
- Rebooting, which deletes all files in /tmp
- Sending data to the pipe after closing the ckb-next window
- Binding a script with the echo command to a key instead of manually running it, and trying it with a delay
- Setting the animation to only one key, or the whole keyboard
- Adding a key code to the echo command

None of these made any changes. I should also mention that when I first tried using the pipe animation, it only updated when I clicked "Edit properties..." on the animation and then pressed OK. Now it doesn't update at all, and I can't seem to reproduce the previous behavior.

tatokis, 2020-04-29:
> Execute echo "ff0000ff" > /tmp/ckbpipe010

The command is "rgb rrggbbaa", not "rrggbbaa".
If it worked before, it's probably because of an accident, or a really old version from before it was merged in master, which might explain why it stopped working.

`echo "rgb ff0000ff" > /tmp/ckbpipe010`
or to just set the r key to red
`echo "rgb r:ff0000ff" > /tmp/ckbpipe010`

selplacei, 2020-04-29:
> The command is "rgb rrggbbaa"

My bad, I typed the wrong thing into the issue, but I did do the correct command when testing. I also tried using key codes, as indicated by the last item in "What I've tried".

selplacei, 2020-04-29:
Also, I rebooted again, this time after doing a full system upgrade, and it seems to work fine. ckb-next itself, though, was the latest version at the time of the problem. Should I close the issue?

ZirconiumX, 2020-05-14:
Since it works fine, I'll close this.

closed 2020-05-14 by ZirconiumX

joker-119, 2024-03-19:
Hate to necro here but I'm having the exact same issue, but I have nothing to update and rebooting didn't solve the issue.