ckb-next/ckb-next #578: Option to load profiles before login (no GUI)

harryausten, 2020-05-02:
Forgive me if I am wrong, but currently due to the nature of the daemon/GUI parts of the program, in order to load saved profiles for Corsair hardware, the GUI part of the program has to be run. This program does not appear to be able to run unless the GUI is active (as it relies on Qt bindings), therefore profiles configured within the `ckb-next` tool cannot be loaded from a headless environment.

As some hardware I own is not (yet?) supported by the *save to hardware* functionality, this means that the keyboard/mouse default to their RGB rainbow effects until I login. Once logged in, the `ckb-next` program starts up in the background (tray icon), the profiles are loaded and the rainbow stops.

I was wondering if there was any use in having a way to run the `ckb-next` program in a headless mode? so that profiles can be loaded from a specified profile in a `systemd` service directly after the `ckb-next-daemon` is initialised. This would allow earlier loading of configured RGB profiles and therefore less rainbow vomit :wink:

owendelong, 2020-05-29:
I may take this on. Depending on the complexity of reading the command files, it seems to me that a CLI based script to read the user's config on login (which could then be added to .<shell>rc shouldn't be too difficult.

OmgImAlexis, 2021-11-25:
As a new user I thought this was a bug. I couldn't workout why even though I had the daemon running my profiles would only work while the GUI was open and I guess this is the reason. :disappointed:

massimopavoni, 2023-03-17:
Same problem here. Do the devices with unsupported write to hardware functionality have some implementation problem that cannot still be resolved, so at some point they might be there, or is this idea of running headless the only solution?

owendelong, 2023-04-28:
Thanks for linking that PR... Indeed, if he can clean up the few remaining issues in the code, this looks very promising. Care to enlighten us, @tatokis as to why you don't like this use case?

Why require X11 to control a keyboard?