ckb-next/ckb-next #579: Support for Corsair Virtuoso ?

Joseon, 2020-05-06:
hey guys just wondering if corsair virtuoso headset will be supported? i saw that there's a WIP headset support but is that only for the void pro? is there anything  I could do to help?

Googlelt, 2020-06-01:
This would be ideal. currently only getting audio through one side of the headset. and the volume is very low.

cap-hector, 2020-07-19:
some news?

Ravenslofty, 2020-07-19:
It's unlikely to be supported any time soon; the Void Pro gave us enough headaches (which is why it's been sat in a branch for a while).

rvl, 2021-07-07:
Hi @Ravenslofty I'm possibly interested in trying to add (partial) support for the corsair virtuoso headset.
Would you be able point me to any work in progress and/or relevant information for headsets support?
What were the headaches for the Void Pro? Is the protocol quite different to that of keyboards for example?
Cheers.

gravityfargo, 2023-04-21:
closing in favor of #785

closed 2023-04-21 by gravityfargo