ckb-next/ckb-next #580: I have to unplug an plug in my Keyboard for it to work.

derbrachialist, 2020-05-10:
Every time I boot, I have to unplug and plug in my K95 Keyboard for it to work.
It does not register keystrokes and the LEDs are set to the default rainbow thing at boot.

Sur3, 2020-05-11:
Did you follow the usb quirks guide? https://github.com/ckb-next/ckb-next/wiki/Troubleshooting

derbrachialist, 2020-05-11:
I need some help with that.
I typed in the command and go the following quirks: `usbhid.quirks=0x1b1c:0x1b2d:0x20000408,0x1b1c:0x1b2e:0x20000408,0x1b1c:0x1b3b:0x20000408`
Do I just add this to my grub boot sequence? Just at the end?

derbrachialist, 2020-05-11:
I edited it to:
`GRUB_CMDLINE_LINUX_DEFAULT="quiet usbhid.quirks=0x1b1c:0x1b2d:0x20000408,0x1b1c:0x1b2e:0x20000408,0x1b1c:0x1b3b:0x20000408 rd.driver.pre=vfio-pci kvm.ignore_msrs=1 cryptdevice=UUID=a7bba2cf-...:luks-a7bba2cf... root=/dev/mapper/luks-a7bba2c... resume=/dev/mapper/luks-a7bba2cf..."`

Will reopen if it does not fix the problem or I did it wrong.

derbrachialist, 2020-05-13:
still got the problem at like every 5th boot

STiAT, 2020-05-14:
At boot or after login?

asking because it could possibly be related to https://github.com/ckb-next/ckb-next/issues/584

If it's at boot time it isn't related, but if it's after login it may be related, since that with the usb mouse occurs at ~20 % of the times as well, and seems to be related to usb detection.

derbrachialist, 2020-05-24:
After login.

I removed the GRUB_CMDLINE because it broke my drm protection and I could no longer watch Netflix.

Mageas, 2020-08-06:
I had the same problem. In my case, all I had to do was stop the daemon service and then turn it on.
I made a script that starts automatically when my computer starts.

```bash
#!/bin/bash

YOUR_DEVICE="CORSAIR SCIMITAR RGB ELITE Gaming Mouse";

get_devices_name () {
    for device in $(ls /sys/bus/usb/devices/*/product); do 
        echo $(cat $device);
    done 
}

force_usb_replug () {
    for device in $(ls /sys/bus/usb/devices/*/product); do 
        device_name=$(cat $device);
        if [[ "$device_name" == "$YOUR_DEVICE" ]]; then
            IFS='/' read -r -a device_ids <<< "$device"
            device_id="${device_ids[5]}"
            
            echo "${device_id}" | sudo tee /sys/bus/usb/drivers/usb/unbind
            echo "${device_id}" | sudo tee /sys/bus/usb/drivers/usb/bind
            sleep 1s
        fi
    done 
}

restart_daemon_service () {
    sudo systemctl stop ckb-next-daemon.service
    sleep 1s
    sudo systemctl start ckb-next-daemon.service
}

if [[ "$1" == "--name" ]]; then
    get_devices_name
elif [[ "$1" == "--force" ]]; then
    force_usb_replug
    restart_daemon_service
else
    restart_daemon_service
fi
```
How the script works:
--name: List of all devices.
--force: Force the USB to disconnect then restart the daemon (You have to change the content of YOUR_DEVICE variable with the name of your device)
Without flag: Restart the daemon

I run the script without any flags.
Hoping this might help you!

tatokis, 2020-08-17:
Can someone affected tell me if they have their device lights off or on before suspend?
Try leaving them on before suspending, and tell me if the keyboard comes back to life after a while.

CerulanLumina, 2020-10-03:
What's the status on this issue? It's still very prevalent for me, occurring on perhaps 80-90% of boots. Oddly enough, sometimes if I start typing in my password to sddm quickly enough I'll get a few characters in before it ceases to work. Even weirder if it stops working in the middle of a key press the key gets spammed repeatedly until unplugging the keyboard.

Ravenslofty, 2020-10-03:
Tasos made some commits to master to attempt to fix the situations he got
the keyboard to stop working when resuming from sleep. You could join us in
IRC and we can try to debug the issue there, but first make sure you are
using the latest Git code.

kenn213, 2021-03-05:
I have an issue where my keyboard won't respond after waking from suspend unless I unplug/replug it. Should I try the quirks or is that only a boot issue fix? Thanks.

maxlemieux, 2021-04-11:
> I have an issue where my keyboard won't respond after waking from suspend unless I unplug/replug it. Should I try the quirks or is that only a boot issue fix? Thanks.

Try the script by @greenfox1505 here - don't forget to make it executable or systemd can't run it: https://github.com/ckb-next/ckb-next/issues/456#issuecomment-529281287

CodexHere, 2023-06-05:
Here we are in mid 2023 and this is still a problem.

Has there been any investigation into this at all by the dev(s)?

I'm considering disabling the service until login, which would then prompt me for the password yet again with `pkexec`.
Hopefully that'll fix it, because it's getting more and more annoying to have to manually go to my PC to unplug/replug because it's not easily accessible.

LukasMueller187, 2023-06-24:
Yep, still an issue for me. I have no clue what the issue is.

CodexHere, 2023-06-24:
I ended up going the autostart route:

1) Disable the service from starting automatically
2) Add my user to `sudoers` file to specifically start the service without asking for password (not a blanket wildcard)
3) Add service start to autostart through a simple shell script.

Should be easy enough to find out how (I believe I've seen similar instruction on here), but if anyone needs info I'll update with better details (I'm on mobile atm)

LukasMueller187, 2023-06-24:
So I retried using the USB quirks listed in the troubleshooting part and with the line `usbhid.quirks=0x1B1C:0x1B11:0x4` (for my K95 RGB) added to the kernel parameters the issue is finally resolved (at least after 3 retries). I left ckb-daemon-next enabled as usual.

In order to add the kernel parameter `usbhid.quirks=0x1B1C:0x1B11:0x4` permanently (even if there's an update and grub config is rebuilt) add the parameter to `GRUB_CMDLINE_LINUX_DEFAULT` in `/etc/default/grub` and rerun `grub-mkconfig` (e.g. `sudo grub-mkconfig -o /boot/grub/grub.cfg`).

I ended up retrying because I noticed `usbhid 1-3:1.1: can't add hid device: -110` in dmesg as soon as the keyboard starts working (after ~40s).