ckb-next/ckb-next #581: udevadm trigger error

STiAT, 2020-05-11:
Hi there,

I'm experiencing an error when calling udevadm trigger when ckb-next-daemon is running.

In 0.4.2 it used to be two errors, in the current [master] it's one error remaining (though, still looking the same):

`Failed to write 'change' to '/sys/devices/virtual/input/input24/uevent': Cannot allocate memory`

This is certainly a device created by ckb-next, though, why it failed to write "change" with the error "Cannot allocate memory" when calling udevadm trigger is a mystery to me.

Maybe you can help out on this, it's currently an issue in solus that this error pops up every time reloading dev configs after updates, several users already mentioned it. A bit light to why this could happen in this combination would be nice.

Solus Bugtracker ID:
https://dev.getsol.us/T8734

ZirconiumX, 2020-05-11:
This seems tricky enough that you should probably join us in IRC so we can debug faster.

STiAT, 2020-05-11:
Spoken on IRC, seems like an issue with the bounds of the message supplied to the kernel in add_uevent_var (kobject_uevent.c).

I'll bugger the kernel guys with it and give you guys a nudge if they get to the conclusion there's something else wrong.

Thanks!

STiAT, 2020-05-11:
Talked to the ckb-next people, very helpful lot!

Short trace (by the ckb-next people), we're it sounds like we have an issue in the kernel itself or at least with the message passed to the kernel.

```
[40677.318139] ------------[ cut here ]------------
[40677.318141] add_uevent_var: buffer size too small
[40677.318183] WARNING: CPU: 2 PID: 25639 at lib/kobject_uevent.c:670 add_uevent_var+0xfa/0x110
[40677.318183] Modules linked in: pci_stub vboxpci(OE) vboxnetadp(OE) vboxnetflt(OE) vboxdrv(OE) binfmt_misc kvm_amd kvm mxm_wmi irqbypass rc_pixelview snd_emu10k1_synth snd_emux_synth crct10dif_pclmul snd_seq_midi_emul crc32_pclmul snd_seq_virmidi ghash_clmulni_intel tuner_simple tuner_types snd_emu10k1 snd_hda_codec_hdmi joydev tuner input_leds tda7432 snd_hda_intel aesni_intel tvaudio snd_util_mem snd_hda_codec snd_ac97_codec bttv ac97_bus snd_hda_core aes_x86_64 glue_helper crypto_simd snd_hwdep tveeprom snd_pcm_oss videobuf_dma_sg snd_mixer_oss cryptd videobuf_core tea575x snd_pcm amdgpu rc_core v4l2_common videodev fam15h_power mc emu10k1_gp gameport snd_seq_midi snd_seq_midi_event sp5100_tco snd_rawmidi gpu_sched snd_seq i2c_algo_bit ttm snd_seq_device drm_kms_helper snd_timer syscopyarea sysfillrect sysimgblt fb_sys_fops drm snd parport_serial soundcore wmi vhba(OE) parport_pc ppdev lp parport autofs4 hid_generic usbhid hid uas usb_storage i2c_piix4 tg3 firewire_ohci ahci
[40677.318216]  firewire_core pata_acpi libahci floppy
[40677.318220] CPU: 2 PID: 25639 Comm: udevadm Tainted: G           OE     5.3.8-uwuvyouwofty #1
[40677.318221] Hardware name: To Be Filled By O.E.M. To Be Filled By O.E.M./990FX Extreme4, BIOS P2.70 06/05/2014
[40677.318223] RIP: 0010:add_uevent_var+0xfa/0x110
[40677.318224] Code: 48 83 c4 50 5b 41 5c 5d c3 48 c7 c7 c8 fe 44 8d e8 a4 ad 69 ff 0f 0b b8 f4 ff ff ff eb d2 48 c7 c7 f0 fe 44 8d e8 8f ad 69 ff <0f> 0b b8 f4 ff ff ff eb bd e8 88 aa 69 ff 0f 1f 84 00 00 00 00 00
[40677.318225] RSP: 0018:ffffa68042cf7d40 EFLAGS: 00010286
[40677.318227] RAX: 0000000000000000 RBX: ffff9573ee690000 RCX: 0000000000000006
[40677.318227] RDX: 0000000000000007 RSI: 0000000000000096 RDI: ffff95751ea974d0
[40677.318228] RBP: ffffa68042cf7da0 R08: 0000000000000542 R09: 0000000000000034
[40677.318229] R10: 0000000000000000 R11: ffffa68042cf7be8 R12: 0000000000000005
[40677.318230] R13: ffff95751ac75300 R14: ffff9573e832f218 R15: 0000000000000000
[40677.318231] FS:  00007f27fc15b880(0000) GS:ffff95751ea80000(0000) knlGS:0000000000000000
[40677.318232] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[40677.318233] CR2: 000055c381d8bcf0 CR3: 000000049a2c8000 CR4: 00000000000406e0
[40677.318234] Call Trace:
[40677.318239]  ? dev_uevent+0xba/0x2e0
[40677.318241]  kobject_uevent_env+0x360/0x790
[40677.318244]  ? do_filp_open+0xa7/0x100
[40677.318245]  kobject_synth_uevent+0x2bb/0x32b
[40677.318247]  uevent_store+0x1c/0x30
[40677.318249]  ? _cond_resched+0x15/0x30
[40677.318251]  kernfs_fop_write+0x108/0x190
[40677.318253]  vfs_write+0xa5/0x1a0
[40677.318255]  ksys_write+0x59/0xd0
[40677.318257]  do_syscall_64+0x55/0x130
[40677.318260]  entry_SYSCALL_64_after_hwframe+0x44/0xa9
[40677.318262] RIP: 0033:0x7f27fc039317
[40677.318263] Code: 64 89 02 48 c7 c0 ff ff ff ff eb bb 0f 1f 80 00 00 00 00 f3 0f 1e fa 64 8b 04 25 18 00 00 00 85 c0 75 10 b8 01 00 00 00 0f 05 <48> 3d 00 f0 ff ff 77 51 c3 48 83 ec 28 48 89 54 24 18 48 89 74 24
[40677.318264] RSP: 002b:00007fff329c4948 EFLAGS: 00000246 ORIG_RAX: 0000000000000001
[40677.318266] RAX: ffffffffffffffda RBX: 0000000000000007 RCX: 00007f27fc039317
[40677.318266] RDX: 0000000000000007 RSI: 00007fff329c4a00 RDI: 0000000000000003
[40677.318267] RBP: 00007fff329c4a00 R08: 0000000000000007 R09: 7269762f73656369
[40677.318268] R10: 0000000000000000 R11: 0000000000000246 R12: 0000000000000007
[40677.318269] R13: 000055c381dc97c0 R14: 0000000000000007 R15: 00007f27fc1138a0
[40677.318270] ---[ end trace 0412a24c9ab61437 ]---
[40677.318272] synth uevent: /devices/virtual/input/input71: failed to send uevent
[40677.318273] input input71: uevent: failed to send synthetic uevent
```

This all boils down to the following kernel code in kobject_uevent.c:
```
int add_uevent_var(struct kobj_uevent_env *env, const char *format, ...)
{
	va_list args;
	int len;

	if (env->envp_idx >= ARRAY_SIZE(env->envp)) {
		WARN(1, KERN_ERR "add_uevent_var: too many keys\n");
		return -ENOMEM;
	}

	va_start(args, format);
	len = vsnprintf(&env->buf[env->buflen],
			sizeof(env->buf) - env->buflen,
			format, args);
	va_end(args);

	if (len >= (sizeof(env->buf) - env->buflen)) {
		WARN(1, KERN_ERR "add_uevent_var: buffer size too small\n");
		return -ENOMEM;
	}

	env->envp[env->envp_idx++] = &env->buf[env->buflen];
	env->buflen += len + 1;
	return 0;
}
```

To get a clue why this is happening there we'll probably need somebody familiar with input interfaces in the kernel, since for me it seems as if the udev message with args would have to exceed bounds of kobj_uevent_env->buf, which is why the error seems to occur. At least if I don't misread the code.

closed 2020-05-11 by STiAT

STiAT, 2020-05-15:
side-note if somebody looks this up: was fixed in ckb-next with a workaround until kernel patches come downstream in c5d3588 and 0ae46be (you need both patches).

rkmax, 2020-10-18:
@STiAT did you know when this will be release I have the version 0.4.2 and the problem is happening